openbmc/openbmc #1143: [Witherspoon w33 ] PowerControl: could not read power GPIO configuration

gkeishin, 2017-02-16:
**Driver info**
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170216042029
root@witherspoon:~# uptime
 05:52:27 up 13 min,  load average: 0.12, 0.32, 0.41
root@witherspoon:~# obmcutil state
 = BMC_READY
root@witherspoon:~#
```

**Log Snippet** 
```
Feb 16 05:40:40 witherspoon power_control.exe[846]: Power GPIO: call to getGpioConfiguration failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files
Feb 16 05:40:40 witherspoon power_control.exe[846]: ERROR PowerControl: could not read power GPIO configuration
Feb 16 05:40:41 witherspoon power_control.exe[846]: (process:846): GLib-CRITICAL **: g_variant_new_string: assertion 'string != NULL' failed
Feb 16 05:40:41 witherspoon power_control.exe[846]: ERROR PowerControl: GPIO setup (rc=2)
Feb 16 05:40:44 witherspoon power_control.exe[846]: ERROR PowerControl: GPIO open error (gpio=(null),rc=2)
```
**Suspecting the org.openbmc.managers.System  but its up and running**

```
root@witherspoon:~# systemctl status org.openbmc.managers.System |cat
● org.openbmc.managers.System.service - Phoshpor System Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.managers.System.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-02-16 05:40:42 UTC; 14min ago
 Main PID: 820 (python)
   CGroup: /system.slice/org.openbmc.managers.System.service
           └─820 python /usr/sbin/system_manager.py

Feb 16 05:40:21 witherspoon systemd[1]: Starting Phoshpor System Manager...
Feb 16 05:40:41 witherspoon system_manager.py[820]: Running System State: BASE_APPS
Feb 16 05:40:41 witherspoon system_manager.py[820]: SystemManager Init Done
Feb 16 05:40:41 witherspoon system_manager.py[820]: Running SystemManager
Feb 16 05:40:42 witherspoon systemd[1]: Started Phoshpor System Manager.
Feb 16 05:40:55 witherspoon system_manager.py[820]: SystemManager Goto System State: BMC_STARTING
Feb 16 05:40:55 witherspoon system_manager.py[820]: Running System State: BMC_STARTING
Feb 16 05:40:55 witherspoon system_manager.py[820]: SystemManager Goto System State: BMC_READY
Feb 16 05:40:55 witherspoon system_manager.py[820]: Running System State: BMC_READY
root@witherspoon:~#
```

- [x] Journal log  
[journald_v1.99.2-60-gcb1cc2b_20170216042029_GPIO.txt](https://github.com/openbmc/openbmc/files/779186/journald_v1.99.2-60-gcb1cc2b_20170216042029_GPIO.txt)

vishwabmc, 2017-02-16:
I looked at this one and saw that there was a racedition between POWR using it and the System Manager getting ready. Need to see if POWR has the sync points and if it does, then some change needs to go into system manager to delay broadcasting the availability

vishwabmc, 2017-02-16:
VISHWANATHA SUBBANNA [11:26 AM] 
looks like a race condition

[11:26]  
`Feb 16 05:40:42 witherspoon systemd[1]: Started Phoshpor System Manager.`

```Feb 16 05:40:40 witherspoon power_control.exe[846]: Power GPIO: call to getGpioConfiguration failed: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.managers.System was not provided by any .service files```

vishwabmc, 2017-02-16:
https://github.com/openbmc/openbmc/blob/171031d20c7ed03900739d51ba53ad0001f98fa5/meta-phosphor/common/recipes-phosphor/chassis/obmc-op-control-power/org.openbmc.control.Power%40.service

I saw 
```
[Unit]
Description=Phosphor Power%i Control
Wants=mapper-wait@-org-openbmc-managers-System.service
After=mapper-wait@-org-openbmc-managers-System.service```

So the change should be with managers.Systems

vishwabmc, 2017-02-16:
@geissonator Can you assign this to someone ?

rfrandse, 2017-06-16:
Andrew, need to prioritize this defect. It's being dragged from sprint to sprint. Should this be put into the backlog.

geissonator, 2017-06-16:
This was 1.99.2 code, @gkeishin you seen this one lately?

gkeishin, 2017-06-16:
I have seen this lately. So we can close perhaps. If I see it again we can re-open it..

closed 2017-06-22 by gkeishin