openbmc/openbmc #1144: Barreleye system unpingable after BMC Reboot via REST

rahulmah, 2017-02-16:
Here is one instance where we hit with system not pingable after BMC Reboot. Seen this issue with latest BMC fw. Not sure if this is a consistent problem.

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.BMC.Transition.Reboot\"}" -X PUT  https://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0/attr/RequestedBMCTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

Serial console logs: 
[Serial console for Barreleye system unpingable.txt](https://github.com/openbmc/openbmc/files/779316/Serial.console.for.Barreleye.system.unpingable.txt)

BMC Fw: 
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@barreleye:~#
root@barreleye:~# cat /etc/version
20170216043607
root@barreleye:~#
```

vishwabmc, 2017-02-16:
I looked the console output and here is what is interesting.
```[  OK  ] Stopped Phosphor Version Software Manager.
         Starting Phosphor Version Software Manager...
[FAILED] Failed to start Phosphor Version Software Manager.
See 'systemctl status xyz.openbmc_proje...oftware.Version.service' for details.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login:
Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: root
0Password:

Login incorrect
barreleye login:
Login timed out after 60 seconds.

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login:          Stopping Phosphor Webserver...```

@bradbishop I wonder if systemd triggers a 'reboot' post certain attempts to Restart a service ?

vishwabmc, 2017-02-16:
@rahulmah Did the BMC boot post this and just that you could not do a network ping ?. 

if so, what was the ifconfig output from the console ?

gkeishin, 2017-02-16:
I think this is the same one as opened earlier for BMC reboot where system become unpingable
https://github.com/openbmc/openbmc/issues/1070

rahulmah, 2017-02-16:
@vishwabmc : System did not boot after this. It went to unpingable state after reboot and then was never pingable. I even tried to reboot system using PDU but still it was not pinging.

@gkeishin : In your defect system was pingable. Problem was with only SSH login. So it is a different case.

gwshan, 2017-03-01:
@rahulmah, could you please catch me from sametime (gwshan@au1.ibm.com) when you get a chance? I want to check several things with you: the issue is reproducible, is there packet flooding from "ifconfig eth0" and the states of GPIO PINs related to reset etc.

From the serial console output, the NCSI channel was bouncing for unknown reason. One thing for sure is the responses for NCSI GLS commands are all missed.

ratagupt, 2017-03-21:
Machine:- Barreleye

random: nonblocking pool is initialized

Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 barreleye ttyS4

barreleye login: occ-i2c 3-0051: ERROR: OCC Poll: 0xe1
ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface up

Seems NCSI interface was going down which caused the network to go down.

rahulmah, 2017-03-21:
@ratagupt: As discussed, I am hitting with a similar issue after BMC reboot. After BMC reboot, system is pingable but SSH and REST Login both are not working. Please find attached serial console log of the failure. 

[Serial console log Barreleye failure.txt](https://github.com/openbmc/openbmc/files/858030/Serial.console.log.Barreleye.failure.txt)

@gwshan : We are hitting  this issue inconsistently.  Serial console hangs after the failure so we cant debug more. Attached serial log are the only logs which I have.

sivassrr, 2017-03-21:
if it is pingable and no SSH / REST then it must be same as https://github.com/openbmc/openbmc/issues/1161

sivassrr, 2017-03-21:
As discussed this morning in local squad meeting, dmesg, journalctl (if reboot not happened ).

geissonator, 2017-03-30:
Dup of #1161

closed 2017-03-30 by geissonator