openbmc/openbmc #1145: Host IPMI errors [Witherspoon w33 v1.99.2-60-gcb1cc2b ]

gkeishin, 2017-02-16:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170216042029
root@witherspoon:~# uptime
 07:18:54 up 39 min,  load average: 0.08, 0.06, 0.13
root@witherspoon:~#
```

```
Feb 16 06:49:17 witherspoon ipmid[1176]: Failed to get /org/openbmc/settings/host0 busname: Device or resource busy
Feb 16 06:49:17 witherspoon ipmid[1176]: ERROR:[0xD4] handling NetFn:[0xA], Cmd:[0x12]
Feb 16 06:49:17 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:17 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:21 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy
Feb 16 06:49:21 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:21 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:26 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy
Feb 16 06:49:26 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:26 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:30 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy
Feb 16 06:49:30 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:30 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:34 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy

```

```
root@witherspoon:~# systemctl status org.openbmc.HostServices.service | cat
● org.openbmc.HostServices.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/org.openbmc.HostServices.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-02-16 06:49:13 UTC; 28min ago
 Main PID: 1176 (ipmid)
   CGroup: /system.slice/org.openbmc.HostServices.service
           └─1176 /usr/sbin/ipmid

Feb 16 06:49:34 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:34 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:38 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy
Feb 16 06:49:38 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:38 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:42 witherspoon ipmid[1176]: Failed to get /org/openbmc/watchdog/host0 bus name: Device or resource busy
Feb 16 06:49:42 witherspoon ipmid[1176]: Failed to call the method: Input/output error
Feb 16 06:49:42 witherspoon ipmid[1176]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 16 06:49:57 witherspoon ipmid[1176]: Failed to get  busname: No such file or directory
Feb 16 06:50:09 witherspoon ipmid[1176]: ERROR:[0xD4] handling NetFn:[0x6], Cmd:[0x6]
```

-[x]  Journal log  
[journald_v1.99.2-60-gcb1cc2b_20170216042029_Host_IPMI.txt](https://github.com/openbmc/openbmc/files/779353/journald_v1.99.2-60-gcb1cc2b_20170216042029_Host_IPMI.txt)

gkeishin, 2017-02-16:
@tomjoseph83 is taking a look at the system.. Awaiting more inputs from him.

geissonator, 2017-02-19:
Updating title to track general issue of services getting this "Device or resource busy".  I hit it during boot and have seen it elsewhere intermittently.

```
Feb 02 17:56:30 witherspoon mapper[1546]: Error finding '/org/openbmc/watchdog/host0' service: Device or resource busy
Feb 02 17:56:30 witherspoon systemd[1]: obmc-start-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 17:56:30 witherspoon systemd[1]: Failed to start Start Watchdog0.
Feb 02 17:56:30 witherspoon systemd[1]: Dependency failed for Chassis0 (Start).
Feb 02 17:56:30 witherspoon systemd[1]: obmc-chassis-start@0.target: Job obmc-chassis-start@0.target/start failed with result 'dependency'.
Feb 02 17:56:30 witherspoon systemd[1]: obmc-chassis-start@0.target: Triggering OnFailure= dependencies.
Feb 02 17:56:30 witherspoon systemd[1]: obmc-start-watchdog@0.service: Unit entered failed state.
Feb 02 17:56:30 witherspoon systemd[1]: obmc-start-watchdog@0.service: Failed with result 'exit-code'.
Feb 02 17:56:31 witherspoon systemd[1]: Reached target Quiesce Target.
```

closed 2017-02-24 by williamspatrick