openbmc/openbmc #1148: phosphor-host-ipmid fails to build in CI

geissonator, 2017-02-16:
We're enabling CI for all repos now, phosphor-host-ipmid hits this one.  Usually indicates the configure.ac is not setup correctly to show dependencies.  @msbarth is the best contact.
```
12:52:49 checking for dirname... dirname
12:52:49 checking for sdbus++... no
12:52:49 configure: error: Cannot find sdbus++
12:52:49 configure: error: ./configure failed for softoff
12:52:49 Traceback (most recent call last):
12:52:49   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 178, in <module>
12:52:49     dep_installed)
12:52:49   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 123, in build_depends
12:52:49     check_call_cmd(pkgdir, './configure', conf_flags)
12:52:49   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 28, in check_call_cmd
12:52:49     check_call(cmd)
12:52:49   File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
12:52:49     raise CalledProcessError(retcode, cmd)
12:52:49 subprocess.CalledProcessError: Command '('./configure', '')' returned non-zero exit status 1
12:52:50 Build step 'Execute shell' marked build as failure
12:52:50 Set build name.
12:52:50 New build name is '#310-Vishwanatha Subbanna'
12:52:50 Variable with name 'BUILD_DISPLAY_NAME' already exists, current value: '#310-Vishwanatha Subbanna', new value: '#310-Vishwanatha Subbanna'
```

https://openpower.xyz/job/openbmc-repository-ci/314/console

msbarth, 2017-02-16:
This problem is due to phosphor-host-ipmid having an "embedded" configure.ac now within the softoff/ subdirectory. The unit test script is not setup to look for additional configure.ac files under subdirs of a package and would need to be enhanced to support this.

vishwabmc, 2017-02-17:
I am working on this issue

vishwabmc, 2017-02-17:
Fixed with : https://gerrit.openbmc-project.xyz/#/c/1984/

closed 2017-02-24 by geissonator