openbmc/openbmc #1149: ipmi-fru-parser fails to build in CI

geissonator, 2017-02-16:
We're enabling CI on all repos, ipmi-fru-parser fails with the below error.  @msbarth is the best contact.

```
12:58:17 /var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser > make
12:58:17   GEN      fru-gen.cpp
12:58:18 make  all-am
12:58:18 make[1]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser'
12:58:18   CXX      strgfnhandler.lo
12:58:18   CXX      libwritefrudata_la-frup.lo
12:58:19   CXX      libwritefrudata_la-fru-gen.lo
12:58:21   CXX      libwritefrudata_la-writefrudata.lo
12:58:21 writefrudata.cpp:18:32: fatal error: sdbusplus/server.hpp: No such file or directory
12:58:21 compilation terminated.
12:58:21 Makefile:648: recipe for target 'libwritefrudata_la-writefrudata.lo' failed
12:58:21 make[1]: *** [libwritefrudata_la-writefrudata.lo] Error 1
12:58:21 make[1]: Leaving directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser'
12:58:21 Makefile:414: recipe for target 'all' failed
12:58:21 make: *** [all] Error 2
12:58:21 Traceback (most recent call last):
12:58:21   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 178, in <module>
12:58:21     dep_installed)
12:58:21   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 124, in build_depends
12:58:21     check_call_cmd(pkgdir, 'make')
12:58:21   File "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/unit-test.py", line 28, in check_call_cmd
12:58:21     check_call(cmd)
12:58:21   File "/usr/lib/python2.7/subprocess.py", line 541, in check_call
12:58:21     raise CalledProcessError(retcode, cmd)
12:58:21 subprocess.CalledProcessError: Command '('make',)' returned non-zero exit status 2
12:58:22 Build step 'Execute shell' marked build as failure
12:58:22 Set build name.
12:58:22 New build name is '#316-Ratan Gupta'
```
https://openpower.xyz/job/openbmc-repository-ci/316/console

dkodihal, 2017-02-16:
This should fix the problem : https://gerrit.openbmc-project.xyz/#/c/2481/

dkodihal, 2017-02-16:
And this : https://gerrit.openbmc-project.xyz/#/c/2482/

ratagupt, 2017-02-17:
After having these two..Still build got failed.Now unable to find the libwritefrudata.

02:34:34 make[1]: Leaving directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser'
02:34:34 /var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser > make install
02:34:34 make  install-am
02:34:34 make[1]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser'
02:34:34 make[2]: Entering directory '/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser'
02:34:34  /bin/mkdir -p '/usr/local/sbin'
02:34:34   /bin/bash ./libtool   --mode=install /usr/bin/install -c phosphor-read-eeprom '/usr/local/sbin'
02:34:34 libtool: warning(B: 'libwritefrudata.la' has not been installed in '/usr/local/lib'(B
02:34:34 libtool: install: /usr/bin/install -c .libs/phosphor-read-eeprom /usr/local/sbin/phosphor-read-eeprom
02:34:34  /bin/mkdir -p '/usr/local/lib/ipmid-providers'
02:34:34  /bin/bash ./libtool   --mode=install /usr/bin/install -c   libstrgfnhandler.la '/usr/local/lib/ipmid-providers'
02:34:34 libtool: warning(B: relinking 'libstrgfnhandler.la'(B
02:34:34 libtool: install: (cd /var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser; /bin/bash "/var/lib/jenkins-slave/workspace/openbmc-repository-ci/openbmc/ipmi-fru-parser/libtool"  --silent --tag CXX --mode=relink g++ -std=c++14 -g -O2 -flto -fPIC -Wall -Werror -version-info 1:0:0 -shared -o libstrgfnhandler.la -rpath /usr/local/lib/ipmid-providers strgfnhandler.lo libwritefrudata.la -lmapper )
02:34:34 /usr/bin/ld: cannot find -lwritefrudata
02:34:35 collect2: error: ld returned 1 exit status
02:34:35 libtool:   error(B: error: relink 'libstrgfnhandler.la' with the above command before installing it(B
02:34:35 Makefile:468: recipe for target 'install-libstrgfnhandlerLTLIBRARIES' failed
02:34:35 make[2]: *** [install-libstrgfnhandlerLTLIBRARIES] Error 1

ratagupt, 2017-02-17:
Seems the above error is getting fixed through .
https://gerrit.openbmc-project.xyz/#/c/2486/1.

closed 2017-02-24 by geissonator