openbmc/openbmc #1150: boot_flags : Setup does not stop at petitboot

trajeswaran, 2017-02-16:
When the boot_flags is set to "Setup", then Host power on is expected to stop at petitboot.  Actually, it proceeds to loading Host OS. The behavior for "default" and "Setup" is same. 

Please refer the issue #1135 "boot_flags does not support the value "BIOS" to stop at petit boot" for further details.

I tried it on w37. Please refer below for the driver details.

cat /etc/version 
20170215223009

cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-55"
VERSION_ID="v1.99.2-55-g8646a1b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-55"
BUILD_ID="v1.99.2"

tomjoseph83, 2017-02-17:
Investigated this issue on w30. I took a look at the detailed trace for host-ipmid and figured out that there is no request to read the Boot Flags (Get System Boot Options). Can we get it checked from the Petitboot side that there is a IPMI command being sent? @geissonator Can you tag the appropriate person?

gkeishin, 2017-02-23:
```
Ubuntu Zesty Zapus (development branch) w37l ttyS0

w37l login: ubuntu
Password:
Last login: Thu Feb 23 03:21:41 CST 2017 on ttyS0
Welcome to Ubuntu Zesty Zapus (development branch) (GNU/Linux 4.10.0-rc8-gc1920f1 ppc64le)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
ubuntu@w37l:~$ sudo nvram --update-config,auto-boot?=false
[sudo] password for ubuntu:
ubuntu@w37l:~$

or

From petitboot

 Petitboot System Configuration
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq

  Autoboot:      (*) Disabled
                 ( ) Enabled

  Network:       (*) DHCP on all active interfaces
                 ( ) DHCP on a specific interface
                 ( ) Static IP configuration

  DNS Server(s):                                   (eg. 192.168.0.2)
                 (if not provided by DHCP server)
  HTTP Proxy:
  HTTPS Proxy:

  Disk R/W       ( ) Prevent all writes to disk
                 (*) Allow bootloader scripts to modify disks

  Boot console:  (*) /dev/hvc0 [IPMI / Serial]

 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
 tab=next, shift+tab=previous, x=exit, h=help

 

```

```
Power off  --> Power On --> Should stop at Petitboot
```

```
To re-enable 
From Petitboot

 Petitboot System Configuration
 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq

  Autoboot:      ( ) Disabled
                 (*) Enabled

  Network:       (*) DHCP on all active interfaces
                 ( ) DHCP on a specific interface
                 ( ) Static IP configuration

  DNS Server(s):                                   (eg. 192.168.0.2)
                 (if not provided by DHCP server)
  HTTP Proxy:
  HTTPS Proxy:

  Disk R/W       ( ) Prevent all writes to disk
                 (*) Allow bootloader scripts to modify disks

  Boot console:  (*) /dev/hvc0 [IPMI / Serial]

 qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq
 tab=next, shift+tab=previous, x=exit, h=help

```

tomjoseph83, 2017-02-24:
To see if Petitboot is trying to retrieve the boot flags please
```
    Set nvram --update-config "petitboot,debug?=true"
    Reboot to Petitboot
    Run pb-sos and save the resulting tar file
```

@gkeishin .. Can you fetch this info for  open-power/skiboot#57

gkeishin, 2017-02-24:
Updated here already  @tomjoseph83  https://github.com/open-power/skiboot/issues/57

tomjoseph83, 2017-05-08:
@trajeswaran Does the Setup stop at Petit Boot? If it is resolved, please close it.

tomjoseph83, 2017-05-11:
@williamspatrick We can have this closed. @trajeswaran Confirmed that the skiboot has fixed this and verified it.

closed 2017-10-17 by geissonator