openbmc/openbmc #1154: Kernel aspeed-g5-pinctrl Warning [ Witherspoon ]

gkeishin, 2017-02-16:
**Driver Info**
```
root@w38:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"
root@w38:~# cat /etc/version
20170216042029
root@w38:~# uptime
 19:12:29 up 17 min,  load average: 0.03, 0.16, 0.33
root@w38:~# obmcutil state
 = BMC_READY
root@w38:~#

```

**Log messages** 

```
Jan 31 18:55:30 w38 kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: No GFX phandle found, some mux configurations may fail
Jan 31 18:55:30 w38 kernel: aspeed-g5-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: No LHC phandle found, some mux configurations may fail

```

- [x] Journal log 
[journald_v1.99.2-60-gcb1cc2b_kernel_PHANDLE.txt](https://github.com/openbmc/openbmc/files/780789/journald_v1.99.2-60-gcb1cc2b_kernel_PHANDLE.txt)

gkeishin, 2017-02-16:
@amboar This doesn't break bones yet but yeah just a concern.. 
Looks like a warning though https://patchwork.kernel.org/patch/9481231/
```
 static int aspeed_g5_pinctrl_probe(struct platform_device *pdev)
 {

+    node = of_parse_phandle(pdev->dev.of_node, "aspeed,external-nodes", 0);
+    map = syscon_node_to_regmap(node);
+    of_node_put(node);
+    if (IS_ERR(map)) {
+        dev_warn(&pdev->dev, "No GFX phandle found, some mux configurations may fail\n");
+        map = NULL;
+    }
```

amboar, 2017-02-16:
This is fixed by http://patchwork.ozlabs.org/patch/727641/ - we are waiting for Joel to apply the series and bump the kernel

gkeishin, 2017-02-17:
Thanks @amboar will close this once the changes are in..

amboar, 2017-02-22:
@gkeishin this issue is resolved: https://gerrit.openbmc-project.xyz/#/c/2574/

gkeishin, 2017-02-23:
Thanks @amboar

closed 2017-02-23 by gkeishin