openbmc/openbmc #1155: phosphor-logging: Create a friendlier "create and commit" error interface

anoo1, 2017-02-16:
Currently the error log flow requires to create an elog (error log) that throws an exception, which the caller needs to catch in order to commit the error. Per Patrick, "We should make an interface to combine this (or maybe not even throw the exception)."

vishwabmc, 2017-02-17:
@geissonator Please tag enhancement

geissonator, 2017-03-23:
Need task list

dkodihal, 2017-03-24:
The task list Devender was proposing is below. I also added a task related to another enhancement, will discuss if that can be covered as well

- Create method in elog.hpp to create and commit an error in one shot. This should not throw an exception, but internally just log to the journal and call commit.
- Add test cases for the new interface.

Nice to have -
- Change existing commit interface to work with exception type rather than exception name, i.e commit(e.what()) -> commit<e>().
- Compile time assert that errors that are being created are of type sdbusplus exception.

anoo1, 2017-03-24:
👍 

For the nice to have, yeah we want to change the commit to take the exception, like commit(xyz::openbmc_project::example::example_exception);

williamspatrick, 2017-03-24:
`commit<xyz::openbmc_project::example::example_exception>()`?  I think the first "nice to have" is a requirement.  Might as well add them all since the last one should be trivial.

williamspatrick, 2017-04-13:
All changes merged for this.

closed 2017-04-13 by williamspatrick