openbmc/openbmc #1156: eSEL logging error

e-liner, 2017-02-16:
I was trying to test logging a HBRT error log on a barreleye machine today and I saw a weird error.  Has this been seen before?

cat /etc/os-release:
root@barreleye:/tmp# cat /etc/os-release
ID=openbmc-phosphor
NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)
VERSION=0.1.0 (jethro)
VERSION_ID=v1.0.5-5-g6044148-dirty
PRETTY_NAME=Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0 (jethro)
BUILD_ID=v1.0.5
root@barreleye:/tmp# 

OpenBmc traces:
root@barreleye:/tmp# IPMI Incoming: Seq 0x27, NetFn 0x04, CMD: 0x2f 
0x000000: 35                                              5               
IPMI GET_SENSOR_TYPE [0x35]
IPMI Response:
0x000000: 00 12 6f                                        ..o             
Failed to call the method: Input/output error
Dest: :1.1, Path: /org/openbmc/HostIpmi/1
IPMI Incoming: Seq 0x28, NetFn 0x0a, CMD: 0x42 
IPMI Handling RESERVE-SEL 0x0007
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x29, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 00 00 00 00 00 00 00 df 00 00 00 00 20 00 .............. .
0x000010: 04 12 35 6f aa 00 00                            ..5o...         
IPMI PARTIAL ESEL for /tmp/esel  Offset = 0 Length = 16
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x2a, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 07 00 10 00 00 50 48 00 30 01 00 1a 00 00 .......PH.0.....
0x000010: 00 00 2a 47 2a 54 9f 00 00 00 2a 47 2b 8c 77 42 ..*G*T....*G+.wB
0x000020: 00 00 05 00 00 00 00 00 00 00 00 00 00 00 00 89 ................
0x000030: 00 00 02 89 00 00 02 55 48 00 18 01             .......UH...    
IPMI PARTIAL ESEL for /tmp/esel  Offset = 16 Length = 53
IPMI Response:
0x000000: 00 07 00  
IPMI Incoming: Seq 0x2b, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 07 00 45 00 00 00 1a 00 8a 03 40 00 00 00 ....E.......@...
0x000010: 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 ..........PS.P..
0x000020: 00 00 02 00 00 09 1a 0b 00 48 00 00 00 e0 00 00 .........H......
0x000030: 14 00 00 00 00 00 00 20 00 00 00 00             ....... ....    
IPMI PARTIAL ESEL for /tmp/esel  Offset = 69 Length = 53
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x2c, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 07 00 7a 00 00 00 00 00 00 00 00 00 00 00 ....z...........
0x000010: 00 00 00 00 00 42 43 38 41 31 41 30 42 20 20 20 .....BC8A1A0B   
0x000020: 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20                 
0x000030: 20 20 20 20 20 55 44 00 1c 01 06 01                  UD.....    
IPMI PARTIAL ESEL for /tmp/esel  Offset = 122 Length = 53
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x2d, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 07 00 af 00 00 00 02 a8 00 02 00 00 00 01 ................
0x000010: 00 00 00 55 06 11 72 e4 00 00 00 00 55 44 00 40 ...U..r.....UD.@
0x000020: 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 ....Hostboot Bui
0x000030: 6c 64 20 49 44 3a 20 68 6f 73 74 62             ld ID: hostb    
IPMI PARTIAL ESEL for /tmp/esel  Offset = 175 Length = 53
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x2e, NetFn 0x32, CMD: 0xf0 
0x000000: 07 00 07 00 e4 00 01 6f 6f 74 2d 31 66 36 37 38 .......oot-1f678
0x000010: 34 64 2d 34 36 39 38 61 39 39 2f 68 62 69 72 74 4d-4698a99/hbirt
0x000020: 2e 62 69 6e 00 00 00                            .bin...    
IPMI PARTIAL ESEL for /tmp/esel  Offset = 228 Length = 32
IPMI Response:
0x000000: 00 07 00                                        ...             
IPMI Incoming: Seq 0x2f, NetFn 0x0a, CMD: 0x44 
[... 76 lines cut ...]
IPMI:I>partial_add_esel:
IPMI:~[0x0000] 07000700 af000000 02a80002 00000001    *................*
IPMI:~[0x0010] 00000055 061172e4 00000000 55440040    *...U..r.....UD.@*
IPMI:~[0x0020] 01010100 486f7374 626f6f74 20427569    *....Hostboot Bui*
IPMI:~[0x0030] 6c642049 443a2068 6f737462             *ld ID: hostb    *
IPMI:rt: calling sync 32:f0  len=60
IPMI:I>partial_add_esel:
IPMI:~[0x0000] 07000700 e400016f 6f742d31 66363738    *.......oot-1f678*
IPMI:~[0x0010] 34642d34 36393861 39392f68 62697274    *4d-4698a99/hbirt*
IPMI:~[0x0020] 2e62696e 000000                        *.bin...         *
IPMI:rt: calling sync 32:f0  len=39
IPMI:I>add_sel:
IPMI:~[0x0000] 0000de00 00000000 00005500 00000007    *..........U.....*
IPMI:rt: calling sync a:44  len=16
IPMI:sel: <<send_esel o_err=00000000, o_cc=x00, sel recID=x70, esel recID=x07
IPMI:sel: I>Doing ack for eid 0x89000002
ERRL:>>ackErrLogInPnor(89000002)
ERRL:<<ackErrLogInPnor returning false
IPMI:sel: <<process_esel
IPMI:sel: <<sendESEL
ERRL:I>sendErrLogToBmc callout size 1
ERRL:<<sendErrLogToBmc
ERRL:<<sendToHypervisor()
UTIL:cmd_logerror> committed error log.
(E): run_command returned 0 but outString is NULL

vishwabmc, 2017-02-20:
@e-liner Please help provide the event data ?. Need to see what that event contains.

vishwabmc, 2017-02-20:
@tomjoseph83 I assume you were working on some eSEL related issue ?. Could you help see this one ?

williamspatrick, 2017-02-20:
This is a v1.0 issue.

gtmills, 2018-06-29:
Closing. No activity in year+.

gtmills, 2018-06-29:
Please reopen if still a problem. Barreleye is also no longer a supported machine. https://github.com/openbmc/openbmc/blob/master/meta-openbmc-machines/meta-openpower/meta-rackspace/meta-barreleye/README.md#power8-barreleye

closed 2018-06-29 by gtmills