openbmc/openbmc #1157: i2c_aspeed i2c-11: Invalid state [witherspoon w38 - v1.99.2-66-g1488f75_20170217003621]

gkeishin, 2017-02-17:
**On rebooting BMC we are seeing this logs on the console in recently few builds..**

```
i2c_aspeed i2c-11: Invalid state (msg   (null), pending 0)?
i2c i2c-11: i2c_ast: timeout waiting for bus free
i2c i2c-11: i2c_ast: timeout waiting for bus free
i2c_aspeed i2c-11: Invalid state (msg   (null), pending 0)?
i2c_aspeed i2c-11: Invalid state (msg   (null), pending 0)?
i2c_aspeed i2c-11: Invalid state (msg   (null), pending 0)?
i2c i2c-11: i2c_ast: timeout waiting for bus free
```

gkeishin, 2017-02-17:
Had a word with @shenki  and he pointed out probably one of  the bellow three devices on w38 causing it.. but not clear on the root cause..

```
        rtc@32 {
                compatible = "epson,rx8900";
                reg = <0x32>;
        };

        eeprom@51 {
                compatible = "atmel,24c64";
                reg = <0x51>;
        };

        ucd90160@64 {
                compatible = "ti,ucd90160";
                reg = <0x64>;
        };
```

gkeishin, 2017-02-17:
@amboar  took a look at the system to probe.. and we could replicate the problem but latter once AC cycled , its seems to have gone away.. 

Any more info on this ?

geissonator, 2017-02-28:
Power cycles fix all :)  Please close out for now @gkeishin .  Can always reopen if you can figure out how to recreate.

gkeishin, 2017-03-01:
Sure... Closing..

closed 2017-03-01 by gkeishin