openbmc/openbmc #1159: Inband power cycle IPMI command fails

spinler, 2017-02-17:
When hostboot does an SBE update,  it sends down a power cycle IPMI command (netfn 0x00 cmd 0x02 subcommand 0x02)  so they can be gracefully restarted.  However, host-ipmid doesn't support that command.  Eventually, hostboot just waits for the watchdog to expire and the reboot does occur.  That causes a big increase in IPL time.

```
	switch(chassis_ctrl_cmd)
	{
		case CMD_POWER_OFF:    <-- 0x00
			rc = ipmi_chassis_power_control("powerOff");
			break;
		case CMD_HARD_RESET:   <-- 0x03
			rc = ipmi_chassis_power_control("reboot");
			break;
		default:
		{
			fprintf(stderr, "Invalid Chassis Control command:[0x%X] received\n",chassis_ctrl_cmd);
			rc = -1;
		}
	}
```

vishwabmc, 2017-02-20:
@spinler So we just happened to not notice this until so far ?. Is it anything specific to witherspoon / the new SBE ?

vishwabmc, 2017-02-20:
@spinler I believe support for this may have been intentionally dropped ( guess tho ) since its listed as optional command to be supported.

`2h = power cycle (optional). This command provides a power off interval`

spinler, 2017-02-20:
@vishwabmc Barreleye is a little bit of a mystery, but it is suspected that nobody in hostboot  ever looked at the traces to see that what was triggering the reboot was the watchdog expiring and not the soft power off command.

geissonator, 2017-02-28:
With the ability for mfg and test to disable automated reboots on errors (like checkstop and watchdog), we can't really count on those paths to handle this situation.  Bumps up the priority a bit on getting this in.

vishwabmc, 2017-03-01:
I looked at IPMI spec and it says : 

```2h = power cycle (optional). This command provides a power off interval
of at least 1 second following the deassertion of the system’s
POWERGOOD status from the main power subsystem. It is
recommended that no action occur if system power is off (S4/S5)
when this action is selected, and that a D5h “Request parameter(s)
not supported in present state.” error completion code be returned.
Note that some implementations may cause a system power up if
a power cycle operation is selected when system power is down.
For consistency of operation, it is recommended that system
management software first check the system power state before
issuing a power cycle, and only issue the command if system
power is ON or in a lower sleep state than S4/S5
```

vishwabmc, 2017-03-01:
I looked at the current implementation of 'reboot' command in chassis_control.py and it does exactly what is mentioned in 
```
Note that some implementations may cause a system power up if
a power cycle operation is selected when system power is down.
```
And if the system is powered on, then it does a Power Off, followed by Power On

https://github.com/openbmc/skeleton/blob/master/pychassisctl/chassis_control.py

vishwabmc, 2017-03-01:
I can solve it in 2 ways.

1/. Handle the POWER_CYCLE command and call back into doing 'reboot' --> quick hack
2/. Modify chassis_control.py to define a new PowerCycle API and do what the spec wants. --> kind of quick too.

I will have both the changes done and push for revire.

geissonator, 2017-03-01:
Thanks Vishwa, we still need to move the ipmi code over to use the new host state functions (i.e. call the Reboot) interface.  So my vote is the quick fix for now.

closed 2017-03-08 by williamspatrick