openbmc/openbmc #1160: phosphor-rest-server: Set property with UInt64 type not supported

mine260309, 2017-02-20:
We can update a property's value by PUT method, e.g.
```
curl -b cjar -k -H "Content-Type: application/json" -X PUT
    -d '{"data": <value>}' \
    https://bmc/org/openbmc/control/flash/bios/attr/flasher_path
```

However, the current REST server does NOT support properties with UInt64 type. E.g. new `phosphor-time-manager` implements `EpochTime.interface` has a `Elapsed` property with `uint64` type.

We can get the property well via REST API, but not for set:
```
$ curl -b cjar -k https://9.3.23.76/xyz/openbmc_project/Time/Host
{
  "data": {
    "Elapsed": 1487576065512412
  },
  "message": "200 OK",
  "status": "ok"
}

$ curl -b cjar -k -H "Content-Type: application/json" -X PUT -d '{"data": {"Elapsed": 1487304700000000000}}' https://9.3.23.76/xyz/openbmc_project/Time/Host
{
  "data": {
    "description": "org.freedesktop.DBus.Error.InvalidArgs: Incorrect parameters for property 'Elapsed', expected 't', got 'x'."
  },
  "message": "403 Forbidden",
  "status": "error"
}
```

The DBus property expect type `'t'(UInt64)`, while REST server gives it as `'x'(Int64)`.

The problem may lie in `phosphor-rest-server/module/obmc/wsgi/apps/rest_dbus.py`:
```
    def do_put(self, path, prop, value=None):
        ...
        try:
            properties_iface.Set(iface, prop, value) ### type(value) is either int or long in python
        ...
        except dbus.exceptions.DBusException, e:
            if e.get_dbus_name() == DBUS_INVALID_ARGS:
                abort(403, str(e))
            raise
```
The type of `value` is either int or long in python, depending on the actual value.

For properties with `uint64`, the call will always fail because it never has a chance to pass it as `UInt64`.

I tried to modify the code:
```
            import dbus.types as types
            v = types.UInt64(long(value))
            properties_iface.Set(iface, prop, v)
```
Then it works fine.

The issue may be valid for uint32 as well, since the value is always passed as signed integers.

I think `phosphor-rest-server` shall add such support for UInt64 types, the problem is, how?

Maybe in exception, check the property's type first and then try to pass the value with expected type?

mine260309, 2017-02-21:
@bradbishop I would like to fix this issue if you feel OK.
Current I'm thinking to add a couple of code in the exception handling:
```
        try:
            properties_iface.Set(iface, prop, value) ### type(value) is either int or long in python
        ...
        except dbus.exceptions.DBusException, e:
            if e.get_dbus_name() == DBUS_INVALID_ARGS:
                # TODO: check the property's type, and try to convert
                # the type to the expected one, and call `Set()` again
                # if still fails, abort with 403
                abort(403, str(e))
            raise
```
1. Do you think it is feasible to get a property's type in `rest_dbus.py`?
2. If not, we can get the type info by parsing the string of the exception.

Or if you have any better solution please let me know.

bradbishop, 2017-02-23:
Can you introspect the object prior to calling Set to get the type?

mine260309, 2017-02-24:
Yes, we can call introspect before `Set()`, but it introduces an additional IPC and it's not efficient.

In most cases the Set() should just work fine, so it's better not to add additional calls on normal path.

My idea is only call introspect (or parse the exception string) on exception occurs.

```
    Call Set();
    if it's OK, return;
    if exception occurs:
        if it's DBUS_INVALID_ARGS:
            Call introspect() or parse the exception string, to get the type
            try convert the value to the specific type;
            if conversion fails:
                raise exception
            call Set() again with the converted value
            if it's OK, return;
            if exception occurs:
                if it's DBUS_INVALID_ARGS:
                    abort(403) 
                else:
                    raise exception
        else:
            raise exception
        fi
```

dkodihal, 2017-03-28:
Another use-case in favor of doing object introspection is that I now can't pass in array[byte] or even byte. The byte gets interpreted as int32 and the d-bus call fails due to a parameter mismatch.

lgon, 2017-04-07:
**# Tasklist**
- [ ] Add method to phosphor-rest server that uses introspection to get type.
- [ ] Modify phosphor-rest-server Set, so on exception, uses introspection method to get type, and for UInt64, convert value, and set again.

dkodihal, 2017-04-07:
FYI, this is not just about uint64, but in general the rest server should be able to correctly figure out the types d-bus expects via introspection. In other words your first task should cover it all. I mentioned a use-case in the story where I had to send an array of bytes and that wouldn't work because a byte would be interpreted as int32.

I'm not sure if @bradbishop agreed to do this on getting an exception versus always before doing a PUT or POST.

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3743 Converts value to expected type
Fixes #1160 phosphor-rest-server: Set property with UInt64 type not supported

closed 2017-06-05 by williamspatrick