openbmc/openbmc #1161: Witherspoon: W37: Loop Test failed from 2nd iteration and serial console hangs

sivassrr, 2017-02-20:
BMC Driver is from Andrew: witherspoon-20170218231615.all.tar
PNOR from Uma: /witherspoon_20170218n.pnor

First basic 10 iterations went fine. 

Enabled some more test and run power on / power off.

First iteration went fine and 2nd iteration on wards timeout and all remaining test failed. 

Error Details as follows:
```
ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /login (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f01817bf210>: Failed to establish a new connection: [Errno 111] Connection refused',)): FAIL != PASS Also teardown failed: Several failures occurred: 1) NoValidConnectionsError: [Errno None] Unable to connect to port 22 on DUMMYIP 2) Connection not open 3) Connection not open 4) Connection not open 5) Connection not open 6) Connection not open 7) NoValidConnectionsError: [Errno None] Unable to connect to port 22 on DUMMYIP 8) Connection not open [ Message content over the limit has been removed. ] 23) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /org/openbmc/records/events/enumerate (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f018215c750>: Failed to establish a new connection: [Errno 111] Connection refused',)) 24) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /login (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f018211b310>: Failed to establish a new connection: [Errno 111] Connection refused',)): FAIL != PASS 25) Resolving variable '${resp.status_code}' failed: AttributeError: 'unicode' object has no attribute 'status_code' 26) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /org/openbmc/inventory/system/enumerate (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f01820b43d0>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

sivassrr, 2017-02-20:
[MFGStage2BMC1823161PNOR20170218n.pdf](https://github.com/openbmc/openbmc/files/787045/MFGStage2BMC1823161PNOR20170218n.pdf)

vishwabmc, 2017-02-20:
@sivassrr this typically means that there is no server running on that witherspoon listening on port-22. If you have a serial console, what is it telling ?

sivassrr, 2017-02-20:
ping is responding, but no ssh

sivassrr, 2017-02-20:
[serialoutputw37.txt](https://github.com/openbmc/openbmc/files/787061/serialoutputw37.txt)
Here is the screen capture

sivassrr, 2017-02-21:
[dmesgbmchang.txt](https://github.com/openbmc/openbmc/files/789363/dmesgbmchang.txt)
[journalctlbmchangfailure.txt](https://github.com/openbmc/openbmc/files/789361/journalctlbmchangfailure.txt)
[systemctl.txt](https://github.com/openbmc/openbmc/files/789362/systemctl.txt)

All are from another system (Barreleye) with similar failure.  This failed observed in the first iteration itself.

sivassrr, 2017-02-21:
BMC level as follows:
```
root@barreleye:/var/lib/systemd/coredump# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-89"
VERSION_ID="v1.99.2-89-gf327ed7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-89"
BUILD_ID="v1.99.2"
root@barreleye:/var/lib/systemd/coredump# cat /etc/version
20170221021702
```

williamspatrick, 2017-02-23:
The console output looks like it was doing a shutdown?  Did we run into a case where the shutdown hung?

sivassrr, 2017-02-23:
Yes, For both Witherspoon and Barreleye, It was during the warm reset. 

But data is there for Witherspoon, Based on the data available from the combined report the failure started from  the Test Case: Test events after openbmc reboot.
Failure Description as follows:
```
Full Name: 	witherspoonCombinedReport.Tests.Test Eventlog.Test events after openbmc reboot
Documentation: 	
This is to test event can be deleted created prior to openbmc reboot Steps: Create event, Power off if ON else no-op Reboot openbmc, Wait for BMC to READY or Powered OFF state Events should exist post reboot, Create two more events, Delete old and new event

===========================================================

This is to test event can be deleted created prior to openbmc reboot Steps: Create event, Power off if ON else no-op Reboot openbmc, Wait for BMC to READY or Powered OFF state Events should exist post reboot, Create two more events, Delete old and new event
Tags: 	bmcreboot, Event_Logging
Start / End / Elapsed: 	20170220 12:11:10.052 / 20170220 12:12:01.075 / 00:00:51.023
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /xyz/openbmc_project/state/BMC0//attr/RequestedBMCTransition (Caused by ProtocolError('Connection aborted.', error(104, 'Connection reset by peer')))

Also teardown failed:
Several failures occurred:

1) SSHException: Error reading SSH protocol banner[Errno 104] Connection reset by peer

2) SSHException: SSH session not active

3) SSHException: SSH session not active

4) SSHException: SSH session not active

5) SSHException: SSH session not active

6) SSHException: SSH session not active

7) SSHException: Error reading SSH protocol banner[Errno 104] Connection reset by peer

8) SSHException: SSH session not active

    [ Message content over the limit has been removed. ]
23) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /org/openbmc/records/events/enumerate (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f0181765950>: Failed to establish a new connection: [Errno 111] Connection refused',))

24) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /login (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f018210d450>: Failed to establish a new connection: [Errno 111] Connection refused',)): FAIL != PASS

25) Resolving variable '${resp.status_code}' failed: AttributeError: 'unicode' object has no attribute 'status_code'

26) ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /org/openbmc/inventory/system/enumerate (Caused by NewConnectionError('<requests.packages.urllib3.connection.VerifiedHTTPSConnection object at 0x7f0182103b10>: Failed to establish a new connection: [Errno 111] Connection refused',))

```

As there is no SSH, We couldnt collect any FFDC data. If any further data needed,  kindly look the combined pdf report.

sivassrr, 2017-02-23:
Let know what are all the data need to be collected in case if we hit similar failure again. 
So that we can collect and provide useful debug data.

sivassrr, 2017-02-24:
With the latest BMC / PNOR sandbox, Could hit it again. This time failure TC is 
as follows:
```
Full Name: 	witherspoonCombinedReport.Tests.Test Association.Association Unchanged After Reboot
Documentation: 	

**GOOD PATH** This test case is to verify that error log association does not change after open bmc reboot.
Tags: 	Association, Association_Unchanged_After_Reboot, bmcreboot
Start / End / Elapsed: 	20170224 14:24:03.073 / 20170224 14:24:44.126 / 00:00:41.053
Status: 	FAIL (critical)
Message: 	ConnectionError: HTTPSConnectionPool(host='DUMMYIP', port=443): Max retries exceeded with url: /xyz/openbmc_project/state/BMC0//attr/RequestedBMCTransition (Caused by ProtocolError('Connection aborted.', error(104, 'Connection reset by peer')))

Also teardown failed:
Several failures occurred:

1) SSHException: Error reading SSH protocol banner[Errno 104] Connection reset by peer

2) SSHException: SSH session not active
```

sivassrr, 2017-02-24:
Found out openstack were also hitting with this issue some days back. 
Here is the details. 
https://bugs.launchpad.net/grenade/+bug/1349617
Some of the important information from the above link as follows:
```
4. The error “Exception: Error reading SSH protocol banner[Errno 104] Connection reset by peer” implies a corruption of some kind (being that it seems communication wasn’t a problem otherwise, there’s a route) -- this seems consistent with too low of an mtu and data getting truncated “occasionally.” In the log [3], the attempt to connect begins with connection refused (before sshd starts), then changes to authentication failure (likely before the guest has tried to pull the key from the metadata service), then changes to the ssh protocol banner read error. Which sounds like the key was retrieved but it’s corrupted (truncated?).

5. Web search for the same error yielded others having problems with mtu setting in the guest, where they can ping but not ssh with key pair, openstack [4] and cirros [5].

Is it at all possible that there’s an issue with the mtu of the guest sometimes? It would explain the randomness and the protocol banner errors, if data is getting truncated sometimes. I’m not sure where to go from here, I didn’t think anything like this would show up in the guest kernel logs.
```

https://github.com/paramiko/paramiko/issues/673

sivassrr, 2017-02-27:
This could be issue with opening lot of SSH connections and not closing it.  We are fixing it in
https://gerrit.openbmc-project.xyz/#/c/2690/

At the same time, Can we have a check in BMC for no. of connections threshold and issue a warning etc.  Another thing is most of the time it observed during reboot.  Is reboot fails because of more no. of connections left it open?

sivassrr, 2017-02-27:
[w37Witherspoon10iterationDr587.pdf](https://github.com/openbmc/openbmc/files/802740/w37Witherspoon10iterationDr587.pdf)
[w42Witherspoon5IterationDr587.pdf](https://github.com/openbmc/openbmc/files/802741/w42Witherspoon5IterationDr587.pdf)

sivassrr, 2017-02-28:
Good to see from Driver 587 on wards, I could see this on every reboot. 
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Phosphor Time Manager daemon.
[  OK  ] Stopped SSH Per-Connection Server (9.3.62.89:49346).
[  OK  ] Stopped Phosphor Webserver.
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:42108).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:42114).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:42116).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:42118).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:42120).

sivassrr, 2017-02-28:
With Driver [588](https://openpower.xyz/job/openbmc-build/588/), Could see this issue again. 
This time we are ensure that we close all open connections from test as well. 
[Dr588Witherspoon10IterationsCombinedReport.pdf](https://github.com/openbmc/openbmc/files/806996/Dr588Witherspoon10IterationsCombinedReport.pdf)

sivassrr, 2017-02-28:
Here is the serial connection output:
```
[  OK  ] Reached target System Time Synchronized.
[  OK  ] Stopped target Slices.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Started Dispatch Password Requests to Console Directory Watch.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Reached target System Initialization.
         Starting Network Service...
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Stopped Network Service.
         Starting Network Service...
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Stopped Network Service.
         Starting Network Service...
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Stopped Network Service.
         Starting Network Service...
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Stopped Network Service.
         Starting Network Service...
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Stopped Network Service.
[FAILED] Failed to listen on Network Service Bus Name.
See 'systemctl status org.freedesktop.network1.busname' for details.
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
[  OK  ] Reached target Network.

```
Nothing after this.

williamspatrick, 2017-02-28:
This indicates the network is not coming up.  Might be a kernel problem.  We need to get the 'dmesg' output and this:

```
[FAILED] Failed to listen on Network Service Bus Name.
See 'systemctl status org.freedesktop.network1.busname' for details.
[FAILED] Failed to start Network Service.
See 'systemctl status systemd-networkd.service' for details.
```

sivassrr, 2017-02-28:
[dmesgw37.txt](https://github.com/openbmc/openbmc/files/807443/dmesgw37.txt)

williamspatrick, 2017-02-28:
What about the two systemctl status commands?

sivassrr, 2017-03-01:
[systemdnetwordservice.txt](https://github.com/openbmc/openbmc/files/811152/systemdnetwordservice.txt)
[dmesgw33Mar1.txt](https://github.com/openbmc/openbmc/files/811151/dmesgw33Mar1.txt)

```
root@witherspoon:# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-6"
VERSION_ID="v1.99.3-6-g1603890-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-6"
BUILD_ID="v1.99.3"
root@witherspoon:# cat /etc/version
20170228221548
```

amboar, 2017-03-03:
Seems there's *some* issue with the network configuration, but I'm not sure it's relevant. From the `systemd-networkd` log:

```
Feb 22 06:05:31 witherspoon systemd-networkd[801]: eth0: Starting DHCPv6 client after NDisc timeout failed: Invalid argument
Feb 22 06:05:31 witherspoon systemd-networkd[801]: eth0: Configured
```

The `dmesg` output looks normal wrt networking.

sivassrr, 2017-03-13:
Had enabled data to collect systemctl, systemd-netword, dmesg and journalctl before the failure hit. 
In the first iteration itself it failed.

sivassrr, 2017-03-13:
[openbmc-test-automation1journalctl0.txt](https://github.com/openbmc/openbmc/files/837811/openbmc-test-automation1journalctl0.txt)
[openbmc-test-automation1dmesg0.txt](https://github.com/openbmc/openbmc/files/837813/openbmc-test-automation1dmesg0.txt)
[openbmc-test-automation1systemctl0.txt](https://github.com/openbmc/openbmc/files/837814/openbmc-test-automation1systemctl0.txt)
[openbmc-test-automation1NS0.txt](https://github.com/openbmc/openbmc/files/837816/openbmc-test-automation1NS0.txt)

sivassrr, 2017-03-13:
Some highlights from Journalctl: 

```
Feb 22 06:11:59 witherspoon systemd[1]: Started SSH Per-Connection Server (9.126.171.88:39432).
Feb 22 06:11:59 witherspoon dropbear[891]: Failed loading /etc/dropbear/dropbear_dss_host_key
Feb 22 06:11:59 witherspoon dropbear[891]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key

Feb 22 06:12:04 witherspoon dropbear[892]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Feb 22 06:12:04 witherspoon dropbear[892]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Feb 22 06:12:10 witherspoon dropbear[891]: Exit (root): Disconnect received
Feb 22 06:12:48 witherspoon systemd[1]: Started SSH Per-Connection Server (9.126.171.88:39514).
Feb 22 06:12:48 witherspoon dropbear[898]: Failed loading /etc/dropbear/dropbear_dss_host_key
Feb 22 06:12:48 witherspoon dropbear[898]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key

Feb 22 06:12:51 witherspoon dropbear[899]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Feb 22 06:12:51 witherspoon dropbear[899]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Feb 22 06:12:56 witherspoon systemd[1]: Started SSH Per-Connection Server (9.126.171.88:39516).
Feb 22 06:12:56 witherspoon dropbear[904]: Failed loading /etc/dropbear/dropbear_dss_host_key
Feb 22 06:12:56 witherspoon dropbear[904]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Feb 22 06:12:56 witherspoon dropbear[904]: Child connection from ::ffff:9.126.171.88:39516

Below may not be causing much problem, I guess:
Feb 22 06:05:39 witherspoon avahi-daemon[764]: Failed to open /usr/share/dbus-1/interfaces/org.freedesktop.Avahi.Server.xml: Invalid argument
```

sivassrr, 2017-03-13:
http://svn.dd-wrt.com/ticket/2279  - Related issue

sivassrr, 2017-03-13:
Journald is current in /tmp and it is not persistance, down the  line there is a plan to move journald to be persistance.

sivassrr, 2017-03-13:
This pattern (Ping Succees, SSH/REST - No),  Could see anywhere from BMC boot to even runtime.

gkeishin, 2017-03-13:
There seems to MAC address related issues on the witherspoon system.. Uma will take a look at the system and rectify the MAC's . 

Because of this the PING/SSH /REST wont respond to the calls made..

sivassrr, 2017-03-14:
@gkeishin do you think the failure "Failed loading /etc/dropbear/dropbear_dss_host_key" could be related to the MAC address related issue.

gkeishin, 2017-03-14:
Yeah.. that's one probably theory for now until Uma verifies the BMC MAC configuration in the lab...

swe12345, 2017-03-15:
Seeing similar issue with Palmetto after BMC reboot. System is pingable but SSH and REST login is failing. Attached are the serial logs from the system. 

Serial console logs:
[Palmetto_ssh_issue_serial_logs.txt](https://github.com/openbmc/openbmc/files/843570/Palmetto_ssh_issue_serial_logs.txt)

sivassrr, 2017-03-20:
If MAC configuration is the issue then why should serial port also should go freeze? 
We are hitting multiple times now.

sivassrr, 2017-03-21:
We keep seeing this issue in multiple systems and on other flavor as well. 
Checked with Uma and he thinks it is not related to Mac. He agreed to double check them.

williamspatrick, 2017-03-21:
The latest palmetto serial log shows we attempted to reboot the BMC. Is this expected?

If you left it running for a while at that point, it appears that the BMC is not going through a reboot correctly.

williamspatrick, 2017-03-21:
@shenki @amboar - This is one of the kernel hang issues we talked about the other day.

sivassrr, 2017-03-23:
With latest driver 647 hit with this issue.  Here is the serial console output.
[  OK  ] Removed slice system-org.openbmc.buttons.Power.slice.
[  OK  ] Stopped Clear one time boot overrides.
[  OK  ] Stopped Wait for /org/openbmc/settings/host0.
         Stopping Phosphor Settings Manager...
[  OK  ] Stopped Wait for /org/openbmc/control/bmc0.
[  OK  ] Stopped Wait for /org/openbmc/NetworkManager/Interface.
[  OK  ] Stopped Wait for /org/openbmc/managers/Download.
[  OK  ] Stopped target Fans Ready.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping Permit User Sessions...
[  OK  ] Removed slice system-xyz.openbmc_project.Hwmon.slice.
[  OK  ] Stopped target Network.
         Stopping Network Service...
[  OK  ] Removed slice system-xyz.openbmc_project.led.controller.slice.
[  OK  ] Removed slice system-obmc\x2dconsole.slice.
[  OK  ] Removed slice system-org.openbmc.control.Chassis.slice.
[  OK  ] Removed slice system-org.openbmc.control.Bmc.slice.
[  OK  ] Removed slice system-org.openbmc.watchdog.Host.slice.
[  OK  ] Stopped Phosphor Settings Manager.
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped Network Service.

rahulmah, 2017-03-24:
I also observed the same issue on Barreleye with today's build #649/20170324001736. This is a blocking issue. I have seen this issue 2 times in last three days while running CT.

sivassrr, 2017-03-26:
Stopping Avahi mDNS/DNS-SD Stack...
         Stopping Phosphor Version Software Manager...
         Stopping Phosphor Chassis State Manager...
         Stopping Phosphor Inventory Manager...
         Stopping Phosphor Console Muxer listening on device /dev/ttyVUART0...
         Stopping Phosphor Host State Manager...
         Stopping Phosphor Power Button0...
         Stopping Phosphor Log Manager...
         Stopping Phosphor Sensor Manager...
         Stopping Phoshpor System Manager...
         Stopping Phosphor User Manager...
         Stopping Phosphor Chassis0 Control...
         Stopping Phosphor MBOX Daemon...
         Stopping Phosphor Power0 Control...
         Stopping Phosphor LED Group Management Daemon...
         Stopping Lightweight SLP Server...
         Stopping Phosphor Watchdog0...
         Stopping Phosphor Host0 Checkstop Monitor...
         Stopping Phosphor BIOS Code Update...
         Stopping Phosphor BMC0 Control...
         Stopping Phosphor Fan Control...
         Stopping Phosphor Inband IPMI...
         Stopping Phosphor BMC Code Update...
         Stopping Phosphor Reset Button0...
         Stopping Phosphor Download Manager...
         Stopping Phosphor BMC State Manager...
         Stopping Phosphor IPMI BT DBus Bridge...
         Stopping Phosphor Inventory Manager...
         Stopping Phosphor Network Manager...
         Stopping Phosphor Time Manager daemon...
[  OK  ] Removed slice system-vcs_off_poweroff.slice.
[  OK  ] Stopped target System Time Synchronized.
         Stopping System Logging Service...
[  OK  ] Removed slice system-op\x2dreset\x2dset\x2dpower\x2don.slice.
[  OK  ] Stopped Wait for /org/openbmc/watchdog/host0.
         Stopping SSH Per-Connection Server (9.126.171.88:57582)...
[  OK  ] Stopped Phosphor BMC0 Control.
[  OK  ] Stopped Phosphor Version Software Manager.
[  OK  ] Stopped Phosphor Console Muxer listening on device /dev/ttyVUART0.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Phosphor sysfs LED controller.
[  OK  ] Stopped Phosphor Network Manager.
[  OK  ] Stopped Phosphor IPMI BT DBus Bridge.
[  OK  ] Stopped Phosphor Watchdog0.
[  OK  ] Stopped Phosphor sysfs LED controller.
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped Phosphor Chassis0 Control.
[  OK  ] Stopped Phosphor sysfs LED controller.
[  OK  ] Stopped Phosphor Sensor Manager.
[  OK  ] Stopped Phosphor Log Manager.
[  OK  ] Stopped Phosphor Inventory Manager.
[  OK  ] Stopped Phosphor User Manager.
[  OK  ] Stopped Phosphor Inventory Manager.
[  OK  ] Stopped Phosphor Download Manager.
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped Phosphor MBOX Daemon.
[  OK  ] Stopped Phosphor Event Manager.
[  OK  ] Stopped Phosphor LED Group Management Daemon.
[  OK  ] Stopped Serial Getty on ttyS4.
[  OK  ] Stopped Lightweight SLP Server.
[  OK  ] Stopped Phoshpor System Manager.
[  OK  ] Stopped Phosphor Webserver.
[  OK  ] Stopped Phosphor DNS Settings Cache Manager.
[  OK  ] Stopped Phosphor Fan Control.
[  OK  ] Stopped Phosphor BIOS Code Update.
[  OK  ] Stopped Phosphor BMC Code Update.
[  OK  ] Stopped Phosphor Inband IPMI.
[  OK  ] Stopped Phosphor Reset Button0.
[  OK  ] Stopped Phosphor Host0 Checkstop Monitor.
[  OK  ] Stopped Phosphor Host State Manager.
[  OK  ] Stopped Phosphor Power0 Control.
[  OK  ] Stopped Phosphor Chassis State Manager.
[  OK  ] Stopped Phosphor Power Button0.
[  OK  ] Stopped Phosphor BMC State Manager.
[  OK  ] Stopped Phosphor Time Manager daemon.
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:57574).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:57582).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:57584).
[  OK  ] Stopped SSH Per-Connection Server (9.126.171.88:57610).
[  OK  ] Stopped Reload mboxd after power off.
[  OK  ] Removed slice system-mboxd\x2dreload.slice.
[  OK  ] Stopped Wait for Power0 to turn off.
[  OK  ] Removed slice system-op\x2dwait\x2dpower\x2doff.slice.
[  OK  ] Removed slice system-dropbear.slice.
[  OK  ] Stopped Wait for /org/openbmc/control/power0.
[  OK  ] Removed slice system-org.openbmc.buttons.Power.slice.
[  OK  ] Removed slice system-org.openbmc.control.Power.slice.
[  OK  ] Removed slice system-org.openbmc.control.Checkstop.slice.
[  OK  ] Stopped Wait for /org/openbmc/managers/System.
[  OK  ] Removed slice system-org.openbmc.buttons.reset.slice.
[  OK  ] Stopped Clear one time boot overrides.
[  OK  ] Stopped Wait for /org/openbmc/settings/host0.
         Stopping Phosphor Settings Manager...
[  OK  ] Stopped Wait for /org/openbmc/control/bmc0.
[  OK  ] Stopped Wait for /org/openbmc/managers/Download.
[  OK  ] Stopped target Fans Ready.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped Wait for /org/openbmc/NetworkManager/Interface.
[  OK  ] Stopped target Webserver (pre).
[  OK  ] Stopped target Network.
         Stopping Network Service...
         Stopping Permit User Sessions...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
[  OK  ] Removed slice system-org.openbmc.control.Chassis.slice.
[  OK  ] Removed slice system-xyz.openbmc_project.Hwmon.slice.
[  OK  ] Removed slice system-org.openbmc.watchdog.Host.slice.
[  OK  ] Removed slice system-xyz.openbmc_project.led.controller.slice.
[  OK  ] Removed slice system-obmc\x2dconsole.slice.
[  OK  ] Removed slice system-org.openbmc.control.Bmc.slice.
[  OK  ] Stopped Phosphor Settings Manager.
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Permit User Sessions.

Driver 655 (Latest) - Hanged after BMC reset. Above is the serial output. Ping fine, No SSH, Serial Connection freeze

gkeishin, 2017-03-30:
One hang failure.. last seen message on serial console

```
 slave@01:00: engine[20]: type 1b, version 1, addr 70000 size 8000
 slave@01:00: engine[21]: type 1b, version 1, addr 78000 size 8000
[  OK  ] Removed slice system-op\x2dreset\x2dset\x2dpower\x2don.slice.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
[  OK  ] Stopped target Timers.
[  OK  ] Removed slice system-op\x2dwait\x2dpower\x2doff.slice.
[  OK  ] Stopped target Chassis0 (Start).
[  OK  ] Stopped Start Watchdog0.
[  OK  ] Stopped target Chassis0 (Power On).
         Stopping SSH Per-Connection Server (9.41.165.94:47800)...
         Stopping SSH Per-Connection Server (9.41.165.94:48182)...
[  OK  ] Removed slice system-fsi\x2dbind.slice.
[  OK  ] Removed slice system-obmc\x2dstart\x2dwatchdog.slice.
[  OK  ] Stopped Wait for /org/openbmc/watchdog/host0.
[  OK  ] Removed slice system-vcs_off_poweroff.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/state/chassis0.
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-mboxd\x2dreload.slice.
         Stopping Phosphor Webserver...
[  OK  ] Removed slice system-op\x2dreset\x2dpgood\x2dcheck.slice.
[  OK  ] Stopped target Host0 (Started).
[  OK  ] Stopped target Start Host0.
[  OK  ] Stopped Start host0 SBE.
[  OK  ] Stopped target Start Host0 (Pre).
[  OK  ] Stopped Set fans to a default speed.
[  OK  ] Removed slice system-fan\x2ddefault\x2dspeed.slice.
[  OK  ] Stopped target Restore Last State.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-op\x2dpower\x2dstop.slice.
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped Wait for /org/openbmc/control/chassis0.
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2don.slice.
[  OK  ] Stopped Wait for /org/openbmc/control/flash/bios.
[  OK  ] Removed slice system-start_host.slice.
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped Turn VCS on.
[  OK  ] Stopped Run VCS workaround on host0.
[  OK  ] Stopped Scan FSI devices.
[  OK  ] Removed slice system-fsi\x2dscan.slice.
[  OK  ] Removed slice system-vcs_on.slice.
[  OK  ] Removed slice system-vcs_workaround.slice.
[  OK  ] Removed slice system-obmc\x2dstop\x2dwatchdog.slice.
[  OK  ] Stopped PCIE Slot Presence Detection For Chassis0.
[  OK  ] Removed slice system-pcie\x2dslot\x2ddetect.slice.
[  OK  ] Stopped target Power0 (On).
[  OK  ] Stopped target Power0 On.
[  OK  ] Stopped Wait for Power0 to turn on.
[  OK  ] Removed slice system-op\x2dwait\x2dpower\x2don.slice.
[  OK  ] Stopped Start Power0.
[  OK  ] Stopped target Fan Control.
[  OK  ] Removed slice system-op\x2dpower\x2dstart.slice.
[  OK  ] Stopped target Power0 On (Pre).
[  OK  ] Stopped target Phosphor System - Standby.
         Stopping Phosphor Download Manager...
         Stopping Phosphor Chassis0 Control...
         Stopping Lightweight SLP Server...
         Stopping Phosphor Fan Control...
         Stopping Phosphor Network Manager...
         Stopping Phosphor DNS Settings Cache Manager...
         Stopping Phosphor BMC State Manager...
         Stopping Phosphor Power0 Control...
         Stopping Phosphor Inband IPMI...
         Stopping Phosphor User Manager...
         Stopping Phosphor Inventory Manager...
         Stopping Phosphor LED Group Management Daemon...
[... 94 lines cut ...]
[  OK  ] Stopped target Webserver (pre).
[  OK  ] Removed slice system-serial\x2dgetty.slice.
         Stopping Permit User Sessions...
[  OK  ] Removed slice system-obmc\x2dconsole.slice.
[  OK  ] Removed slice system-xyz.openbmc_project.led.controller.slice.
[  OK  ] Removed slice system-org.openbmc.watchdog.Host.slice.
[  OK  ] Removed slice system-org.openbmc.control.Chassis.slice.
[  OK  ] Removed slice system-org.openbmc.control.Bmc.slice.
[  OK  ] Stopped Phosphor Settings Manager.
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Network.
         Stopping Network Service...
[  OK  ] Stopped Wait for /org/openbmc/sensors.
[  OK  ] Removed slice system-mapper\x2dwait.slice.
[  OK  ] Stopped target Phosphor Object Mapper.
         Stopping Phosphor DBus Service Discovery Manager...
[  OK  ] Stopped Network Service.
[  OK  ] Stopped Phosphor DBus Service Discovery Manager.
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF
ftgmac100 1e660000.ethernet eth0: [ISR] = 0x57: NO_RXBUF
```

gkeishin, 2017-03-30:
I think this problem went back when the BMC reboot was implemented https://github.com/openbmc/openbmc/issues/1070

so we may want to look at how the BMC reboot is actually forcing BMC to reset @andrewg

geissonator, 2017-03-30:
If you search for "systemd reboot.target hangs" you'll see quite a few hits.  @saqibkh is working on updating systemd next for us in openbmc, I'm hoping we maybe pull in a fix there.

sivassrr, 2017-04-05:
Didnt observe much difference in serial connection output during the normal IPL and BMC hang time.

saqibkh, 2017-04-06:
systemd was upgraded to version 232 today. Please test the master branch for similar failures.

sivassrr, 2017-04-09:
Could able to recreate with driver 682. Here is the combined test report. Last 3 iteration complete failure. At-least till 7th iteration some test cases passed. 
[logwitherspoonCombinedIterations10Report.pdf](https://github.com/openbmc/openbmc/files/908072/logwitherspoonCombinedIterations10Report.pdf)

sivassrr, 2017-05-22:
With the [latest](https://openpower.xyz/job/openbmc-build/751/) BMC driver , And latest PNOR, Still hit the same issue. BMC failed to ssh, Ping success. 

root@witherspoon:~# cat /etc/timestamp
20170519212253
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-60"
VERSION_ID="v1.99.5-60-gecfa662"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.
5-60"
BUILD_ID="v1.99.5"
root@witherspoon:~#

gkeishin, 2017-05-26:
```
•	BUILD ID   : https://openpower.xyz/job/openbmc-build/766/
•	VERSION_ID : v1.99.6-10-g1d22c22
•	BMC Kernel : Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri May 26 05:22:23 UTC 2017 armv6l GNU/Linux
```
we are hit on new kernel 4.10

geissonator, 2017-05-26:
Here's the same section in the shutdown, but on one that works, anyone see an obvious service that's not stopping correctly in the fail conditions?
```
[  OK  ] Stopped Phosphor LED Group Management Daemon.
[  OK  ] Stopped OpenBMC Software Update Manager.
[  OK  ] Stopped OpenPower Software Update Manager.
[  OK  ] Stopped Phosphor Settings Manager.
[  OK  ] Stopped Phosphor Hwmon Poller.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Network.
         Stopping Network Service...
[  OK  ] Stopped Wait for /org/openbmc/sensors.
[  OK  ] Removed slice system-mapper\x2dwait.slice.
[  OK  ] Stopped target Phosphor Object Mapper.
         Stopping Phosphor DBus Service Discovery Manager...
[  OK  ] Stopped Phosphor DBus Service Discovery Manager.
[  OK  ] Stopped Network Service.
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped Forward Password Requests to Wall Directory Watch.
[  OK  ] Stopped Dispatch Password Requests to Console Directory Watch.
[  OK  ] Stopped target Slices.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed Phosphor Host Console SSH Per-Connection socket.
[  OK  ] Closed Phosphor DBUS Browser socket.
[  OK  ] Closed Phosphor Webserver socket.
[  OK  ] Closed phosphor-ipmi-net.socket.
[  OK  ] Closed dropbear.socket.
[  OK  ] Closed Avahi mDNS/DNS-SD Stack Activation Socket.
[  OK  ] Closed Syslog Socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target System Initialization.
```

geissonator, 2017-06-28:
Spent some time on this last week and today.  It seems like the issue here is that systemd is getting hung up on the reboot.  Patrick and I found a few interesting reads in this area:
- https://bbs.archlinux.org/viewtopic.php?id=170756&p=2
- https://github.com/systemd/systemd/issues/4955

I tried the solution proposed in one but I think our dbus.service is required by basic.target so making it run Before= caused a cyclical dependency.  

One theory is that the automated test software is coming in via REST during the reboot process and starting up the REST/network service.  I've asked @gkeishin to try a test where he adds a delay between the reboot request and the first SSH/REST interaction as a data point.

geissonator, 2017-06-28:
Been running a recreate attempt with the following:
```
vi /etc/systemd/journald.conf
ForwardToConsole=yes
```
So far no luck, getting the journal to the console on the fail should really help us debug the problem, but if it's a race condition, this may end up changing the timings.

I also had a bash loop going in parallel doing curl command logins.

gkeishin, 2017-06-28:
There u go here is the recreate Serial log
```
[  198.194306] systemd[1]: Failed to start Network Service.
[  198.224451] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  198.264321] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  198.304249] systemd[1]: systemd-networkd.service: Service has no hold-off time, scheduling restart.
[  198.344226] systemd[1]: Stopped Network Service.
[  198.374335] systemd[1]: systemd-networkd.service: Start request repeated too quickly.
[  198.414543] systemd[1]: Failed to start Network Service.
[  198.444390] systemd[1]: systemd-networkd.socket: Unit entered failed state.
[  198.484595] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  198.524363] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  198.564437] systemd[1]: Reached target Network.
```
[BMC_hang_on_reset.txt](https://github.com/openbmc/openbmc/files/1107954/BMC_hang_on_reset.txt)

geissonator, 2017-06-28:
That's good stuff, as you noted, the "196.344450] systemd[1]: Starting Network Service..." looks very bad, and is most likely what's causing the systemd shutdown hang.  The key question is what's causing that service to start.  One thing I noticed a bit earlier was this timesyncd service and target starting up:
```
[  195.554330] systemd-timesyncd[671]: Network configuration changed, trying to establish connection.
...
[  196.234502] systemd[1]: Reached target System Time Synchronized.
...
[  196.344450] systemd[1]: Starting Network Service...
...
[  198.564437] systemd[1]: Reached target Network.
```

I'm wondering if that could be what's starting the network service up.

https://github.com/systemd/systemd/blob/2340bfbfccb0c1361becbb527bf9fe3e8c174cf0/src/timesync/timesyncd-manager.c#L1075

My guess is a race condition on that code looking at /run/systemd/netif/state to determine if the network is online.  A simple test would be to ensure the timesyncd and networkd have the right relationship in their service files.

geissonator, 2017-06-29:
There was a workaround proposed in https://github.com/systemd/systemd/issues/4065#issuecomment-311827007.  The change did ensure the systemd-timesyncd stopped prior to the systemd-networkd but that just gave us even more interesting data!

```
[  146.869375] systemd[1]: Stopped Network Service.
[  146.996911] systemd[1]: Stopped Phosphor DBus Service Discovery Manager.
[  147.042357] systemd[1]: dev-hugepages.mount: Couldn't determine result for ConditionVirtualization=!private-users, assuming failed: No such file or directory
[  147.086210] systemd[1]: Reached target Final Step.
[  147.117166] systemd[1]: Stopping D-Bus System Message Bus...
[  147.157102] systemd[1]: Stopped D-Bus System Message Bus.
[  147.204214] systemd[1]: Stopped target Basic System.
[  147.251985] systemd[1]: Stopped target Paths.
[  147.303592] systemd[1]: Stopped Forward Password Requests to Wall Directory Watch.
[  147.356637] systemd[1]: Stopped target Slices.
[  147.418234] systemd[1]: Starting Network Time Synchronization...
[  147.485401] systemd[1]: Stopped target Sockets.
[  147.562871] systemd[1]: Closed Phosphor Host Console SSH Per-Connection socket.
[  147.642702] systemd[1]: Closed dropbear.socket.
[  147.724879] systemd[1]: Closed Syslog Socket.
[  147.806081] systemd[1]: Closed Phosphor DBUS Browser socket.
[  147.897721] systemd[1]: Closed Avahi mDNS/DNS-SD Stack Activation Socket.
[  147.996902] systemd[1]: Closed phosphor-ipmi-net.socket.
[  148.103618] systemd[1]: Closed Phosphor Webserver socket.
[  148.135497] systemd[1]: Closed D-Bus System Message Bus Socket.
[  148.174802] systemd[1]: Starting Network Service...
[  148.204364] systemd-networkd[1158]: Could not connect to bus: Connection refused
[  148.242952] systemd[1]: systemd-networkd.service: Main process exited, code=exited, status=1/FAILURE
[  148.284514] systemd[1]: Failed to start Network Service.
[  148.314426] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  148.344296] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  148.384213] systemd[1]: systemd-networkd.service: Service has no hold-off time, scheduling restart.
[  148.424224] systemd[1]: Stopped Network Service.
[  148.454257] systemd[1]: Starting Network Service...
[  148.484179] systemd[1]: Started Network Time Synchronization.
[  148.514280] systemd[1]: Reached target System Time Synchronized.
[  148.544677] systemd-networkd[1159]: Could not connect to bus: Connection refused
[  148.582799] systemd[1]: systemd-networkd.service: Main process exited, code=exited, status=1/FAILURE
[  148.624273] systemd[1]: Failed to start Network Service.
...
[  149.734398] systemd[1]: Failed to start Network Service.
[  149.764292] systemd[1]: systemd-networkd.socket: Unit entered failed state.
[  149.804563] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  149.844455] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  149.884316] systemd[1]: Reached target Network.
```

Note the "[  147.418234] systemd[1]: Starting Network Time Synchronization..."  What the heck is going on here??

geissonator, 2017-06-29:
I don't see the "[  147.042357] systemd[1]: dev-hugepages.mount: Couldn't determine result for ConditionVirtualization=!private-users, assuming failed: No such file or directory" in the good path reboots.

geissonator, 2017-06-29:
Don't think that above error is related, but this code looks a bit suspicious if it's somehow getting called in the shutdown:
https://github.com/openbmc/phosphor-networkd/blob/master/netman.py#L75

Looks like it's triggered via a dbus call to UpdateUseNtpField which is called from https://github.com/openbmc/phosphor-time-manager/blob/master/time-config.cpp#L256

gkeishin, 2017-06-29:
Still the same failure with that 
```
[  152.714277] systemd[1]: Stopped Network Service.
[  152.744287] systemd[1]: Starting Network Service...
[  152.774483] systemd-networkd[1158]: Could not connect to bus: Connection refused
[  152.812783] systemd[1]: systemd-networkd.service: Main process exited, code=exited, status=1/FAILURE
[  152.854253] systemd[1]: Failed to start Network Service.
[  152.884345] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  152.924215] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  152.964568] systemd[1]: systemd-networkd.service: Service has no hold-off time, scheduling restart.
[  153.004360] systemd[1]: Stopped Network Service.
[  153.034338] systemd[1]: systemd-networkd.service: Start request repeated too quickly.
[  153.074382] systemd[1]: Failed to start Network Service.
[  153.104315] systemd[1]: systemd-networkd.socket: Unit entered failed state.
[  153.154339] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  153.194365] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  153.234379] systemd[1]: Reached target Network.
```
- [x] Console log 
[BMC_hang_on_reset_new.txt](https://github.com/openbmc/openbmc/files/1111279/BMC_hang_on_reset_new.txt)

geissonator, 2017-06-29:
Looks like I've at least got a workaround that made it through 30 iterations.  @gkeishin could you give it a shot?
```
vi /lib/systemd/system/systemd-networkd.service
Restart=no

vi /lib/sysetmd/system/systemd-timesyncd.service
After=systemd-remount-fs.service systemd-tmpfiles-setup.service systemd-sysusers.service systemd-network.service
Before=time-sync.target shutdown.target
```

williamspatrick, 2017-06-29:
Changing systemd-networkd.service to `Restart=no` is not a workable solution:

```
Restart=
           Configures whether the service shall be restarted when the service
           process exits, is killed, or a timeout is reached.
```

That means if there is a crash of networkd we will never get it restarted.

williamspatrick, 2017-06-29:
Wouldn't adding `After` and maybe `Wants` in `timesyncd.service` on `systemd-network.service` be sufficient?

gkeishin, 2017-06-30:
Look like it hasn't fixed with the above service changes
```
09:45:11.891 	INFO 	Repeating keyword, round 1/50. 	
09:48:27.757 	INFO 	Repeating keyword, round 2/50. 	
09:51:16.379 	INFO 	Repeating keyword, round 3/50. 	
09:53:55.569 	INFO 	Repeating keyword, round 4/50
```
With the above changes broke much earlier within 5 iterations then earlier iterations.

```
[  134.864233] systemd[1]: Closed D-Bus System Message Bus Socket.
[  134.894223] systemd[1]: Starting Network Service...
[  134.924326] systemd-networkd[1143]: Could not create runtime directory: File exists
[  134.962732] systemd-networkd[1143]: Could not create runtime directory 'links': File exists
[  135.002685] systemd-networkd[1143]: Could not create runtime directory 'leases': File exists
[  135.042656] systemd-networkd[1143]: Could not connect to bus: Connection refused
[  135.072850] systemd[1]: systemd-networkd.service: Main process exited, code=exited, status=1/FAILURE
[  135.114546] systemd[1]: Failed to start Network Service.
[  135.144272] systemd[1]: systemd-networkd.service: Unit entered failed state.
[  135.184483] systemd[1]: systemd-networkd.service: Failed with result 'exit-code'.
[  135.224343] systemd[1]: Reached target Network.
[  135.254180] systemd[1]: Started Network Time Synchronization.
[  135.284252] systemd[1]: Reached target System Time Synchronized
```
- [x] Serial console log 
[BMC_hang_on_reset_log.txt](https://github.com/openbmc/openbmc/files/1114127/BMC_hang_on_reset_log.txt)

rfrandse, 2017-07-01:
<https://gerrit.openbmc-project.xyz/5062> Make the BMC reboot irreversible
Resolves  #1161 Witherspoon: bmc intermittently hangs on reboot, serial console hangs

[2 more replies]

closed 2017-07-06 by williamspatrick