openbmc/openbmc #1162: ipmi-fru-parser: update system specific config YAML

dkodihal, 2017-02-20:
Currently only a default system specific FRU config YAML exists. Define one per OpenPOWER system.

ratagupt, 2017-02-23:
Hoping this story is for creating the system specific inventory.yaml which can be used by the code for building the inventory.

ratagupt, 2017-02-23:
- [ ] Palmetto

- [ ] Barreleye

- [ ] Romulus

- [ ] Firestone

ratagupt, 2017-02-23:
- [ ] Garrison

dkodihal, 2017-02-23:
For systems with MRW, https://github.com/openbmc/openbmc/blob/master/meta-phosphor/common/recipes-phosphor/ipmi/phosphor-ipmi-fru-inventory-mrw-native.bb generates the config.yaml.

For systems without MRW, this needs to be hand-written. This could be done via a system-specific bbappend or a native recipe, such that the hand-written config.yaml ends up in the same location as the MRW one. This way phosphor-ipmi-fru.bb will always find it here -
EXTRA_OECONF = "YAML_GEN=${STAGING_DIR_NATIVE}${config_datadir}/config.yaml"

Similar work is needed for the "additional properties" out.yaml coming via https://gerrit.openbmc-project.xyz/#/c/2537/.

closed 2017-03-10 by rfrandse