openbmc/openbmc #1164: [Barreleye] Host not booting after host power on with latest BMC build

rahulmah, 2017-02-20:
In Barreleye, Host not booting after host power on with latest BMC build. See step to recreate to more for info.

1. Make sure host state is off and BMC at Ready state
```
bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                           {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  },
  "message": "200 OK",
  "status": "ok"
}
bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0
{
  "data": {
    "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
    "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
  },
  "message": "200 OK",
  "status": "ok"
}
```

2. Power on host using rest
```
bash-4.1$curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.On\"}" -X PUT  https://xx.xx.xx.xx/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```

3. check host and chassis state (here we will see that both HOST and chassis state remain at off)
```
bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                           {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$

bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/chassis0
{
  "data": {
    "CurrentPowerState": "xyz.openbmc_project.State.Chassis.PowerState.Off",
    "RequestedPowerTransition": "xyz.openbmc_project.State.Chassis.Transition.Off"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

obmcutil state reflect powering on
```
root@barreleye:~# obmcutil state
 = HOST_POWERING_ON
root@barreleye:~#
```
FFDC : 
[Journal HOST not powering on.txt](https://github.com/openbmc/openbmc/files/787859/Journal.HOST.not.powering.on.txt)

BMC fw:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-68"
VERSION_ID="v1.99.2-68-gb792af2-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-68"
BUILD_ID="v1.99.2"
root@barreleye:~# cat /etc/version
20170218050016
root@barreleye:~#
```

ratagupt, 2017-02-20:
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-71"
VERSION_ID="v1.99.2-71-gbc49f79-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-71"
BUILD_ID="v1.99.2"
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# obmcutil state
 = HOST_POWERED_ON
root@witherspoon:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 36:3E:37:64:6B:00  
          inet addr:9.3.185.174  Bcast:9.3.185.255  Mask:255.255.255.0
          inet6 addr: fe80::343e:37ff:fe64:6b00/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:334969 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9866 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20615888 (19.6 MiB)  TX bytes:1225031 (1.1 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:41124 errors:0 dropped:0 overruns:0 frame:0
          TX packets:41124 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:3125648 (2.9 MiB)  TX bytes:3125648 (2.9 MiB)

ratagupt, 2017-02-20:
```root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-71"
VERSION_ID="v1.99.2-71-gbc49f79-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-71"
BUILD_ID="v1.99.2"
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# 
root@witherspoon:~# obmcutil state
 = HOST_POWERED_ON
root@witherspoon:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 36:3E:37:64:6B:00  
          inet addr:9.3.185.174  Bcast:9.3.185.255  Mask:255.255.255.0
          inet6 addr: fe80::343e:37ff:fe64:6b00/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:334969 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9866 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20615888 (19.6 MiB)  TX bytes:1225031 (1.1 MiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:41124 errors:0 dropped:0 overruns:0 frame:0
          TX packets:41124 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:3125648 (2.9 MiB)  TX bytes:3125648 (2.9 MiB)
```

ratagupt, 2017-02-20:
@rahulmah :- I tried on the above box which is having latest firmware and it is powered on.

rahulmah, 2017-02-21:
@ratagupt : Even I also noticed and mentioned above that obmcutil state changes to ON here. But I dont think that HOST is really booting as host state is not changing to Running with new xyz interface. I am also noticing that obmc-console-client does not show anything after powering on the Host. Tested with today's latest build 564/v1.99.2-89-gf327ed7/20170221021702 and still seeing the problem.

ratagupt, 2017-02-21:
I looked at the logs shared by the @rahulmah .

Feb 01 03:02:28 barreleye systemd[1]: Failed to start Phosphor Version Software Manager.
Feb 01 03:02:28 barreleye systemd[1]: xyz.openbmc_project.Software.Version.service: Unit entered failed state.
Feb 01 03:02:28 barreleye systemd[1]: xyz.openbmc_project.Software.Version.service: Failed with result 'timeout'.
Feb 01 03:02:28 barreleye systemd[1]: xyz.openbmc_project.Software.Version.service: Service hold-off time over, scheduling restart.

rahulmah, 2017-02-21:
@ratagupt : Hold on this one.  After rebooting system using PDU, this issue is not seen. Let me get the exact steps to reproduce it.

rahulmah, 2017-02-23:
@ratagupt : Here are the steps to recreate this issue on Barreleye. Once we hit this issue after following below steps we need to do power cycle to get host in Running state. Tested on Barreleye system with latest BMC and pnor image. FFDC attached below.

```
BMC fw : v1.99.2-104-g0fe26a0/20170222222246
PNOR image : https://openpower.xyz/job/openpower-op-build/distro=ubuntu,target=barreleye/305/artifact/images/
root@barreleye:~# cat /tmp/out.txt
open-power-barreleye-v1.14-55-ga397ed9
        buildroot-2016.11-4-g6b9fd80
        skiboot-5.4.3
        hostboot-fe5c556
        linux-4.9.8-openpower1-pe71c806
        petitboot-v1.4.1-pc5f30a7
        barreleye-xml-253622f
        occ-f018940
        hostboot-binaries-541a2bd
        capp-ucode-1bb7503
root@barreleye:~#
```
 
1. Powercycle system using PDU. (This is needed, if you want to start on the system where host is not booting to Host Running state)

2. After power cycle, Host will come to off state and BMC to State

```
bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0"
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off"

bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0
{
  "data": {
    "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
```

3. Now power on the Host via REST. Here you will see that HOST will come to Running state with any problem.
```
bash-4.1$ curl -c cjar -b cjar -k-H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.On\"}" -X PUT  https://xx.xx.xx.xx/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                          {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}
```

4. Once host reaches running state, trigger warmreset via REST. Host will come to off state after it.

```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": []}" -X POST  https://xx.xx.xx.xx/org/openbmc/control/bmc0/action/warmRest
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
bash-4.1$ curl -c cjar -b cjar -k-H "Content-Type: application/json" -X POST https://xx.xx.xx.xx/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"            {
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                          {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/BMC0
{
  "data": {
    "CurrentBMCState": "xyz.openbmc_project.State.BMC.BMCState.Ready",
    "RequestedBMCTransition": "xyz.openbmc_project.State.BMC.Transition.None"
  },
  "message": "200 OK",
  "status": "ok"
}
```

5. Now power on the host again via REST. Here you will see that HOST state will remain off waited for 30 mins. We will continue to see this problem even after multiple power off and on, bmc reboot. Only after pdu reboot, host will boot to Running state

```
bash-4.1$ curl -c cjar -b cjar -k-H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.On\"}" -X PUT  https://xx.xx.xx.xx/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0   
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",            <---------- Current Host state will remain off. waited for 20-30 mins
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}
```
6. obmcutil state will keep showing HOST_POWERING_ON and does not go to HOST_BOOTED state

```
root@barreleye:~# obmcutil state
 = HOST_POWERING_ON
root@barreleye:~#
```
FFDC
[20170223031649589340_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/796067/20170223031649589340_MyFFDCLogs.zip)

geissonator, 2017-02-23:
ok, makes sense, story #1094 tracks this function.  We can close this once that code is in and verified.

geissonator, 2017-05-11:
Master now support a BMC reboot while the host is running.  BMC will come back up, discover chassis is on and host is running, and update obmcutil states accordingly.  There is a pgood gpio bug on witherspoon that breaks this function but it works fine on barreleye.

closed 2017-05-11 by geissonator