openbmc/openbmc #1165: Host OS goes to quiesce [Witherspoon v1.99.2-71-gbc49f79 ]

gkeishin, 2017-02-20:
```
Power on  to OS, reboot OS .. OS goes to quiesce mode
```
From journal data
`
Feb 02 17:29:12 witherspoon systemd[1]: Reached target Quiesce Target.
`

- [x] Journald log 
[journald_v1.99.2-71-gbc49f79_Quiesce.txt](https://github.com/openbmc/openbmc/files/788038/journald_v1.99.2-71-gbc49f79_Quiesce.txt)

gkeishin, 2017-02-20:
```
SYSTEM     : W38 ( Witherspoon)
BUILD_ID   : witherspoon-20170218231615.all.tar  ( openbmc master + openFSI enablement for MFG )
VERSION_ID : v1.99.2-71-gbc49f79
BUILD TIME : 20170218231615
PNOR VERSION:

IBM-witherspoon-ibm-OP9_v1.14_1.4
        op-build-5f2e132
        buildroot-6b9fd80
        skiboot-8c51fda
        hostboot-6782606
        linux-4.9.8-10d96bf
        petitboot-v1.4.1-42f37cc
        witherspoon-xml-3fd670a
        occ-4141b5f
        hostboot-binaries-541a2bd
        capp-ucode-1bb7503
```

spinler, 2017-02-20:
```Feb 02 17:29:12 witherspoon mapper[1531]: Error finding '/org/openbmc/watchdog/host0' service: Device or resource busy
Feb 02 17:29:12 witherspoon systemd[1]: obmc-start-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 17:29:12 witherspoon systemd[1]: Failed to start Start Watchdog0.
Feb 02 17:29:12 witherspoon systemd[1]: Dependency failed for Chassis0 (Start).```

geissonator, 2017-02-20:
Yeah, this is broken (or really has never worked with new code).  Working on design with Brad on how to make it work.

geissonator, 2017-02-23:
This is the same as #1164 and since that one is 1 digit earlier, we'll use that one to track this issue.

closed 2017-02-23 by geissonator