openbmc/openbmc #1169: Inventory Versioning [ Witherspoon v1.99.2-89-gf327ed7 ]

gkeishin, 2017-02-21:
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/564/distro=ubuntu,target=witherspoon/
- [x] VERSION_ID : v1.99.2-89-gf327ed7
- [x] BUILD TIME : 20170221021702
- [x] PNOR VERSION:
```
IBM-witherspoon-ibm-OP9_v1.14_1.4
        op-build-5f2e132
        buildroot-6b9fd80
        skiboot-8c51fda
        hostboot-6782606
        linux-4.9.8-10d96bf
        petitboot-v1.4.1-42f37cc
        witherspoon-xml-3fd670a
        occ-4141b5f
        hostboot-binaries-541a2bd
        capp-ucode-1bb7503
```
After having a discussion with @ratagupt and @dkodihal , we decided to open a defect for this

**Example: Inventory Data**

```
The versioning and type of data in it are not consistent

1.  Non-Hex  "Version": "10"     vs  "Version": "0x31"
2.  Even on the same FRU's example DIM the versioning varies 
"Version": "0x00"  vs "Version": "0x31"

Note: On Barreleye from the org inventory data the versioning shows empty "". 
```

```
  "data": {
    "/xyz/openbmc_project/Inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "", 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01HL322", 
      "Present": 0, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "YA3933741571", 
      "Version": "10"
    }, 
    "/xyz/openbmc_project/Inventory/system/chassis/motherboard/dimm0": {
      "BuildDate": "", 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 0, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb28b", 
      "Version": "0x31"
    }, 
   
   "/xyz/openbmc_project/Inventory/system/chassis/motherboard/dimm2": {
      "BuildDate": "", 
      "Manufacturer": "0xad80", 
      "Model": "HMA41GR7AFR4N-UH    ", 
      "PartNumber": "", 
      "Present": 0, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x51906156", 
      "Version": "0x00"
    }, 
    "/xyz/openbmc_project/Inventory/system/chassis/motherboard/dimm3": {
      "BuildDate": "", 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 0, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb1e5", 
      "Version": "0x31"
    },
```

dkodihal, 2017-02-21:
I don’t think hex vs ascii is an issue at all. In fact the 0x helps you figure out that the data is hex.

Dimm2 doesn’t have the version info encoded like others, so you see a 0x0 there.

The only issue I see is, that too this is a vpd programming issue, is that the DIMM manufacturer is 0x2c80, and PrettyName is 0x0c. These are not obvious values for manufacturer and name, need to get these checked up.

geissonator, 2017-02-23:
It doesn't sound like this is anything the openbmc team is going to fix?  Maybe a internal CQ defect on the GSI team and close this out?

gkeishin, 2017-02-26:
@dkodihal @ratagupt Anything here we need to know before we open an internal CQ for GI ? Let me know

dkodihal, 2017-02-26:
No, I think we can go with this information.

geissonator, 2017-02-28:
Please open CQ on our GSI team.

closed 2017-02-28 by geissonator

sivassrr, 2017-03-02:
FW665454 CQ Issue Done.

williamspatrick, 2017-03-02:
I think a potential feature request that could come in would be to ask the BMC to do translation of the JEDEC MFG IDs to strings.  It isn't something I think we should consider high enough priority to tackle now though.