openbmc/openbmc #1170: eSEL data not visible via REST [ Witherspoon v1.99.2-72-gd11ec2e ]

gkeishin, 2017-02-21:
- [x] SYSTEM     : W37 ( Witherspoon)
- [x] BUILD_ID   : witherspoon-20170221015852.all.tar
- [x] VERSION_ID : v1.99.2-72-gd11ec2e
- [x] BUILD TIME : 20170221015852
- [x] PNOR VERSION:

```
IBM-witherspoon-ibm-OP9_v1.14_1.4
        op-build-5f2e132
        buildroot-6b9fd80
        skiboot-8c51fda
        hostboot-6782606
        linux-4.9.8-10d96bf
        petitboot-v1.4.1-42f37cc
        witherspoon-xml-3fd670a
        occ-4141b5f
        hostboot-binaries-541a2bd
        capp-ucode-1bb7503
```

eSEL via dbus 
```
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        1 "ESEL=00 00 df 00 00 00 00 20 00 04... emits-change writable
.Id                                 property  u         1                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1486056355620                            emits-change writable
root@witherspoon:~#

```

**From REST call** 
```
rango@ubuntu:~/Desktop$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET  https://xx.xx.xx.xx/xyz/openbmc_project/Logging/Entry/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Logging/Entry"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}rango@ubuntu:~/Desktop$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X  GET  https://xx.xx.xx.xx/xyz/openbmc_project/Logging/enumerate
{
  "data": {
    "/xyz/openbmc_project/Logging/Internal/Manager": {}
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

- [x]  Journal log 
[journald_v1.99.2-72-gd11ec2e_20170221015852_eSEL.txt](https://github.com/openbmc/openbmc/files/790591/journald_v1.99.2-72-gd11ec2e_20170221015852_eSEL.txt)

bradbishop, 2017-02-21:
I have a feeling org.freedesktop.DBus.ObjectManager.InterfacesAdded signals are not being emitted when the SEL object is created.

closed 2017-02-22 by williamspatrick

gkeishin, 2017-02-22:
**Verified**
```
==============================================================================
Test Esel :: This suite is for testing esel's mechanism of checking Reserva...
==============================================================================
Verify eSEL Using REST :: Generate eSEL log and Verify using REST.    .URI:https://xx.xx.xx.xx/xyz/openbmc_project/Logging/Entry/1, method:Get, args:{}
Response code:404, Content:{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/Logging/Entry/1"
  }, 
  "message": "404 Not Found", 
  "status": "error"
}
.......URI:https://xx.xx.xx.xx/xyz/openbmc_project/Logging/Entry/1, method:Get, args:{}
Response code:200, Content:{
  "data": {
    "AdditionalData": [
      "ESEL=00 00 df 00 00 00 00 20 00 04 12 35 6f aa 00 00 "
    ], 
    "Id": 1, 
    "Message": "org.open_power.Error.Host.Event.Event", 
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
    "Timestamp": 1486060795107
  }, 
  "message": "200 OK", 
  "status": "ok"
}
Verify eSEL Using REST :: Generate eSEL log and Verify using REST.    | PASS |
------------------------------------------------------------------------------
```