openbmc/openbmc #1173: [Barreleye] Presence Detect failing on fans (even speed being read Ok), and value==127 for fake fan records

yadron, 2017-02-21:
The speed values for Barreleye correct, so I expect that Presence should be True (looking to func "fan_update_present(fan_info_t *info)" of build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-control-fan/1.0-r1/git/fanctl/fan_control.c  ).
For some reason, it is False.

If I add one more line of code to fan_control.c (and both some debug fprintf stderr output)
(
```
int start_fan_services(fan_info_t *info)
{
...
        }

        fan_update_present(info); // **this is a new line**

        for (;;) {
                /* Process requests */
                rc = sd_bus_process(info->bus, NULL);

```
), then the result is:

```
root@barreleye:~# ~/fan_control.exe -f 6
Started
Failed to GetObject: The name org.openbmc.ObjectMapper was not provided by any .service files
fanctl: Failed to get bus name for /org/openbmc/sensors/tach/fan0H
Read H 0 for fan 0
Failed to GetObject: The name org.openbmc.ObjectMapper was not provided by any .service files
fanctl: Failed to get bus name for /org/openbmc/sensors/tach/fan0L
Read H 0 for fan 0
fan0 speed: 0
Set fan 0 presence to 0
Failed to GetObject: The name org.openbmc.ObjectMapper was not provided by any .service files
fanctl: Failed to get bus name for /org/openbmc/inventory/system/chassis/fan0
fanctl: Failed to set fan present status
```

Could it be that org.openbmc.ObjectMapper.conf missing or incorrect? See similar:
https://github.com/openbmc/meta-phosphor/blob/master/common/recipes-phosphor/dbus/obmc-mapper/org.openbmc.ObjectMapper.conf
I've found the issue that says how to fix this path: https://github.com/openbmc/openbmc/issues/837
, but I'm not sure if this is right, because tried and it doesn't work still. Maybe other issue too linked from 837: 
https://github.com/openbmc/openbmc/issues/835
, but I tried fixing it to "xyz.openbmc_project.ObjectMapper" and still doesn't work:

```
root@barreleye:~# ~/fan_control.exe -f 6
service_name xyz.openbmc_project.ObjectMapper object_name /org/openbmc/ObjectMapper intf_name xyz.openbmc_project.ObjectMapper
Failed to GetObject: Method "GetObject" with signature "s" on interface "xyz.openbmc_project.ObjectMapper" doesn't exist

```

Is it a bug?
How to fix it and who owns fan_control.c and how it works, when "fan_update_present(info)" should be called ?

Looking to the "obmcutil getsensors" (below), all the required values exists and fan0L/fan0H > 0 (so presence should be True):

```
   "/org/openbmc/sensors/tach/fan0H": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon3/fan6_input", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 12385, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    }, 
    "/org/openbmc/sensors/tach/fan0L": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon3/fan5_input", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 9854, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    }, 
    "/org/openbmc/sensors/tach/fan1H": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon3/fan4_input", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 11250, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    }, 
    "/org/openbmc/sensors/tach/fan1L": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon3/fan3_input", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 9926, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    }, 
    "/org/openbmc/sensors/tach/fan2H": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon3/fan2_input", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 11538, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    },

```

And Present:False for fans:

```
    "/org/openbmc/inventory/system/chassis": {
      "fault": "False", 
      "fru_type": "SYSTEM", 
      "is_fru": 1, 
      "present": "True", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan0": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    }, 
    "/org/openbmc/inventory/system/chassis/fan1": {
      "fault": "False", 
      "fru_type": "FAN", 
      "is_fru": 1, 
      "present": "False", 
      "version": ""
    },

```

Other fake fan values==127, why? 

```
    "/org/openbmc/sensors/speed/fan0": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon2/pwm1", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 127, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    }, 
    "/org/openbmc/sensors/speed/fan1": {
      "critical_lower": 0, 
      "critical_upper": 0, 
      "emergency_enabled": 0, 
      "error": 0, 
      "filename": "/sys/class/hwmon/hwmon2/pwm2", 
      "offset": 0, 
      "scale": 1, 
      "threshold_state": "NORMAL", 
      "thresholds_enabled": 0, 
      "units": "", 
      "value": 127, 
      "warning_lower": 0, 
      "warning_upper": 0, 
      "worst_threshold_state": "NORMAL"
    },

```
-- why value==127?
Probably the answer somewhere in document
https://www.nuvoton.com/resource-files/NCT7904D_Datasheet_V1.44.pdf
, read PWM* there, but I don't understand it yet. Looks like this is a registers for "smart fan control" using PWM values, but in Barreleye it is being read as Fan/speed instead.

root@barreleye:~# cat /sys/class/hwmon/hwmon2/pwm2
127

Is it a bug inside hwmon
work/armv5e-openbmc-linux-gnueabi/linux-libc-headers/4.4-r0/linux-4.4/drivers/hwmon/hwmon.c
?

For me it looks like device "nct7904"/pwm being read incorrectly as fan, but the value 127 itself could be correct for pwm (then not a bug inside hwmon/drivers).
Is it related to the fact that no "/org/openbmc/sensors/speed/fan1", but "/org/openbmc/sensors/tach/fan0H" and "/org/openbmc/sensors/tach/fan0L" instead in someway, and logic with pwm incomplete and buggy still then?

root@barreleye:~# cat /sys/class/hwmon/hwmon2/device/name 
nct7904
...

For the fan0H the correct value is in fan1_input, but pwmX still == 127:
```

root@barreleye:~# cat /sys/class/hwmon/hwmon3/fan1_input 
9782
root@barreleye:~# 
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm        
pwm1         pwm1_enable  pwm2         pwm2_enable  pwm3         pwm3_enable  pwm4         pwm4_enable  
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm1
127
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm2
127
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm3
127
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm4
127
root@barreleye:~# cat /sys/class/hwmon/hwmon3/pwm1_enable 
1
root@barreleye:~# cat /sys/class/hwmon/hwmon3/device/name 
nct7904
```

Also, let me note that there are two values for each fan in Barreleye (L and H) and the answer why was in sources codes comment:

```
build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-control-fan/1.0-r1/git/fanctl/fan_control.c :

203 /*
 * FAN_TACH_OFFSET is specific to Barreleye.
 * Barreleye uses NTC7904D HW Monitor as Fan tachometoer.
 * The 13-bit FANIN value is made up Higer part: [12:5],
 * and Lower part: [4:0], which are read from two sensors.
 * see: https://www.nuvoton.com/resource-files/NCT7904D_Datasheet_V1.44.pdf
 */

```

geissonator, 2017-02-23:
Hi, could you please provide a "cat /ets/os-release" so we can know obmc code level?

yadron, 2017-02-28:
It is both for current for previous 1.99.1-dirty:

checked for current (everything the same):

cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-dirty"
VERSION_ID="v1.99.3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-dirty"
BUILD_ID="v1.99.3"

originally written for prev.previous:

cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.1-dirty"
VERSION_ID="v1.99.1-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.1-dirty"
BUILD_ID="v1.99.1"

geissonator, 2017-02-28:
@msbarth I know fan control function is just ramping up for us right now.  Is the above information expected with where we are right now?  

@yadron - We're tracking our fan function under the https://github.com/openbmc/openbmc/issues/347 epic right now.  That's where Matt is working on presence, fan speed setting, and dynamic fan control.  It will be rolling in over the next 2 months.

bradbishop, 2017-02-28:
This is partially resolved with https://gerrit.openbmc-project.xyz/#/c/2701/...

I forgot that we updated the mapper api too...so the call needs to be updated to pass the 'interfaces' argument.

yadron, 2017-02-28:
@geissonator , got the link to #347 

I've just described the FANs topic for Barreleye at #1218 as I see it now (for example, dynamic fan control maybe not required at all for Barreleye).

Thanks for answering both here or in other related issues.

gtmills, 2018-05-23:
@yadron Still an issue or can this be closed?

gtmills, 2018-07-18:
No updates in over a year, Barreleye is no longer a supported machine, and this is a very old build. Closing. Please reopen if still an issue. Thanks!

closed 2018-07-18 by gtmills