openbmc/openbmc #1176: iio to hwmon bridge driver

rfrandse, 2017-02-21:


bradbishop, 2017-03-02:
This seems to work...

```root@witherspoon:/sys/class/hwmon/hwmon5# pwd
/sys/class/hwmon/hwmon5
root@witherspoon:/sys/class/hwmon/hwmon5# realpath of_node
/sys/firmware/devicetree/base/iio-hwmon
root@witherspoon:/sys/class/hwmon/hwmon5# realpath device
/sys/devices/platform/iio-hwmon
root@witherspoon:/sys/class/hwmon/hwmon5# cat temp1_input
24070

bradbishop, 2017-03-02:
enabled in DT looks like this:

```diff --git a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts b/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
index e15bd25..8a3ec5b 100644
--- a/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
+++ b/arch/arm/boot/dts/aspeed-bmc-opp-witherspoon.dts
@@ -58,6 +58,11 @@
                enable-gpios = <&gpio ASPEED_GPIO(D, 0) GPIO_ACTIVE_HIGH>;
                trans-gpios = <&gpio ASPEED_GPIO(R, 2) GPIO_ACTIVE_HIGH>;
        };
+
+       iio-hwmon {
+               compatible = "iio-hwmon";
+               io-channels = <&bmp 1>;
+       };
 };

 &fmc {
@@ -160,9 +165,10 @@
         *    Regs 3.3VS, 1.1VCS, 5VCS, 1.8V
         */

-       bmp180@77 {
-               compatible = "bosch,bmp180";
+       bmp: bmp180@77 {
+               compatible = "bosch,bmp280";
                reg = <0x77>;
+               #io-channel-cells = <1>;
        };

        max31785@52 {

bradbishop, 2017-03-02:
finally need this in defconfig:

```CONFIG_SENSORS_IIO_HWMON=y```

bradbishop, 2017-03-02:
The only snag I see at this point is how to map the iio-hwmon device back to actual bus device:

```root@witherspoon:/sys/class/hwmon/hwmon5# pwd
/sys/class/hwmon/hwmon5
root@witherspoon:/sys/class/hwmon/hwmon5# ls -als
total 0
     0 drwxr-xr-x    2 root     root             0 Feb 26 07:49 .
     0 drwxr-xr-x    3 root     root             0 Feb 26 07:49 ..
     0 lrwxrwxrwx    1 root     root             0 Feb 26 07:50 device -> ../../../iio-hwmon
     0 -r--r--r--    1 root     root          4096 Feb 26 07:50 name
     0 lrwxrwxrwx    1 root     root             0 Feb 26 07:50 of_node -> ../../../../../firmware/devicetree/base/iio-hwmon
     0 lrwxrwxrwx    1 root     root             0 Feb 26 07:49 subsystem -> ../../../../../class/hwmon
     0 -r--r--r--    1 root     root          4096 Feb 26 07:50 temp1_input
     0 -rw-r--r--    1 root     root          4096 Feb 26 07:49 uevent
```
There doesn't seem to be any to get from

/sys/class/hwmon/hwmonX

back to a real device in the /sys/devices tree.

mdmillerii, 2017-03-02:
The io-channel property should be a phandle and an index ... for each pair ... so that should be able to be followed, if a bit involved.

bradbishop, 2017-03-02:
thx Milton!

bradbishop, 2017-03-28:
https://gerrit.openbmc-project.xyz/#/c/3331/
https://patchwork.ozlabs.org/patch/744250/
https://patchwork.ozlabs.org/patch/744249/

closed 2017-03-30 by bradbishop