openbmc/openbmc #1177: vdd reg temps

rfrandse, 2017-02-21:


geissonator, 2017-02-23:
We need an hwmon driver for this i2c device.  The main use case here is to read the temperature from the device.

shenki, 2017-02-27:
What device are we referring to here?

jk-ozlabs, 2017-02-27:
What's the actual sensor device?

spinler, 2017-02-27:
It's an IR35221.  The temp is obtained with the 0x8D READ_TEMPERATURE_1 PMBUS command

sammj, 2017-03-23:
I've coincidentally started writing the `IR35221` driver already (along with the `IR3806x`), I'll jump back on that and finish it up.

sammj, 2017-04-21:
http://patchwork.ozlabs.org/patch/753119/

shenki, 2017-05-01:
v2: http://patchwork.ozlabs.org/patch/756975/

shenki, 2017-05-04:
This has been accepted by the upstream maintainer. I have merged it into the dev-4.10 tree.

Once the device tree and config changes are merged we can close this issue.

bradbishop, 2017-05-19:
Moving to @cbostic to get the DT and kernel config updated.

bradbishop, 2017-05-19:
Adding @cbostic task list from #1612:

- [x] Review device driver and understand all device tree requirements, bus type etc...
- [x] Obtain card schematics including details for IR35221.
- [ ] Identify i2c bus and address details for Witherspoon
- [ ] Code and integrate changes to dts file.
- [ ] Load kernel on test system and verify device is identified and probed

cbostic, 2017-05-19:
Enabled the driver and the ir35221's are showing up as expected
```
root@witherspoon:/sys/bus/i2c/drivers/ir35221# ls
4-0070  4-0071  5-0070  5-0071  bind    uevent  unbind
```

bradbishop, 2017-06-01:
The scope of this issue was (perhaps repurposed) only to make the DT and config updates, which is complete.   We have #1702 to track unbinding prior to use with i2cset.  Closing...

closed 2017-06-01 by bradbishop