openbmc/openbmc #1178: openFSI: Add options to obmcutil to enable cronus

yadlapati, 2017-02-22:


rfrandse, 2017-02-22:
Apparently needed by Friday. 2/24
Andrew: The actual work is a bit vague right now from Dean and Norm, but I think it’s doing that openFSI unbind operation, and maybe a gpio write.

spinler, 2017-02-22:
```
echo "fsi-master" > /sys/bus/platform/drivers/fsi-master-gpio/unbind
gpioutil -p d0 -d out -v 1
```

fortunately, the FSI_ENABLE GPIO is D0 on all systems

williamspatrick, 2017-02-22:
I think the preference would be that the FSI driver ensures the mux is in the right state when we bind / unbind.  Then you just run unbind and the mux is set to "Cronus mode" automatically.  If we can get that done, I don't see any need for a utility.  We have set an overall direction to not write little helper utilities for things that are pretty straight-forward.

shenki, 2017-02-23:
I thought we could have a systemd target that ensures

1. Power is applied to the CPUs
1. The FSI driver is unbound
1. The mux is set to "external fsi"
1. The system can't reach the host-boot state until this mode is disabled (so we don't try to power on and fail due to FSI not being available)

@geissonator Is this realistic?

ozbenh, 2017-02-23:
Patrick, while I agree with you on the driver cleaning up after itself, I rather strongly disagree on doing unbinds by hand being straighforward. We should definitely have an obmcutil option to turn on cronus mode

ozbenh, 2017-02-23:
To followup on @shenki , I also thought all along a systemd target would be a good idea. It would allow cronus-boots when used for poweron, live debug when "switched" from poweron to that mode, and would also prevent all the checkstop handling gunk. The FSI switch is only one part of the picture here.

ozbenh, 2017-02-23:
In fact it's a bit more complicated ... there are two things that are sort-of orthogonal:

 - The debug mode (disable reboot-on-xstop, don't auto trigger boot in CFAM so you can do cronus boots, etc...)
 - The FSI mode (external or internal)

IE, one might want to run in debug mode with a BMC-local croserver for example.

I still think that from a user perspective it should be a single systemd "debug" target. As for which FSI interface is to be used when in debug mode, well that could be driven via some config setting. In any case, I agree the actual cleanup should be done by the FSI driver on ubind, the question is whether to unbind or not.

geissonator, 2017-02-27:
@cbostic has agreed that when we unbind the FSI driver, it will switch the mux over for cronus.  A systemd debug target that does the unbind, disables auto reboot, ensures we wont try to start the host (i.e. conflict with host start target), and potentially powers on the chassis does seem reasonable though.

cbostic, 2017-02-28:
I assume that the FSI driver is to switch mux to 0b'0' for switch to cronus control?  FSI driver initializes to 0b'1' during setup.

paulgrobertson, 2017-02-28:
FYI.  Current FSP box Cronus users are broken if systems are flashed up the latest BMC code.   I have attempted this power on procedure,  but sometime during cronus IPL, or later, f/w powers off system unexpectedly.  Need a work around,  or good sol'n.

Cronus power on for FSP box
------------------------------
1. power on
obmcutil poweron

2. wait a minute

3. issue these command to allow access from FSP box:
echo "fsi-master" > /sys/bus/platform/drivers/fsi-master-gpio/unbind
pdbg -b fsi -d p9w -p0 getcfam 0xc09
gpioutil -p d0 -d out -v 1

geissonator, 2017-03-01:
@yadlapati had this working Paul.  Those look like the steps he was using.  Although, maybe not the obmcutil poweron, that will boot the host, if you just want cec power, you can use the "on" script or the systemd command.

williamspatrick, 2017-03-01:
Please stop recommending arbitrary scripts. We already have a systemd target for turning on the power supplies and that is what people should use. 

We already do not "auto reboot" when someone just turns on the power supplies. What else is part of "debug mode"?  I really want to get away from that terminology. 

1. Power supply on + FSI mux for Cronus
2. Disable auto reboot and watchdogs.

These are two different orthogonal host debug-oriented use cases, correct?

I suspect there is another case of doing a normal boot with case 2 and then switching the mux for the actual debug. For this reason I see there being 3 different "utilities".

paulgrobertson, 2017-03-01:
Patrick, I want to use a supported f/w sol'n to power on and cronus IPL (external FSP box).  What, exactly is the sol'n?  Step by step.  Cronus users are broken with new f/w without a crisp how to.

williamspatrick, 2017-03-01:
My understanding is there are (at least) two different use-cases:

```
Begin:
1. Power on chassis.
2. Disable OpenFSI / Enable external Cronus.

Later:
1. Power off chassis.
```

```
Begin:
1. Boot system normally with watchdog / auto-reboot disabled.

Later:
1. Disable OpenFSI / Enable external Cronus.

Even later:
1. Power off system and/or reboot.
```

Do we have any others?  @sannerd @paulgrobertson @ozbenh @wghoffa

williamspatrick, 2017-03-01:
@paulgrobertson - Understand we need it documented.  We're first trying to figure out what is really needed.  Different users have different use cases.

williamspatrick, 2017-03-01:
@wghoffa mentioned another use-case through internal channels.

```
Begin:
1. Power on chassis.
2. Connect Cronus to OpenFSI.

Later:
1. Power off chassis.
```

sannerd, 2017-03-02:
I think you have most of the use cases:

1) power on, do not start IPL.  Use external Cronus
      a) disable watchdog, xstop reboot
      b) set FSI clock to external
      c) switch FSI mux to external

2) Power on, do not start IPL, Use softFSI Cronus
     a) disable watchdog, xstop reboot

3) Power on, startIPL, but enter debug mode where 
     a) disable watchdog, xstop reboot

4) At any time connect Cronus via internal
5) At any time connect Cronus via external
     a) set FSI clock to external
      b) switch FSI mux to external

Since there are common actions I guess I see it boiling down to these basic commands which can be mixed and matched as the user desires:
**Enter/exit debug mode** -- handles toggling of auto start, disabling watchdog, etc
**Enter/exit external cronus mode** -- handles toggling between external cronus and internal
        Note that the external cronus mode will prevent OCC comm (eventually)
**poweron, poweroff** -- already present, but need to take into account the debug/external cronus modes

The last point may be open to debate -- I think the debug modes should ride through the power cycles, but others may disagree?

geissonator, 2017-03-16:
https://gerrit.openbmc-project.xyz/#/c/2927/

closed 2017-03-16 by geissonator

spinler, 2017-03-16:
To summarize how this will work:

systemctl start obmc-power-chassis-on@0.target
* Just power on a system and not start the SBE

systemctl stop host-failure-reboots.service
* Turns off automatic reboots on checkstops and watchdog timeouts
* reboots get enabled again on a BMC reset
* Enable again by starting the service

systemctl start fsi-disable.service
* Unbinds the FSI driver, and switches the GPIOs for external cronus use
* Do this anytime after power on
* Will need to do this each power on, as the act of powering on requires FSI for the VCS workaround

systemctl start fsi-enable.service
* Loads the BMC's FSI Driver again
* Don't need to explicitly run this for just a power cycle