openbmc/openbmc #1179: Software Version Service process hung ? [ Witherspoon v1.99.2-100-g443bb00 ]

gkeishin, 2017-02-22:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-100"
VERSION_ID="v1.99.2-100-g443bb00-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-100"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170222035820
root@witherspoon:~#

```

```
root@witherspoon:~# systemctl status xyz.openbmc_project.Software.Version.service | cat
● xyz.openbmc_project.Software.Version.service - Phosphor Version Software Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Software.Version.service; enabled; vendor preset: enabled)
   Active: activating (start) since Thu 2017-02-02 17:23:01 UTC; 14s ago
 Main PID: 887 (phosphor-versio)
   CGroup: /system.slice/xyz.openbmc_project.Software.Version.service
           └─887 /usr/sbin/phosphor-version-software-manager

Feb 02 17:23:01 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Service hold-off time over, scheduling restart.
Feb 02 17:23:01 witherspoon systemd[1]: Stopped Phosphor Version Software Manager.
Feb 02 17:23:01 witherspoon systemd[1]: Starting Phosphor Version Software Manager
```

- [x] Journald 
[journald_v1.99.2-100-g443bb00_20170222035820_Software.Version.service.txt](https://github.com/openbmc/openbmc/files/793976/journald_v1.99.2-100-g443bb00_20170222035820_Software.Version.service.txt)

gkeishin, 2017-02-22:
@gtmills is taking a look at the system in fail state.. Thanks

gkeishin, 2017-02-22:
This shows up during Poweron / reboot..  
```
Feb 02 18:11:10 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Start operation timed out. Terminating.
Feb 02 18:11:10 witherspoon systemd[1]: Failed to start Phosphor Version Software Manager.
Feb 02 18:11:10 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Unit entered failed state.
Feb 02 18:11:10 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Failed with result 'timeout'.
Feb 02 18:11:11 witherspoon systemd[1]: xyz.openbmc_project.Software.Version.service: Service hold-off time over, scheduling restart.
Feb 02 18:11:11 witherspoon systemd[1]: Stopped Phosphor Version Software Manager.
Feb 02 18:11:11 witherspoon systemd[1]: Starting Phosphor Version Software Manager..
```

gtmills, 2017-02-22:
https://gerrit.openbmc-project.xyz/#/c/2623/ should fix.

closed 2017-02-22 by williamspatrick