openbmc/openbmc #1180: Need pdbg CFAM workaround for openFSI

geissonator, 2017-02-22:
We've got a pretty little hack for a certain CFAM address in https://github.com/openbmc/openbmc/blob/d3661bd14892437925a09c37cf99393f97eaf0ad/meta-openbmc-machines/meta-openpower/common/recipes-bsp/pdbg/pdbg/0001-CFAM-reg-0x2918-access-hack.patch

We now need to port this patch over to the new openFSI interfaces in pdbg (https://github.com/open-power/pdbg/blob/master/libpdbg/kernel.c#L103)

apopple, 2017-02-22:
I think it would be best if we also added a warning when someone tries to do a get/putcfam 0x900 as there are currently scripts all over the place that need updating.

We can revert the warning later.

chinaridinesh, 2017-03-15:
Updated pdbg.bb with latest commit ID and changed vcs_workaround.sh file  updating fsi slave address to 0x840.
https://gerrit.openbmc-project.xyz/#/c/3003/

bradbishop, 2017-03-27:
I'm told this is fixed here (@chinaridinesh):

https://gerrit.openbmc-project.xyz/#/c/3049/

closed 2017-03-27 by bradbishop