openbmc/openbmc #1181: Auto reboot for Quiesce Mode  [ Witherspoon MFG Test ]

gkeishin, 2017-02-22:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-100"
VERSION_ID="v1.99.2-100-g443bb00-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-100"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170222035820
root@witherspoon:~#
```

**OS checkstop ... Auto reboot enabled, Retries for Power On**

```
Feb 02 18:04:08 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:04:08 witherspoon kernel[766]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:05:20 witherspoon chassis_control.py[769]: Watchdog Error, Going to quiesce
Feb 02 18:05:20 witherspoon phosphor-host-state-manager[831]: Auto reboot enabled. Beginning reboot...
Feb 02 18:05:21 witherspoon phosphor-host-state-manager[831]: Auto reboot enabled. Beginning reboot...
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:05:26 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:05:26 witherspoon kernel[766]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:06:36 witherspoon chassis_control.py[769]: Watchdog Error, Going to quiesce
Feb 02 18:06:36 witherspoon phosphor-host-state-manager[831]: Auto reboot enabled. Beginning reboot...
Feb 02 18:06:37 witherspoon phosphor-host-state-manager[831]: Auto reboot enabled. Beginning reboot...
ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:06:41 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Feb 02 18:06:41 witherspoon kernel[766]: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
```

**And finally hits Quiesce  mode**

```
Feb 02 18:07:32 witherspoon systemd[1]: Reached target Quiesce Target.
Feb 02 18:07:32 witherspoon phosphor-host-state-manager[831]: Error in mapper call
``` 

**This seems to be happening more than expected before going to Quiesce  as per @mtritz** 

- [x]  Journald 
[journald_v1.99.2-100-g443bb00_20170222035820_Autoboot.txt](https://github.com/openbmc/openbmc/files/794336/journald_v1.99.2-100-g443bb00_20170222035820_Autoboot.txt)

gkeishin, 2017-02-22:
Auto reboot seems to vary from test to test as well as per observation, one example is above.. so if there is a counter for it where if it retries Max's out , it should stop.. ?

geissonator, 2017-02-22:
I think we still need to implement the max count on reboots.  I believe the mapper call fail is related to #1145

closed 2017-03-20 by williamspatrick