openbmc/openbmc #1183: Temperatures from OCC

rfrandse, 2017-02-22:


bradbishop, 2017-05-05:
Use this story to add phospor-hwmon config files for the OCC.

spinler, 2017-05-15:
Current open question:
- what path do I use for the config file under /etc/default/obmc/hwmon?  I2C hwmon files look like:  `./ahb/apb/i2c@1e78a000/i2c-bus@400/tmp275@4a.conf`

bradbishop, 2017-05-16:
Had a meeting with @jk and @eddiejames tonight about this.  The direction coming out of that was that we'll use the same setup as the existing P8 OCC driver.  The general idea is that the IPMI sensor number will go in the label attribute:

temp1_label = 121

The config files will have to map this number to a name like core0.

One open question I have is whether or not the index will be stable.  As in, for a given system, can userspace assume that temp1 will always have the same label?  If userspace cannot..we'll have to implement a level of indirection to go from the label -> input in phosphor-hwmon that we currently don't have.

bradbishop, 2017-05-16:
@eddiejames when you know what its going to be, can you post in here what the the paths will look like under /sys/devices and /sys/firmware?

jk, 2017-05-16:
I'm pretty sure I had no meeting with @bradbishop.

bradbishop, 2017-05-18:
Need to know what of_node and device in /sys/class/hwmon/hwmonX will point at for occ devices....
@eddiejames ^

eddiejames, 2017-05-18:
/sys/firmware/devicetree/base/gpio-fsi/cfam@0,0/sbefifo@2400/occ@1/occ-hwmon@1
/sys/devices/platform/gpio-fsi/fsi0/slave\@00\:00/00\:00\:00\:06/sbefifo1-dev0/occ1-dev0/hwmon/hwmon6/

for second proc:
/sys/firmware/devicetree/base/gpio-fsi/cfam@1,0/sbefifo@2400/occ@2/occ-hwmon@2
/sys/devices/platform/gpio-fsi/fsi0/slave\@00\:00/00\:00\:00\:0a/fsi1/slave\@01\:00/01\:00\:00\:06/sbefifo1-dev0/occ1-dev0/hwmon/hwmon6/

geissonator, 2017-05-23:
Questions and answers from todays meeting
- What specifically does this track?  Just OCC sensors or VDD and Ambient?
-- It tracks all of them
-- VDD is for processors (ir chip, cbostic getting into device tree)
-- ambient is in iio, Brandon has story to make this work (bmp280 (non ga sensor)) DTS310 is coming (will need a different config) - cards coming middle of May
-- Put conf file in for DTS310, wont be testable but we'll have it and can tweak if needed when hw shows up
- Can't guarantee instance numbers for occ sensors?
-- #1633 is where this gets addressed, review it and see if more questions

geissonator, 2017-05-23:
Task List:
- [x] Add in DPS310 config file - https://gerrit.openbmc-project.xyz/#/c/4338/
- [ ] Add in IR35221 config file (dependency on #1177 - no hwmon yet)
- [ ] Add in OCC config file (this will require integration with #1633) 
- [ ] Verify all required sensors are available via dbus objects

rfrandse, 2017-05-30:
https://gerrit.openbmc-project.xyz/4440 Add in OCC hwmon config file
Resolves #1183 hwmon config file for OCC, VDD, Ambient

closed 2017-06-12 by williamspatrick