openbmc/openbmc #1190: [palmetto VERSION_ID="v1.99.2-94-g390c080-dirty"] Palmetto Reboot Failure

swe12345, 2017-02-23:
Issue: Palmetto Reboot Failure

Log Info from serial port console:
```
palmetto login: jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC 011f9179 != calculated CRC 912c3e96 for node at 0022000c
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!
jffs2: Node CRC ffffffff != calculated CRC f09e7845 for node at 0015852c
jffs2: read of old metadata failed in jffs2_garbage_collect_metadata(): -5
jffs2: Error garbage collecting node at 0015852c!

```
System Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-94"
VERSION_ID="v1.99.2-94-g390c080-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-94"
BUILD_ID="v1.99.2"

Please find attached the dmesg and the journal logs
[dmesg.txt](https://github.com/openbmc/openbmc/files/795780/dmesg.txt)
[journal_palm309.txt](https://github.com/openbmc/openbmc/files/795782/journal_palm309.txt)

ratagupt, 2017-02-23:
Feb 22 07:07:19 palmetto kernel: SQUASHFS error: Unable to read fragment cache entry [693638]
Feb 22 07:07:19 palmetto kernel: SQUASHFS error: Unable to read page, block 693638, size 7b34
Feb 22 07:07:19 palmetto kernel: SQUASHFS error: Unable to read fragment cache entry [693638]
Feb 22 07:07:19 palmetto kernel: SQUASHFS error: Unable to read page, block 693638, size 7b34
Feb 22 07:07:19 palmetto kernel[683]: SQUASHFS error: Unable to read fragment cache entry [693638]
Feb 22 07:07:19 palmetto kernel[683]: SQUASHFS error: Unable to read page, block 693638, size 7b34
Feb 22 07:07:19 palmetto kernel[683]: SQUASHFS error: Unable to read fragment cache entry [693638]
Feb 22 07:07:19 palmetto kernel[683]: SQUASHFS error: Unable to read page, block 693638, size 7b34
Feb 22 07:07:20 palmetto systemd[1]: Started SSH Per-Connection Server (9.3.62.216:34260).

sivassrr, 2017-02-23:
yup, correct @ratagupt  Thats why I requested Sweta to open this defect.

geissonator, 2017-02-27:
We need more here guys, what was done leading up to the failure?  Please provide recreate instructions.  I assume just doing a "reboot" works fine in general on systems?  Must have been something special here.

mdmillerii, 2017-02-27:
looks like something happened and we lost access to the flash.  @cyrilbur-ibm ?

geissonator, 2017-10-17:
Please try and recreate

closed 2017-10-17 by geissonator