openbmc/openbmc #1191: phosphor-inventory-manager fails to build on x86-64

mine260309, 2017-02-23:
When MACHINE is set to `qemux86-64` the build of phosphor-inventory-manager fails with below log.
But it's OK with `qemux86` or p8/p9 machines.

Build Configuration:
```
BB_VERSION        = "1.30.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Ubuntu-14.04"
TARGET_SYS        = "x86_64-openbmc-linux"
MACHINE           = "qemux86-64"
DISTRO            = "openbmc-phosphor"
DISTRO_VERSION    = "0.1.0"
TUNE_FEATURES     = "m64 core2"
TARGET_FPU        = ""
meta
meta-poky
meta-oe
meta-networking
meta-python
meta-virtualization
meta-phosphor     = "master:2be34d27d1431dea6bc71c54b4271dd2567b829d"
```

```
| generated.cpp:581:30: error: invalid initialization of reference of type 'std::map<sdbusplus::message::details::string_wrapper<sdbusplus::message::details::object_path_type>, std::map<std::__
cxx11::basic_string<char>, std::map<std::__cxx11::basic_string<char>, mapbox::util::variant<bool, long int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >
 >&&' from expression of type '<brace-enclosed initializer list>'
|                              }))
|                               ^
| In file included from ../git/manager.hpp:10:0,
|                  from generated.cpp:2:
| ../git/functor.hpp:97:13: note: in passing argument 1 of 'auto phosphor::inventory::manager::functor::createObjects(std::map<sdbusplus::message::details::string_wrapper<sdbusplus::message::de
tails::object_path_type>, std::map<std::__cxx11::basic_string<char>, std::map<std::__cxx11::basic_string<char>, mapbox::util::variant<bool, long int, std::__cxx11::basic_string<char, std::char_
traits<char>, std::allocator<char> > > > > >&&)'
|  inline auto createObjects(
|              ^
| generated.cpp:582:18: error: no matching function for call to 'std::vector<std::function<void(sdbusplus::bus::bus&, phosphor::inventory::manager::Manager&)> >::vector(<brace-enclosed initiali
zer list>)'
|                  })),
...... // A lot of infos by gcc
```

bradbishop, 2017-02-27:
I'm wondering if https://gerrit.openbmc-project.xyz/#/c/2657/ fixes this.

mine260309, 2017-02-28:
Yes, this issue is fixed by the patch and it's now merged: 905226c5cd38887231572c5948ef16bed01904e9
Close the issue.

closed 2017-02-28 by mine260309