openbmc/openbmc #1192: Able configure broadcast address to bmc

prkatti1, 2017-02-23:
System is allowing to configure broadcast IP address which is not an unicast address.
System should not allow to configure brodcaste IP address to ethernet interface.
Try to configure x.x.x.255.

System info
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"

prkatti1, 2017-02-23:
@sivassrr opened this defect, although it is an valid address but it has special purpose to broadcast packets to LAN

ratagupt, 2017-02-23:
Yes,we have not made that check for broadcast address.

ratagupt, 2017-02-23:
first, second, third, fourth = [int(part) for part in ip_parts]
        if first == 0 and second == 0 and third == 0 and fourth == 0:
            return "Invalid" 	# "this" network disallowed
        if first == 169 and second == 254:
            return "Link Local"
        if first >= 224:
            return "Invalid"	# class D multicast and class E disallowed
        if first == 192 and second == 88 and third == 99:
return "Invalid"	# ipv6 relay

geissonator, 2017-02-23:
Is it the job of openbmc code to validate what people want to set for ethernet settings?  If they want to shoot themselves in the foot, why not just let them?  
@williamspatrick

geissonator, 2017-02-27:
Similar to the other issue in this area, openBMC code isn't going to write arbitrary rules like this.  It's up to the person doing the network programming to know what they're doing.

closed 2017-02-27 by geissonator