openbmc/openbmc #1193: Able configure Network ID to bmc

prkatti1, 2017-02-23:
We are able to assign network ID (ie xx.xx.xx.0)  ip address to bmc interface.
System should throw an error when we try to assign it.

System version info
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-60"
VERSION_ID="v1.99.2-60-gcb1cc2b-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-60"
BUILD_ID="v1.99.2"

prkatti1, 2017-02-23:
@sivassrr opened this defect, although it is an valid address but it has special purpose

prkatti1, 2017-02-23:
Used ifconfig command to configure this 
ifconfig eth0 x.x.x.0 255.255.255.0

bradbishop, 2017-02-25:
We are not hacking up the ifconfig command to prevent this.

If you are able to do this with a valid user interface like the REST API or over IPMI then go ahead and mention which interface you used and reopen.

closed 2017-02-25 by bradbishop