openbmc/openbmc #1194: Chassis and Host state services show incorrect states when targets exit

geissonator, 2017-02-23:
The chassis and host code monitors for certain targets to reach "JobRemoved" to know when a target has completed. This works great initially when the target is run. But later, when the target is stopped, we get this same event. This doesn't really hurt anything but it's confusing in the journal because you're stopping the target (i.e. powering down) and you see a journal log indicating we powered on. Later in the power down process, we get the correct journal entries but still confusing initially.

JoshDKing, 2017-03-02:
https://gerrit.openbmc-project.xyz/#/c/2805/
https://gerrit.openbmc-project.xyz/#/c/2804/1

JoshDKing, 2017-04-05:
There has been some suggested code design changes for the state manager code surrounding this story.  After discussing with Vishwa and Andrew, we feel that these design changes may be for the better.  Rather than apply all those lengthy changes to this story, we thought it be best to create a new story to address the repetitive code and design changes. (Story has yet to be created, but will provide link when story is created)

closed 2017-04-18 by rfrandse