openbmc/openbmc #1195: Support fake hwmon chip

rfrandse, 2017-02-23:


williamspatrick, 2017-02-23:
What does this mean?

bradbishop, 2017-02-23:
I wanted to have a configure option that allows you to point at:

/foo/fakesysfs/hwmon/hwmonx

instead of the real one.

Then we can actually run hwmon instances in qemu.

shenki, 2017-02-24:
If this is for qemu, a better solution would be to add a device that has a hwmon driver to the model.

@legoater @amboar any suggestions?

bradbishop, 2017-02-24:
For that to be "better" it would need to be the default qemu model that comes with Yocto, preferably both the arm and x86 models as I know those are two models most (userspace) devs run.

bradbishop, 2017-02-24:
Or, if we can figure out how to change the default model to something more appropriate to our distro that would work too.

amboar, 2017-02-24:
@bradbishop @kc8apf opened an issue recently about migrating our yocto setup to use the aspeed qemu models. Yocto have improved their handling of qemu, so it should be a matter of writing some configuration files.

shenki, 2017-02-24:
Yeah. We should get them using the machine that we've spent the past year working on

legoater, 2017-02-24:
And we can write more model for sensors. These are easy to write when the logic is just a question of getting and setting values from the QEMU monitor.  see https://github.com/openbmc/qemu/commit/1fa5a71977c8806fa374600af27bee4f4b240bf6 for that.

amboar, 2019-04-02:
Closing this, as it's morphed into two separate problems: Implementing hwmon models and integrating our machines into yocto.

closed 2019-04-02 by amboar