openbmc/openbmc #1196: hwmon for infinion DPS 310 device driver.

rfrandse, 2017-02-23:


spinler, 2017-02-24:
The Infineon DPS310 is the new ambient sensor on the Teak fan riser card.  It is showing up in the ITC level of the cards currently scheduled for the first or second week in May.  It is replacing the BMP280 that has too near an EOL.

I have a datasheet if anyone needs it.   Also, I have a contact who can have a part reworked onto a Teak in Rochester or Austin.

It is also a pressure sensor, but we don't have a hard requirement to display altitude, only ambient temperature (as an input to fan control).  The BMP280 had an IIO subsystem driver instead of hwmon, probably because it read pressure.  We will need it to show up in hwmon, but I believe the iio-hwmon driver can bridge that if necessary.

jk-ozlabs, 2017-02-27:
Datasheet is at http://www.infineon.com/dgdl/Infineon-DPS310-DS-v01_00-EN.pdf?fileId=5546d462576f34750157750826c42242

shenki, 2017-02-27:
@spinler This will be i2c attached?

spinler, 2017-02-27:
@shenki Yea, it's I2C.

shenki, 2017-02-28:
@spinler okay. I have a driver to test once we have hardware ready. Let me know how you go with the reworked system

shenki, 2017-04-27:
A working driver was posted here:

 http://patchwork.ozlabs.org/patch/755499/

Upstream suggested we use the IIO framework instead, so it will be reworked to use that API.

shenki, 2017-05-04:
The iio version was posted today:

 https://patchwork.kernel.org/patch/9710865/

bjwyman, 2017-08-21:
Isn't this done, or is it waiting on some upstream process?

amboar, 2017-08-23:
Well, we have it in openbmc/linux dev-4.10: https://github.com/openbmc/linux/blob/dev-4.10/drivers/iio/pressure/dps310.c

Upstream were asking @shenki to implement the pressure sensor side of it, but it's probably closed for OpenBMC purposes?

closed 2017-08-23 by spinler