openbmc/openbmc #1200: Inventory shows  8 DIMMs when 16 DIMMs available [ Witherspoon MFG]

gkeishin, 2017-02-24:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-133"
VERSION_ID="v1.99.2-133-gfba842e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-133"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170224160133
root@witherspoon:~#
```

- [x] Journal log 
[journald_v1.99.2-133-gfba842e_20170224160133_inventory.txt](https://github.com/openbmc/openbmc/files/799971/journald_v1.99.2-133-gfba842e_20170224160133_inventory.txt)

```
{
  "data": {
    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
      "BuildDate": "", 
      "Manufacturer": "0000000000000000", 
      "Model": "", 
      "PartNumber": "0000000", 
      "Present": 0, 
      "PrettyName": "SYSTEM PLANAR   ", 
      "SerialNumber": "000000000000"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01HL322", 
      "Present": 1, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "YA3933741574", 
      "Version": "10"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "IBM", 
      "Model": "", 
      "PartNumber": "01HL322", 
      "Present": 1, 
      "PrettyName": "PROCESSOR MODULE", 
      "SerialNumber": "YA3933755254", 
      "Version": "10"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm0": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0xce80", 
      "Model": "M393A1G40EB1-CRC    ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x0300cf4f", 
      "Version": "0x00"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm1": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb2f6", 
      "Version": "0x31"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm2": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0xce80", 
      "Model": "M393A1G40EB1-CRC    ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x0300cf5d", 
      "Version": "0x00"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm3": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb158", 
      "Version": "0x31"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm4": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb147", 
      "Version": "0x31"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm5": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb2f1", 
      "Version": "0x31"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm6": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb152", 
      "Version": "0x31"
    }, 
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm7": {
      "BuildDate": "", 
      "Cached": 0, 
      "FieldReplaceable": 1, 
      "Manufacturer": "0x2c80", 
      "Model": "18ASF1G72PZ-2G3B1   ", 
      "PartNumber": "", 
      "Present": 1, 
      "PrettyName": "0x0c", 
      "SerialNumber": "0x13afb2f2", 
      "Version": "0x31"
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

williamspatrick, 2017-02-25:
I think this was determined to be a host firmware issue, right?

geissonator, 2017-02-25:
Yean, please open issue on hostboot repo (https://github.com/open-power/hostboot)

closed 2017-02-25 by geissonator

gkeishin, 2017-02-26:
Done.. Opened https://github.com/open-power/hostboot/issues/89