openbmc/openbmc #1201: Host ipmid core dump [ Witherspon MFG ]

gkeishin, 2017-02-24:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-133"
VERSION_ID="v1.99.2-133-gfba842e-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-133"
BUILD_ID="v1.99.2"
root@witherspoon:~# cat /etc/version
20170224160133
root@witherspoon:~#

```

```
root@witherspoon:~# systemctl status org.openbmc.HostServices.service | cat
● org.openbmc.HostServices.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/org.openbmc.HostServices.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-02-02 18:01:48 UTC; 6min ago
 Main PID: 1772 (ipmid)
   CGroup: /system.slice/org.openbmc.HostServices.service
           └─1772 /usr/sbin/ipmid

Feb 02 18:05:10 witherspoon ipmid[1772]: Dest: :1.7, Path: /org/openbmc/HostIpmi/1
Feb 02 18:05:14 witherspoon ipmid[1772]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x48]
Feb 02 18:05:14 witherspoon ipmid[1772]: Failed to call the method: Input/output error
Feb 02 18:05:14 witherspoon ipmid[1772]: Dest: :1.7, Path: /org/openbmc/HostIpmi/1
Feb 02 18:05:18 witherspoon ipmid[1772]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x49]
Feb 02 18:05:18 witherspoon ipmid[1772]: Failed to call the method: Input/output error
Feb 02 18:05:18 witherspoon ipmid[1772]: Dest: :1.7, Path: /org/openbmc/HostIpmi/1
Feb 02 18:05:22 witherspoon ipmid[1772]: ERROR:[0xFF] handling NetFn:[0xA], Cmd:[0x48]
Feb 02 18:05:22 witherspoon ipmid[1772]: Failed to call the method: Input/output error
Feb 02 18:05:22 witherspoon ipmid[1772]: Dest: :1.7, Path: /org/openbmc/HostIpmi/1
root@witherspoon:~#
```

- [x]  Core dump file  
[core_ipmid.1335.gz](https://github.com/openbmc/openbmc/files/800013/core_ipmid.1335.gz)

- [x] Journald 
[journald_v1.99.2-133-gfba842e_20170224160133_ipmid_core.txt](https://github.com/openbmc/openbmc/files/800014/journald_v1.99.2-133-gfba842e_20170224160133_ipmid_core.txt)

anoo1, 2017-02-24:
```
$ $GDB ipmid core_ipmid.1335
warning: exec file is newer than core file.
[New LWP 1335]
Cannot access memory at address 0x49c56e20
Cannot access memory at address 0x49c56e1c
Core was generated by `/usr/sbin/ipmid'.
Program terminated with signal SIGABRT, Aborted.
#0  0x49bed0d8 in ?? ()
(gdb)
```

geissonator, 2017-02-24:
Seems to recreate well on w42, probably going to need some printf's.

rfrandse, 2017-03-01:
Fixed: 
https://gerrit.openbmc-project.xyz/#/c/2728/

closed 2017-03-01 by rfrandse