openbmc/openbmc #1202: OpenFSI  Hub is generating duplicate sysfs file during scan

cbostic, 2017-02-24:
During scan hub is attempting to create a file with the same name as a previously created one:

Feb 02 17:23:54 witherspoon kernel[760]: bus[0000001a] created
Feb 02 17:23:56 witherspoon kernel: bus[0000001b] created
Feb 02 17:23:56 witherspoon kernel: bus[0000001c] created
Feb 02 17:23:56 witherspoon kernel: bus[0000001d] created
Feb 02 17:23:56 witherspoon kernel: RESUME   eng[00f5112c]
Feb 02 17:23:56 witherspoon kernel: iic_boe_reset
Feb 02 17:23:56 witherspoon kernel: iic_boe_eng_init: P8/Z7PLUS mode
Feb 02 17:23:56 witherspoon kernel: clean port busy port_busy[0001ffff]
Feb 02 17:23:56 witherspoon kernel: iic_boe_eng_init: P8/Z7PLUS mode
Feb 02 17:23:56 witherspoon kernel:  slave@01:00: engine[8]: type 22, version 1, addr 2400 size 400
Feb 02 17:23:56 witherspoon kernel:  slave@01:00: engine[9]: type 20, version 2, addr 2800 size 800
Feb 02 17:23:56 witherspoon kernel:  slave@01:00: engine[11]: type 1a, version 1, addr 3000 size 400
Feb 02 17:23:56 witherspoon kernel: ------------[ cut here ]------------
Feb 02 17:23:56 witherspoon kernel: WARNING: CPU: 0 PID: 1150 at /home/fspcibld/ubuntu-witherspoon-custom/openbmc/build/tmp/work-shared/witherspoon/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
Feb 02 17:23:56 witherspoon kernel: sysfs: cannot create duplicate filename '/devices/hub@00'
Feb 02 17:23:56 witherspoon kernel: CPU: 0 PID: 1150 Comm: pdbg Not tainted 4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Feb 02 17:23:56 witherspoon kernel: Hardware name: ASpeed SoC
Feb 02 17:23:56 witherspoon kernel: [<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
Feb 02 17:23:56 witherspoon kernel: [<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
Feb 02 17:23:56 witherspoon kernel: [<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
Feb 02 17:23:56 witherspoon kernel: [<80110d2c>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
Feb 02 17:23:56 witherspoon kernel: [<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
Feb 02 17:23:56 witherspoon kernel: [<801f8970>] (sysfs_create_dir_ns) from [<802a2940>] (kobject_add_internal+0x108/0x30c)
Feb 02 17:23:56 witherspoon kernel: [<802a2940>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
Feb 02 17:23:56 witherspoon kernel: [<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
Feb 02 17:23:56 witherspoon kernel: [<802e5fec>] (device_add) from [<8034fd6c>] (fsi_master_register+0x434/0x8ac)
Feb 02 17:23:56 witherspoon kernel: [<8034fd6c>] (fsi_master_register) from [<8035010c>] (fsi_master_register+0x7d4/0x8ac)
Feb 02 17:23:56 witherspoon kernel: [<8035010c>] (fsi_master_register) from [<80350684>] (store_scan+0x7c/0x90)
Feb 02 17:23:56 witherspoon kernel: [<80350684>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
Feb 02 17:23:56 witherspoon kernel: [<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
Feb 02 17:23:56 witherspoon kernel: [<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
Feb 02 17:23:56 witherspoon kernel: [<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
Feb 02 17:23:56 witherspoon kernel: [<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Feb 02 17:23:56 witherspoon kernel: ---[ end trace b0440d7da7bdfdac ]---
Feb 02 17:23:56 witherspoon kernel: ------------[ cut here ]------------
Feb 02 17:23:56 witherspoon kernel: WARNING: CPU: 0 PID: 1150 at /home/fspcibld/ubuntu-witherspoon-custom/openbmc/build/tmp/work-shared/witherspoon/kernel-source/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
Feb 02 17:23:56 witherspoon kernel: kobject_add_internal failed for hub@00 with -EEXIST, don't try to register things

Will investigate.

cbostic, 2017-02-24:
fsi_slave_scan( ) is creating the first hub master detected with the name "hub@00" which is correct.  This represents the hub master of the first P9 CFAM scanned.   Problem is that the second hub master found which would be on the second P9 CFAM is trying to assign the same name and so get the error messages.

There is a bug where the hub name is set right after allocating and zeroing out the hub_master struct.  So, the %02x portion of the name will always be '00'.    Fix would involve using the master->idx field of the local slave struct.  Once that is done the second hub would be named "hub@01" meaning hub off of master 1.

yadlapati, 2017-02-27:
Seeing the same issue on w37 when I tried to poweron

jk-ozlabs, 2017-02-27:
From:

```
Feb 02 17:23:56 witherspoon kernel: sysfs: cannot create duplicate filename '/devices/hub@00'
```

It looks like you're not setting `dev->parent` on the hub device there, as it's trying to create `hub@00` at the root of the device hierarchy.

jk-ozlabs, 2017-02-27:
... and I guess we don't have `hub->master.idx` initialised when we call `dev_set_name`, as we haven't registered the master yet (which allocates the idx). I'd suggest we use a slave-specific identifier instead there, like the slot index from the configuration table.

cbostic, 2017-02-28:
The solution I came up with was to use the hub master engine's upstream master id as a differentiating id.   So for first p9 we'd have hub@00, as before.  And hub off of that hub master would be hub@01.

                       `hub = kzalloc(sizeof(*hub), GFP_KERNEL);
                        if (!hub)
                                return -ENOMEM;

                        device_initialize(&hub->dev);
                        dev_set_name(&hub->dev, "hub@%02x", slave->master->idx);`

gkeishin, 2017-03-06:
We still see this  on
```
root@witherspoon:~# cat /etc/os-release | grep VERSION_ID
VERSION_ID="v1.99.3-39-g2cac4f7"
root@witherspoon:~# cat /etc/version
20170306031742
```

Had a word with @shenki and he confirms this the same.

- [x] Journald log 
[journald_v1.99.3-39-g2cac4f7_20170306031742_FSI_trace.txt](https://github.com/openbmc/openbmc/files/821497/journald_v1.99.3-39-g2cac4f7_20170306031742_FSI_trace.txt)

```
Feb 22 06:16:44 witherspoon kernel[777]: ------------[ cut here ]------------
Feb 22 06:16:44 witherspoon kernel[777]: WARNING: CPU: 0 PID: 1567 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/fs/sysfs/dir.c:31 sysfs_warn_dup+0x50/0x70
Feb 22 06:16:44 witherspoon kernel[777]: sysfs: cannot create duplicate filename '/devices/hub@00'
Feb 22 06:16:44 witherspoon kernel[777]: CPU: 0 PID: 1567 Comm: sh Not tainted 4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1
Feb 22 06:16:44 witherspoon kernel[777]: Hardware name: ASpeed SoC
Feb 22 06:16:44 witherspoon kernel[777]: [<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
Feb 22 06:16:44 witherspoon kernel[777]: [<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
Feb 22 06:16:44 witherspoon kernel[777]: [<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
Feb 22 06:16:44 witherspoon kernel[777]: [<80110d2c>] (warn_slowpath_fmt) from [<801f88dc>] (sysfs_warn_dup+0x50/0x70)
Feb 22 06:16:44 witherspoon kernel[777]: [<801f88dc>] (sysfs_warn_dup) from [<801f8970>] (sysfs_create_dir_ns+0x74/0x88)
Feb 22 06:16:44 witherspoon kernel[777]: [<801f8970>] (sysfs_create_dir_ns) from [<802a2940>] (kobject_add_internal+0x108/0x30c)
Feb 22 06:16:44 witherspoon kernel[777]: [<802a2940>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
Feb 22 06:16:44 witherspoon kernel[777]: [<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
Feb 22 06:16:44 witherspoon kernel[777]: [<802e5fec>] (device_add) from [<8034fd70>] (fsi_master_register+0x438/0x8b0)
Feb 22 06:16:44 witherspoon kernel[777]: [<8034fd70>] (fsi_master_register) from [<80350698>] (store_scan+0x8c/0xb0)
Feb 22 06:16:44 witherspoon kernel[777]: [<80350698>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
Feb 22 06:16:44 witherspoon kernel[777]: [<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Feb 22 06:16:44 witherspoon kernel[777]: ---[ end trace 3a86725a58ea980a ]---
Feb 22 06:16:44 witherspoon kernel[777]: ------------[ cut here ]------------
Feb 22 06:16:44 witherspoon kernel[777]: WARNING: CPU: 0 PID: 1567 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/lib/kobject.c:240 kobject_add_internal+0x290/0x30c
Feb 22 06:16:44 witherspoon kernel[777]: kobject_add_internal failed for hub@00 with -EEXIST, don't try to register things with the same name in the same directory.
Feb 22 06:16:44 witherspoon kernel[777]: CPU: 0 PID: 1567 Comm: sh Tainted: G        W       4.7.10-2763f3eb775cf760b1c0f28d283da6df6c42f061 #1
Feb 22 06:16:44 witherspoon kernel[777]: Hardware name: ASpeed SoC
Feb 22 06:16:44 witherspoon kernel[777]: [<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
Feb 22 06:16:44 witherspoon kernel[777]: [<80105558>] (show_stack) from [<80110cbc>] (__warn+0xc4/0xf0)
Feb 22 06:16:44 witherspoon kernel[777]: [<80110cbc>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
Feb 22 06:16:44 witherspoon kernel[777]: [<80110d2c>] (warn_slowpath_fmt) from [<802a2ac8>] (kobject_add_internal+0x290/0x30c)
Feb 22 06:16:44 witherspoon kernel[777]: [<802a2ac8>] (kobject_add_internal) from [<802a2bc0>] (kobject_add+0x7c/0xb4)
Feb 22 06:16:44 witherspoon kernel[777]: [<802a2bc0>] (kobject_add) from [<802e5fec>] (device_add+0x118/0x53c)
Feb 22 06:16:44 witherspoon kernel[777]: [<802e5fec>] (device_add) from [<8034fd70>] (fsi_master_register+0x438/0x8b0)
Feb 22 06:16:44 witherspoon kernel[777]: [<8034fd70>] (fsi_master_register) from [<80350698>] (store_scan+0x8c/0xb0)
Feb 22 06:16:44 witherspoon kernel[777]: [<80350698>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
Feb 22 06:16:44 witherspoon kernel[777]: [<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
Feb 22 06:16:44 witherspoon kernel[777]: [<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Feb 22 06:16:44 witherspoon kernel[777]: ---[ end trace 3a86725a58ea980b ]---
Feb 22 06:16:45 witherspoon systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
```

cbostic, 2017-03-06:
The original problem for which this issue was opened was that the two hub devices detected on first and second P9's were both named 'hub@00'.   The fix that went in ensured that the second P9's hub was named 'hub@01'.   Problem though is that on re-scan the old hub devices are not removed prior to trying to add hub@00 and hub@01 again.   So, we get that same issue but for a slightly different reason.

Will put up a fix for this today.

gkeishin, 2017-03-06:
@cbostic awesome..  Let me know when the fix is up.. Thanks

cbostic, 2017-03-07:
These will fix the rescan producing duplicate file name issue:

http://patchwork.ozlabs.org/patch/735992/
The following depends on the previous:
http://patchwork.ozlabs.org/patch/736019/

williamspatrick, 2017-03-10:
This should be closed now?

gkeishin, 2017-03-10:
I still see till yesterday build.. today, in new build  this haven't shown up yet.. 

I'll close it down once verified in the build.

gkeishin, 2017-03-10:
@cbostic I see this build is good and didn't see in multiple poweron.. 
unless you have something else to address.. we can close this.
```
root@witherspoon:~# cat /etc/os-release  | grep VERSION_ID
VERSION_ID="v1.99.3-65-ga6f406c"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-ec62e5e4b9bb4f4e9df718fa7aa4d452fa906e1e #1 Fri Mar 10 03:21:37 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

rfrandse, 2017-03-22:
Closing based on last comment on build is good

closed 2017-03-22 by rfrandse