openbmc/openbmc #1203: OBMC images not working on Zaius

yadlapati, 2017-02-24:
Code update on Zaius system failed. Aaron took the BMC chip out and flashed https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=zaius/lastSuccessfulBuild/artifact/images/zaius/flash-zaius-20170224171607 using dediprog. 

Here is the error on serial console. No messages after "booting the kernel"

DRAM Init-V0B-DDR4
0abc1-8Gb-Done
Read margin-DL:0.3647/DH:0.3823 CK (min:0.35)

U-Boot 2016.07 (Feb 09 2017 - 23:37:12 +0000)

DRAM:  1008 MiB
Flash: 64 MiB
In:    serial
Out:   serial
Err:   serial
Net:   aspeednic#1
Hit any key to stop autoboot:  0 
## Loading kernel from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x20080124
     Data Size:    1682280 Bytes = 1.6 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x80001000
     Entry Point:  0x80001000
     Hash algo:    sha1
     Hash value:   a6f069cd718142779251043ffb28287add064ec8
   Verifying Hash Integrity ... sha1+ OK
## Loading ramdisk from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'ramdisk@1' ramdisk subimage
     Description:  obmc-phosphor-initramfs
     Type:         RAMDisk Image
     Compression:  lzma compressed
     Data Start:   0x20220854
     Data Size:    1569805 Bytes = 1.5 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: unavailable
     Entry Point:  unavailable
     Hash algo:    sha1
     Hash value:   97492baf1fd7ad3add81c0bef18de3c1be3a7882
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 20080000 ...
   Using 'conf@1' configuration
   Trying 'fdt@1' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x2021ad80
     Data Size:    23058 Bytes = 22.5 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   80cfa457b9cab74c3bf49d364f6e1a728be9b8d0
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x2021ad80
   Loading Kernel Image ... OK
   Loading Ramdisk to bea0a000, end beb8940d ... OK
   Loading Device Tree to bea01000, end bea09a11 ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.

Dean said he is using this image with out any issues:
https://github.com/sannerd/op-build/tree/p9_bu_17_02_27.canidate_3

geissonator, 2017-02-24:
Joel, a quick look at what went in from Patrick and I, pointed this one being the most suspicious here - https://github.com/openbmc/linux/commit/0ea834fe59dc26f4df5d8bae097a33669da887bd

yadlapati, 2017-02-24:
https://gerrit.openbmc-project.xyz/#/c/2697/
Build with that commit fixed issue on Zaius.

GHF, 2017-02-24:
I was able to reproduce the failure to boot here. I suspected that the issue is this part:

```
Loading Ramdisk to bea0a000, end beb8940d ... OK
Loading Device Tree to bea01000, end bea09a11 ... OK
```

Those are in the newly reserved region and I guessed Linux doesn't like that. In U-Boot, I changed the load locations for the dtb and initrd by setting `fdt_high` and `initrd_high` to `a0000000`:

```
=> setenv fdt_high a0000000
=> setenv initrd_high a0000000
```

When I boot (I'm using a more verbose U-Boot build):

```
   Loading Kernel Image ... OK
   kernel loaded at 0x80001000, end = 0x8019bdb8
## initrd_high = 0xa0000000, copy_to_ram = 1
   Loading Ramdisk to 9fe80000, end 9ffff87e ... OK
   ramdisk load start = 0x9fe80000, ramdisk load end = 0x9ffff87e
Initial value for argc=3
Final value for argc=3
Initial value for argc=3
Final value for argc=3
using: FDT
## device tree at 2021afd0 ... 202209e5 (len=35350 [0x8A16])
   Loading Device Tree to 9fe77000, end 9fe7fa15 ... OK
Initial value for argc=3
Final value for argc=3
## Transferring control to Linux (at address 80001000)...
```

And that does boot.

I also tested that loading either one of dtb or initrd to the reserved region will cause a boot failure, so both need to respect the reserved region.

This means that U-Boot's image repositioning code needs to know about the reserved region(s) specified in the dtb.

GHF, 2017-02-24:
Also to fix the boot failure immediately, we can roll back the kernel change.

Any bricked machines with unhappy kernels can be recovered by booting with the U-Boot `setenv` commands above.

GHF, 2017-02-24:
It looks like Romulus and Witherspoon were unaffected because they had 0x98000000–0x9C000000 reserved, leaving the final 16 MiB of memory for VGA.

U-Boot most likely put the initrd and dtb in that final 16 MiB on those servers' BMCs.

williamspatrick, 2017-02-25:
This gives me a little concern from a security perspective. We cannot have u-boot putting the initramfs and VGA in the same place. Otherwise, when we go through a reset with the host up, the host can modify code after the FIT verification has been done.

kc8apf, 2017-02-25:
U-Boot is just putting everything at the top of memory as it knows it. If
there needs to be a section of RAM reserved because it is used by hardware
directly, U-Boot needs to know that.

shenki, 2017-02-28:
Spent some time looking into fixing the kernel instead. There is precedent for this; the powerpc kernel moves the device tree into a safe place early in the boot process.

The early boot code is quite hairy so it will take some time to work through.

kc8apf, 2017-02-28:
U-Boot already subtracts the VGA region from available memory. What
prevents us from doing that for this other region as well?

closed 2017-04-26 by yadlapati