openbmc/openbmc #1207: openFSI: segfault on reboot when Cronus is enabled

yadlapati, 2017-02-27:
Executing following commands on openFSI enabled BMC driver enables Cronus successfully.
echo "fsi-master" > /sys/bus/platform/drivers/fsi-master-gpio/unbind
pdbg -b fsi -d p9w -p0 getcfam 0xc09
gpioutil -p d0 -d out -v 1

However, I am seeing a segfault and system does not come back after doing a reboot. This is seen on w37 ( single socket pass-1 system). I am not seeing this issue on w42 (two socket pass-2 system)

Steps to recreate on w37:
1. obmcutil poweron
2. run these three commands to enable cronus
3. echo "fsi-master" > /sys/bus/platform/drivers/fsi-master-gpio/unbind
    pdbg -b fsi -d p9w -p0 getcfam 0xc09
    gpioutil -p d0 -d out -v 1
4. obmcutil poweroff
5. reboot

BMC console (telnet 9.3.185.26 2006) will show seg fault details.
Remaining mounts:
tmpfs / tmpfs rw,nosuid,nodev,mode=755 0 0
dev /dev devtmpfs rw,relatime,size=215184k,nr_inodes=53796,mode=755 0 0
proc /proc proc rw,relatime 0 0
sys /sys sysfs rw,relatime 0 0
tmpfs /run tmpfs rw,nosuid,nodev,mode=755 0 0
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1447 at /var/lib/jenkins-slave/workspace/openbmc-build/distro/ubuntu/target/witherspoon/openbmc/build/tmp/work-shared/witherspoon/kernel-source/lib/kobject.c:597 kobject_get+0x30/0x98
Unable to handle kernel paging request at virtual address 31656434
pgd = 9d9bc000
[31656434] *pgd=00000000
Internal error: Oops: 5 [#1] ARM
CPU: 0 PID: 1447 Comm: reboot Not tainted 4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
task: 9e7d39a0 ti: 9d7d4000 task.ti: 9d7d4000
PC is at string+0x2c/0x74
LR is at 0xffffffff
pc : [<802a7028>]    lr : [<ffffffff>]    psr: 00000093
sp : 9d7d5d28  ip : 8072cef4  fp : 000003e0
r10: 9d7d5d54  r9 : 805b1a7f  r8 : 8072cb14
r7 : 8072cef4  r6 : 9d7d5e40  r5 : 8072cb20  r4 : 9d7d5e3c
r3 : 805b24de  r2 : 31656434  r1 : 00000000  r0 : 8072cb20
Flags: nzcv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: 9d9bc008  DAC: 00000051
Process reboot (pid: 1447, stack limit = 0x9d7d4188)
Stack: (0x9d7d5d28 to 0x9d7d6000)
5d20:                   8072cb20 8072cef4 31656434 802a8e80 8072cb20 ffffff04
5d40: ffff0a00 00000c08 8070200c 0000000c 000000cc ffffff04 ffff0a00 61eca1bd
5d60: 807086e0 000003e0 00000000 ffffffff 00000009 802a2390 807086e0 00000000
5d80: 00000000 802a9558 00000000 8013401c 60000013 80175008 00000000 8070200c
5da0: 805b18db 00000000 00000000 00000000 60000013 00000000 00000000 9d7d5e24
5dc0: 805b18db 00000000 00000009 802a2390 00000255 00000000 00000000 801343d0
5de0: 805b1a71 9d7d5e3c 61eca1bd 80110c84 00000255 802a2390 802a2390 8070200c
5e00: 9e798230 9e6c5050 80725560 9e6c9754 807389c0 80110d2c 00000000 9d7d5e24
5e20: 00000000 805b1a71 9d7d5e3c 61eca1bd 9e775118 802a2390 805b1a71 31656434
5e40: 9e775118 00000000 9e79823c 802e4b6c 9e79823c 802e6728 00000000 80707e9c
5e60: 01234567 8070200c 00099ee8 fee1dead 00000000 80128470 01234567 80128720
5e80: 8070200c 80127670 00000000 000000cf 9e4b4000 61eca1bd 0000c350 9d7d5fb0
5ea0: 9d137370 00000054 49e09b50 9e7d39a0 9d7ca0c0 80000007 9d7ca0f4 8010abec
5ec0: 9d7d5ee0 00000001 9d7d5f20 9d7d4000 49e09b50 00000000 00000000 00000000
5ee0: 0000c350 00000007 8070200c 49e09b50 9d7d5fb0 80705f0c 7ec5df98 000ae768
5f00: 00000000 801013cc 00000001 8013e270 00000000 801c5b54 0000c350 00000001
5f20: 7fffffff 9d11cc00 00000000 00000001 00000010 00000002 00000000 00000000
5f40: 9d7d5f1c 61eca1bd 00000000 9d11cc00 9d11c400 9d11cc44 801c8d98 00000000
5f60: 8070cf50 801a57fc 00000000 8070200c 00000000 8070200c 7ec5deb4 00000024
5f80: 80102384 61eca1bd 00000000 00000002 00000004 7ec5deb4 00000058 80102384
5fa0: 9d7d4000 801021c0 00000002 00000004 fee1dead 28121969 01234567 00099ee8
5fc0: 00000002 00000004 7ec5deb4 00000058 00000002 7ec5df98 000ae768 00000000
5fe0: 000ae250 7ec5dd00 0007fc4c 49e09c38 00000010 fee1dead 00000000 00000000
[<802a7028>] (string) from [<802a8e80>] (vsnprintf+0x25c/0x420)
[<802a8e80>] (vsnprintf) from [<802a9558>] (vscnprintf+0xc/0x24)
[<802a9558>] (vscnprintf) from [<8013401c>] (vprintk_emit+0x1c4/0x43c)
[<8013401c>] (vprintk_emit) from [<801343d0>] (vprintk_default+0x1c/0x24)
[<801343d0>] (vprintk_default) from [<80110c84>] (__warn+0x8c/0xf0)
[<80110c84>] (__warn) from [<80110d2c>] (warn_slowpath_fmt+0x44/0x6c)
[<80110d2c>] (warn_slowpath_fmt) from [<802a2390>] (kobject_get+0x30/0x98)
[<802a2390>] (kobject_get) from [<802e4b6c>] (get_device+0x14/0x1c)
[<802e4b6c>] (get_device) from [<802e6728>] (device_shutdown+0x2c/0x120)
[<802e6728>] (device_shutdown) from [<80128470>] (kernel_restart+0xc/0x50)
[<80128470>] (kernel_restart) from [<80128720>] (SyS_reboot+0x124/0x1f4)
[<80128720>] (SyS_reboot) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: e3a01000 31a02003 e151000e 0a000002 (e4d23001) 
---[ end trace 66e92f99de7edad0 ]---
Segmentation fault
Execute reboot -f if all unmounted ok, or exec /init

geissonator, 2017-02-28:
@cbostic or @shenki ?

mdmillerii, 2017-02-28:
Ahh fun.  The WARN is trying to print the name of the kobject to complain that its not initialized.   Obviously it was more corrupt than that :-)

bradbishop, 2017-03-27:
I don't have the exact commit but this is supposed to be fixed as of at least https://github.com/openbmc/linux/commit/168b8cbfd4818342f13e5a9f3998a3da517cbbec (probably earlier).

closed 2017-03-27 by bradbishop