openbmc/openbmc #1214: Romulus: incorrect LPC to SPI address mapping

mine260309, 2017-02-28:
The expected LPC to SPI address mapping should be:
```
devmem 0x1e789088 32 0x30000C00
devmem 0x1e78908c 32 0xFC0003FF
devmem 0x1e630034 32 0x70680000
devmem 0x1e630030 32 0x68600000
devmem 0x1e789080 32 0x00000500
```

But the current Romulus default setting is:
```
root@romulus:/tmp# devmem 0x1e789088
0x30000C00
root@romulus:/tmp# devmem 0x1e78908c
0xFC0003FF
root@romulus:/tmp# devmem 0x1e630034
0x70640000        <== Wrong
root@romulus:/tmp# devmem 0x1e630030
0x64600000        <== Wrong
root@romulus:/tmp# devmem 0x1e789080
0x00000500
```

The prevents hostboot from getting PNOR correctly.

mine260309, 2017-02-28:
Romulus is missing the PNOR SPI address mapping related code in `arch/arm/mach-aspeed/aspeed.c`.
It's the same as Witherspoon and the code is now added and sent to mailing list for review:
```
diff --git a/arch/arm/mach-aspeed/aspeed.c b/arch/arm/mach-aspeed/aspeed.c
index 726b8fa..ec9eecf 100644
--- a/arch/arm/mach-aspeed/aspeed.c
+++ b/arch/arm/mach-aspeed/aspeed.c
@@ -221,6 +221,24 @@ static void __init do_witherspoon_setup(void)
 static void __init do_romulus_setup(void)
 {
        do_common_setup();
+
+       /* Setup PNOR address mapping for 64M flash
+        *
+        *   ADRBASE: 0x3000 (0x30000000)
+        *   HWMBASE: 0x0C00 (0x0C000000)
+        *  ADDRMASK: 0xFC00 (0xFC000000)
+        *   HWNCARE: 0x03FF (0x03FF0000)
+        *
+        * Mapping appears at 0x60300fc000000 on the host
+        */
+       writel(0x30000C00, AST_IO(AST_BASE_LPC | 0x88));
+       writel(0xFC0003FF, AST_IO(AST_BASE_LPC | 0x8C));
+
+       /* Set SPI1 CE1 decoding window to 0x34000000 */
+       writel(0x70680000, AST_IO(AST_BASE_SPI | 0x34));
+
+       /* Set SPI1 CE0 decoding window to 0x30000000 */
+       writel(0x68600000, AST_IO(AST_BASE_SPI | 0x30));
 }

 #define SCU_PASSWORD   0x1688A8A8
```

mine260309, 2017-03-02:
Romulus only has 1 PNOR chip with 64MiB.

But both CE0/1 decoding window shall be defined, because:
1. We need CE0 to map 0x30000000 ~ 0x34000000 (64MiB);
2. By default CE1 maps 0x32000000 ~ 0x38000000 (96MiB)

If we do not write CE1, it will have overlap between CE0 and CE1, and cause the SPI flash controller behave incorrectly.

Quote from AST2500 datasheet:
> The valid address segment for each CE is: StartAdr ≤ CEx < EndAdr.

legoater, 2017-03-02:
This is true. It is not that important for user mode but it would be for command mode which should be used on the LPC bus.

legoater, 2017-03-02:
Lei, 

If you have some time, could you please give a try to my branch : https://github.com/legoater/linux/commits/openbmc-4.7. It contains the mainline driver, read optimisations and the segment registers configuration. 

I have checked it under a palmetto, an ast2500 evb and QEMU.

Thanks,
C.

mine260309, 2017-03-02:
@legoater Yes, I will try your branch tomorrow or maybe next Monday.

mine260309, 2017-03-03:
The linux bricks a Romulus machine (I have not serial console to it), so I have to assume the branch is not working well on Romulus.

legoater, 2017-03-03:
brick ? what do you mean ?  
It runs on witherspoon so I suppose there is an issue in the kernel config, which is not provided by the branch.

mine260309, 2017-03-03:
I mean the OpenBMC failed to boot.
I'll try to get a machine with serial console and test it again.

legoater, 2017-03-03:
ok. Try netbooting first before flashing. I can provide an .its image if you wish.

legoater, 2017-03-03:
I gave it a try and it looks good

```
[    4.130000] aspeed-smc 1e620000.flash-controller: Using IRQ 17
[    4.130000] aspeed-smc 1e620000.flash-controller: DMA support disabled.
[    4.130000] aspeed-smc 1e620000.flash-controller: mx25l25635e (32768 Kbytes)
[    4.130000] aspeed-smc 1e620000.flash-controller: CE0 segment window is 32MB [ 0x20000000- 0x22000000 ]
[    4.130000] aspeed-smc 1e620000.flash-controller: CE1 segment window is 128MB [ 0x22000000- 0x2a000000 ]
[    4.130000] 5 ofpart partitions found on MTD device bmc
[    4.130000] Creating 5 MTD partitions on "bmc":
[    4.130000] 0x000000000000-0x000000060000 : "u-boot"
[    4.130000] 0x000000060000-0x000000080000 : "u-boot-env"
[    4.130000] 0x000000080000-0x0000004c0000 : "kernel"
[    4.390000] 0x0000004c0000-0x000001c00000 : "rofs"
[    4.390000] 0x000001c00000-0x000002000000 : "rwfs"
[    4.390000] aspeed-smc 1e620000.flash-controller: unrecognized JEDEC id bytes: 00, 00, 00
[    4.390000] aspeed-smc 1e620000.flash-controller: Aspeed SMC probe failed -2
[    4.390000] aspeed-smc 1e620000.flash-controller: No DMA. bad IRQ status: 600
[    4.390000] aspeed-smc: probe of 1e620000.flash-controller failed with error -2
[    4.390000] aspeed-smc 1e630000.flash-controller: n25q512ax3 (65536 Kbytes)
[    4.390000] aspeed-smc 1e630000.flash-controller: CE0 segment window is 64MB [ 0x30000000- 0x34000000 ]
[    4.390000] aspeed-smc 1e630000.flash-controller: CE1 segment window is 64MB [ 0x34000000- 0x38000000 ]
```

we should remove the second FMC flash in the device tree

mine260309, 2017-03-23:
This can be closed by http://patchwork.ozlabs.org/patch/734023/ is merged

closed 2017-03-23 by mine260309