openbmc/openbmc #1215: [Witherspoon-W37] - Version ID not fully popluated - Difficult to validate with Driver 588

sivassrr, 2017-02-28:
Problem Description: Taken image from https://openpower.xyz/job/openbmc-build/588/
And its version id details usually embedded in the string  "588-c85f23e19ae273".

Here is the current output of cat /etc/os-release:
```
root@witherspoon:~# cat /etc/version
20170227221530
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-dirty"
VERSION_ID="v1.99.3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-dirty"
BUILD_ID="v1.99.3"
root@witherspoon:~# 
```

Usual Expectation of cat /etc/os-release as follows:
```
root@barreleye:/var/lib/systemd/coredump# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.2-89"
VERSION_ID="v1.99.2-89-gf327ed7-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.2-89"
BUILD_ID="v1.99.2"
root@barreleye:/var/lib/systemd/coredump# cat /etc/version
20170221021702
```

sivassrr, 2017-05-29:
Didnt observe that in the latest release. 
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-12"
VERSION_ID="v1.99.6-12-gcd24073"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-12"
BUILD_ID="v1.99.6"

Hence closing it.

williamspatrick, 2017-05-29:
I don't think this was actually ever an issue. I think that build was originally 1.99.3 exactly.

closed 2017-05-29 by williamspatrick