openbmc/openbmc #1217: eSEL Logging REST data format [ Witherspoon ]

gkeishin, 2017-02-28:
- [x]  eSEL message
`"Message": "org.open_power.Error.Host.Event.Event", `   why there is two `Event` string in the object path name ?

- [x] eSEL duplicate ESEL=  `AdditionalData` data logged ?

- [x] Journal log  
[journald_20170227221530_v1.99.3_IPL_ESEL.txt](https://github.com/openbmc/openbmc/files/806472/journald_20170227221530_v1.99.3_IPL_ESEL.txt)

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-dirty"
VERSION_ID="v1.99.3-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-dirty"
BUILD_ID="v1.99.3"
root@witherspoon:~# cat /etc/version
20170227221530
root@witherspoon:~#

```
```
root@witherspoon:~# busctl tree xyz.openbmc_project.Logging | cat
└─/xyz
  └─/xyz/openbmc_project
    └─/xyz/openbmc_project/logging
      ├─/xyz/openbmc_project/logging/entry
      │ ├─/xyz/openbmc_project/logging/entry/1
      │ └─/xyz/openbmc_project/logging/entry/2
      └─/xyz/openbmc_project/logging/internal
        └─/xyz/openbmc_project/logging/internal/manager
root@witherspoon:~#
```

```
{
  "data": {
    "/xyz/openbmc_project/logging/entry/1": {
      "AdditionalData": [
        "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 ba df 87 1a 00 00 00 04 ba e1 72 1d 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 20 90 00 00 20 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 04 7d 88 00 00 00 00 40 04 5a 88 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 "
      ], 
      "Id": 1, 
      "Message": "org.open_power.Error.Host.Event.Event", 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
      "Timestamp": 1487743704623
    }, 
    "/xyz/openbmc_project/logging/entry/2": {
      "AdditionalData": [
        "ESEL=00 00 df 00 00 00 00 20 00 04 07 a4 07 aa 00 00 50 48 00 30 01 00 07 00 00 00 00 05 2f ba 1d 58 00 00 00 05 2f bb 2a be 42 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 21 90 00 00 21 55 48 00 18 01 00 07 00 10 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 07 05 00 48 00 00 00 e0 00 00 05 00 00 00 00 00 00 20 00 00 00 01 00 00 00 05 00 01 02 00 ff ff 01 90 00 48 42 43 31 30 30 37 30 35 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 12 23 01 00 02 00 05 01 00 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 01 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 50 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 2b a3 94 00 00 00 00 40 2b ac 68 00 00 00 00 40 2b ad c4 00 00 00 00 40 2b cd 08 00 00 00 00 40 2b ce f0 00 00 00 00 40 0d c0 0c 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 55 44 00 fc 01 15 31 00 01 28 00 42 49 32 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f4 00 00 00 00 00 00 00 f4 00 00 00 05 00 00 00 00 00 00 00 2b 1c ce 72 d7 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 01 00 00 00 24 00 00 00 2b 1c d1 ef 0b 00 00 00 c7 00 10 43 4f 10 87 d4 ee 00 00 0c cc 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 2b 1c d2 21 95 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 2b 1c d2 d1 80 00 00 00 c7 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 2b 1e 37 7f e7 00 00 00 c7 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 ", 
        "ESEL=00 00 df 00 00 00 00 20 00 04 07 a4 07 aa 00 00 50 48 00 30 01 00 07 00 00 00 00 05 2f ba 1d 58 00 00 00 05 2f bb 2a be 42 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 21 90 00 00 21 55 48 00 18 01 00 07 00 10 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 07 05 00 48 00 00 00 e0 00 00 05 00 00 00 00 00 00 20 00 00 00 01 00 00 00 05 00 01 02 00 ff ff 01 90 00 48 42 43 31 30 30 37 30 35 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 12 23 01 00 02 00 05 01 00 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 01 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 50 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 2b a3 94 00 00 00 00 40 2b ac 68 00 00 00 00 40 2b ad c4 00 00 00 00 40 2b cd 08 00 00 00 00 40 2b ce f0 00 00 00 00 40 0d c0 0c 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 55 44 00 fc 01 15 31 00 01 28 00 42 49 32 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f4 00 00 00 00 00 00 00 f4 00 00 00 05 00 00 00 00 00 00 00 2b 1c ce 72 d7 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 01 00 00 00 24 00 00 00 2b 1c d1 ef 0b 00 00 00 c7 00 10 43 4f 10 87 d4 ee 00 00 0c cc 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 2b 1c d2 21 95 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 2b 1c d2 d1 80 00 00 00 c7 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 2b 1e 37 7f e7 00 00 00 c7 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 "
      ], 
      "Id": 2, 
      "Message": "org.open_power.Error.Host.Event.Event", 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
      "Timestamp": 1487743708210
    }, 
    "/xyz/openbmc_project/logging/internal/manager": {}
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

gkeishin, 2017-02-28:
It's easily recreatable..   

```
{
  "data": {
    "/xyz/openbmc_project/logging/entry/1": {
      "AdditionalData": [
        "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 b3 98 92 61 00 00 00 04 b3 99 e0 1a 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 26 90 00 00 26 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 04 7d 88 00 00 00 00 40 04 5a 88 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 "
      ], 
      "Id": 1, 
      "Message": "org.open_power.Error.Host.Event.Event", 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
      "Timestamp": 1487743754538
    }, 
    "/xyz/openbmc_project/logging/entry/2": {
      "AdditionalData": [
        "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 b2 62 81 23 00 00 00 04 b2 63 7e 5b 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 29 90 00 00 29 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 04 7d 88 00 00 00 00 40 04 5a 88 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 mc_project.Logging.Entry", 
        "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 b2 62 81 23 00 00 00 04 b2 63 7e 5b 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 29 90 00 00 29 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 04 7d 88 00 00 00 00 40 04 5a 88 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 63 33 31 37 30 63 30 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 mc_project.Logging.Entry"
      ], 
      "Id": 2, 
      "Message": "org.open_power.Error.Host.Event.Event", 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
      "Timestamp": 1487744140075
    }, 
    "/xyz/openbmc_project/logging/internal/manager": {}
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

dkodihal, 2017-02-28:
FYI @tomjoseph83 this could be what you're chasing, seems to be a case of an non-terminated string?

williamspatrick, 2017-03-01:
I agree with the duplicate error name being removed.  I suspect this just requires changing the file from 'host/event.yaml' to 'host.yaml'.

The duplicate additional data might be fixed by https://gerrit.openbmc-project.xyz/#/c/2689/

williamspatrick, 2017-03-01:
Deepak also mentioned the missing null terminator problem which was just merged into openbmc master.

gkeishin, 2017-03-02:
Today on latest driver, I see this one time so far where the Addtional data is missing. Trying to recreate, but here is some info see if that helps

```
Feb 22 06:11:11 witherspoon ipmid[872]: Received Host Event
Feb 22 06:11:11 witherspoon ipmid[872]: A host system event was received
Feb 22 06:11:11 witherspoon phosphor-log-manager[782]: Failed to find metadata
Feb 22 06:12:02 witherspoon system_manager.py[813]: Running System State: HOST_BOOTED
```

```
root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/1 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        0                                        emits-change writable
.Id                                 property  u         1                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1487743871755                            emits-change writable
root@witherspoon:~#

/xyz/openbmc_project/logging/enumerate
{
  "data": {
    "/xyz/openbmc_project/logging/entry/1": {
      "AdditionalData": [], 
      "Id": 1, 
      "Message": "org.open_power.Error.Host.Event.Event", 
      "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency", 
      "Timestamp": 1487743871755
    }, 
    "/xyz/openbmc_project/logging/internal/manager": {}
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

- [x] Jounral log 
[journald_v1.99.3-20-g98927fe_20170301221558_eSEL.txt](https://github.com/openbmc/openbmc/files/812783/journald_v1.99.3-20-g98927fe_20170301221558_eSEL.txt)

geissonator, 2017-03-02:
I don't see the normal ESEL traces from IPMID in this case, it's almost like the host just said "here's an error" but didn't send down the data?  Unless these input/output error's were the data?
```
Feb 22 06:11:07 witherspoon ipmid[872]: Failed to call the method: Input/output error
Feb 22 06:11:07 witherspoon ipmid[872]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 22 06:11:07 witherspoon phosphor-gevent[1266]: ::e063:c176:8e91:f402%1207800232 - - [2017-02-22 06:11:07] "POST /login HTTP/1.1" 200 320 0.199865
Feb 22 06:11:08 witherspoon ipmid[872]: Failed to call the method: Input/output error
Feb 22 06:11:08 witherspoon ipmid[872]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 22 06:11:08 witherspoon ipmid[872]: Failed to call the method: Input/output error
Feb 22 06:11:08 witherspoon ipmid[872]: Dest: :1.4, Path: /org/openbmc/HostIpmi/1
Feb 22 06:11:11 witherspoon ipmid[872]: Received Host Event
Feb 22 06:11:11 witherspoon ipmid[872]: A host system event was received
Feb 22 06:11:11 witherspoon phosphor-log-manager[782]: Failed to find metadata
Feb 22 06:12:02 witherspoon system_manager.py[813]: Running System State: HOST_BOOTED
```

gkeishin, 2017-03-03:
@anoo1 @geissonator   Another one hit on latest driver
```
Feb 22 06:41:03 witherspoon ipmid[885]: Received Host Event
Feb 22 06:41:03 witherspoon ipmid[885]: A host system event was received
Feb 22 06:41:07 witherspoon ipmid[885]: Received Host Event
Feb 22 06:41:07 witherspoon ipmid[885]: A host system event was received

		
/xyz/openbmc_project/logging/enumerate
{
    "/xyz/openbmc_project/logging/internal/manager": {},
    "/xyz/openbmc_project/logging/entry/2": {
        "Timestamp": 1487745667378,
        "AdditionalData": [
            "ESEL=00 00 df 00 00 00 00 20 00 04 07 a4 07 aa 00 00 50 48 00 30 01 00 07 00 00 00 00 05 aa 5d 24 24 00 00 00 05 aa 5e 37 e9 42 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 0f 90 00 00 0f 55 48 00 18 01 00 07 00 10 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 07 05 00 48 00 00 00 e0 00 00 05 00 00 00 00 00 00 20 00 00 00 01 00 00 00 05 00 01 02 00 ff ff 01 90 00 48 42 43 31 30 30 37 30 35 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 24 01 06 01 00 01 54 41 4b 00 00 00 06 00 00 00 00 00 00 00 00 00 00 00 30 23 01 00 02 00 05 01 00 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 01 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 50 01 03 01 00 00 00 00 00 00 03 f3 24 00 00 00 00 00 03 b6 94 00 00 00 00 40 2a 93 94 00 00 00 00 40 2a 9c 68 00 00 00 00 40 2a 9d c4 00 00 00 00 40 2a bd 08 00 00 00 00 40 2a be f0 00 00 00 00 40 0d c0 0c 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 61 30 34 33 37 62 32 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 55 44 00 fc 01 15 31 00 01 28 00 42 49 32 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 f4 00 00 00 00 00 00 00 f4 00 00 00 05 00 00 00 00 00 00 00 2f 1d d3 78 36 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 01 00 00 00 24 00 00 00 2f 1d d7 5f 77 00 00 00 c7 00 10 43 4f 10 87 d4 ee 00 00 0c cc 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 2f 1d d7 8c 82 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 2f 1d d8 26 47 00 00 00 c7 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 2f 1f 52 c0 e3 00 00 00 c7 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 "
        ],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 2,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    },
    "/xyz/openbmc_project/logging/entry/1": {
        "Timestamp": 1487745664094,
        "AdditionalData": [],
        "Message": "org.open_power.Error.Host.Event.Event",
        "Id": 1,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Emergency"
    }
}

root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/1 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        0                                        emits-change writable
.Id                                 property  u         1                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1487745664094                            emits-change writable
root@witherspoon:~#
```
`Journald pretty print for eSEL`

```
root@witherspoon:~# journalctl MESSAGE="Received Host Event" -o json-pretty | cat
{
        "__CURSOR" : "s=30746501b36b4f11bfd377510a289abc;i=753;b=25768ded8d47417292eccc94340f7614;m=4af685ab;t=54918c7d08f6c;x=6277282dc717d93d",
        "__REALTIME_TIMESTAMP" : "1487745663930220",
        "__MONOTONIC_TIMESTAMP" : "1257670059",
        "_BOOT_ID" : "25768ded8d47417292eccc94340f7614",
        "PRIORITY" : "6",
        "_UID" : "0",
        "_GID" : "0",
        "_SYSTEMD_SLICE" : "system.slice",
        "_MACHINE_ID" : "b14a51f848ab4ffda6bfe645978e47d6",
        "_HOSTNAME" : "witherspoon",
        "_TRANSPORT" : "journal",
        "_CAP_EFFECTIVE" : "3fffffffff",
[... cut ...] 00 01 00 00 00 00 00 00 00 02 00 00 00 2c 00 00 00 2f 1d d7 8c 82 00 00 00 c7 00 08 43 4f 83 b9 0f 9a 00 00 0d 0f 00 00 00 00 00 00 00 02 00 00 00 24 00 00 00 2f 1d d8 26 47 00 00 00 c7 00 18 43 4f 1b ce fc ec 00 00 0a 1e 00 00 00 00 00 05 00 01 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 34 00 00 00 2f 1f 52 c0 e3 00 00 00 c7 00 08 43 4f 65 c9 23 ee 00 00 07 6b 00 01 00 00 00 00 00 00 00 00 00 24 ",
        "_SOURCE_REALTIME_TIMESTAMP" : "1487745667334100"
}
root@witherspoon:~#
```

gkeishin, 2017-03-03:
- [x] Journal log 
[journald_v1.99.3-32-g05b7eba_20170302221516_eSEL.txt](https://github.com/openbmc/openbmc/files/816010/journald_v1.99.3-32-g05b7eba_20170302221516_eSEL.txt)

closed 2017-03-09 by williamspatrick