openbmc/openbmc #1218: [BarrelEye] need FANs control abilities for Chip W83795G that is more or less similar with NCT7904D chip of BarrelEye

yadron, 2017-02-28:
**Chip W83795G (fans auto-control)**
Planning usage of chip W83795G with OpenPower+OpenBMC. Overall what I read:
Linux driver: http://lxr.free-electrons.com/source/drivers/hwmon/w83795.c
Chip spec doc: http://www.nuvoton.com/resource-files/Nuvoton_W83795G_W83795ADG_Datasheet_V1.43.pdf (see p.94 with "REGISTER SUMMARY – BANK2" for fans registers that describes how to setup auto-control fans by this chip).

**Chip NCT7904D (fans auto-control in BarrelEye)**
Similar BarrelEye what I read:
Chip NCT7904D used in BarrelEye
Chip spec doc: https://www.nuvoton.com/resource-files/NCT7904D_Datasheet_V1.44.pdf
Linux driver: http://lxr.free-electrons.com/source/drivers/hwmon/nct7904.c

Talking about similar func of OpenBMC for both chips.

Before Dec 2016 fans were always max-speed with OpenBMC on BarrelEye and it is so with stable 1.0 OpenBMC build still _(why? does it mean that NCT7904D registers were not initialized correctly on BMC boot before Dec 2016?)_.

Starting from master build (after ~Jan 2017), fans being silent on BarrelEye _(but the fan_control.c of "obmc-control-fan" have nothing apart from set_max and in-dev set_present functions, see below, so it is a question for me what is the difference between release stable 1.0 and master - probably nct7904.c was disabled in release 1.0, so default registers of NCT7904D were not initialized correctly or something similar; what was the difference for fans between stable 1.0 and master?)_.

Regarding needs of new features, need fans control in OpenBMC that will be almost the same both for BarrelEye and W83795G based platform (means that need ability turning off fans-smart-control by chip for manual controlling of fans or at least need ability updating registers of chip with max/low values of expected temperatures/values), see: #939  
Simple scenario example: if OpenBMC sees that CPU temp too high (and CPU Freq becomes lower than expected as result), then OpenBMC updates min/max values for temp in registers of FANs chip (it looks like chip itself only uses expected motherboard temp sensors, so software control of CPUs params could be required -see below-, at least if talking about W83795G, but in Barreleye with NCT7904D maybe the same sensors used).

Currently OpenBMC doesn't support manual control of fans speed, but it is planned before OpenBMC 2.0 release (as "Phase 4"): #1185 
Apart from manual control, it would be a good idea if I can say "make fans a step faster" and "make fans a step lower" instead of disabling auto-control (if I get it right, when I do manual control of FANs speed, then auto-control by chip turned off).

It must be emergency turn on max FANs speed by firmware when temp higher than some level controlled by chip (how to test if it works?). With these chips I guess this should work without OpenBMC working at all or even if by usage of OpenBMC overheated (say manually).

The auto-control fans in OpenBMC (based on temperatures from OCC too) looks like planned also: #940  
, but it looks like this is not required for these two chips (because this already auto-controlled by "SMART FANTM control" and both POWER8/OpenPower CPU can help itself like OCC decreases CPU freq if CPU temp becomes too high).
I think that for these two chips it is enough setting best consts of chip registers once instead of auto-update by OpenBMC like #940 , is it right?

**OpenBMC currently**
Current code status, see #725   and https://gerrit.openbmc-project.xyz/#/c/120/ (fan_control.c that only allows set_to_max and in-dev set_present).
Fans speed monitoring issue: #1173  

**OpenBMC sources**
For Barreleye, the only module being included to OpenBMC currently is "obmc-control-fan" (the daemon of BMC named /usr/sbin/fan_control.exe). Other fan modules doesn't exist or not found.

**Linux Kernel**
This issue describes where driver enabled in Linux kernel setup:
#499  
(see "
$ grep NCT ./meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig
CONFIG_SENSORS_NCT7904=y
")
and similar W83795G probably need these kernel params (not used currently in OpenBMC):
SENSORS_W83795,
SENSORS_W83795_FANCTRL
(looks like nobody tried it yet in OpenBMC, is it so?).

yadron, 2017-02-28:
BTW, I've found fans-auto-control code for Barreleye (and looks like not committed to OpenBMC yet): https://lists.ozlabs.org/pipermail/openbmc/2016-May/002953.html
- see fan_algorithm.c there.

Probably the author of this algorithm can answer if it is or similar required for Barreleye.

adamliyi, 2017-03-14:
FYI. There is currently new FAN control module under review: https://gerrit.openbmc-project.xyz/#/c/2514/7

williamspatrick, 2017-03-15:
@yadron : You titled this issue 'Barreleye' but that doesn't seem accurate.  The Barreleye doesn't have a W83795G chip in it.

If you are working on another system and would like to have W83795 placed into the kernel config, you are welcome to issue a patch for the kernel config on the mailing list.  I believe an alternative is in Yocto to place a config snippet for the machine you are building.

What does the W83795_FANCTRL enable?  Does this enable setting the fan speeds via the hwmon interface?  Or does it enable some fan control algorithm?  We would want hwmon interface for fan speed control but probably not a native / built-in fan control algorithm.

yadron, 2017-05-18:
@williamspatrick , we've just started development of fans-control algorithm for W83795.

I'm seeing similar phosphor-fan and fan-monitor development already started in OpenBMC too (ex. build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-fan/1.0-r1/git/control/*pp files ).

What did you mean by "We would want hwmon interface for fan speed control, but not a native / built-in fan control algorithm"? Do you mean that phosphor-hwmon (that is meta-phosphor/common/recipes-phosphor/sensors/phosphor-hwmon) should control fans? How it will work? Will "phosphor-fan" service ask phosphor-hwmon service for setting fans speed or what?
Can I describe it like
`"sensors -> phosphor-hwmon -> phosphor-fan -> phosphor-hwmon -> LinuxDriver -> fans"`
then meaning that hwmon read data from sensors and phosphor-fan from hwmon and then phosphor-fan asks hwmon updating fans speed?
And what service "fan-monitor" does then?

Looking to https://github.com/openbmc/openbmc/issues/1584 , which mode should we use for Barreleye or other similar platforms? There are these modes available:
"0=disabled, 1=pwm, 2=rpm, 3=automatic"
inside chips (both W83795G and NCT7904D).

Should we use "0==disabled" mode so OpenBMC fully controls fans or any other could be used too?
Which modes of these will be supported by phosphor-fan+fan-monitor services?

For more information, read current issues:
1. https://github.com/openbmc/openbmc/issues/1584 - outdated fans services being removed right now.
2. https://github.com/openbmc/openbmc/issues/1567 - describes what phosphor-fan-control will do during start booting host.
3. https://github.com/openbmc/openbmc/issues/1600 - overall list of issues for latest fans-development sprints.
4. https://github.com/openbmc/openbmc/issues/1342 - describes some ideas behind fans-control from the point of watchdog view (like set fans to max speed before booting host and then start auto-fans control only if BMC booted successfully and all services "fan control, monitor, hwmon, presence-detect, maybe others" works as expected).

williamspatrick, 2017-05-18:
The data flow is something like this:

`hw tach sensor -> linux driver (hwmon) -> phosphor-hwmon -> phosphor-fan`
`hw temp sensor -> ... same`
`phosphor-fan calculates fan adustments -> phosphor-hwmon -> linux driver -> fan controller hw`

When I said:
> We would want hwmon interface for fan speed control, but not a native / built-in fan control algorithm
I think I was referring to not using "mode 3" but using mode 1 or mode 2 so that the BMC is actively controlling the fan speed directly.  I assume the "mode 3" algorithm in the hardware can only take into account its own thermal sensor, where as you likely want the BMC to include temperatures that come from the processor chips via the OCC.  phosphor-hwmon currently only supports "mode 2" behavior.

There were some email discussions on the mailing list between @pstrinkle and @msbarth.  I would suggest reading through those and either continuing a discussion with @msbarth via the mailing list or contacting him on IRC.  If you would like me to arrange a phone conference between our fan control team and yourself I can make that happen as well.

zahrens, 2017-07-21:
@yadron Are there any other discussions needed in this issue?  Thank you

gtmills, 2018-05-23:
@yadron Still an issue or can this be closed?

gtmills, 2018-07-05:
Closing. no update in a year+ and barreleye is no longer supported. Please reopen if still an issue. thanks!

closed 2018-07-06 by gtmills