openbmc/openbmc #1222: OpenFSI: interrupt support for hub links.

rfrandse, 2017-02-28:


cbostic, 2017-03-08:
Work on this is in process.

cbostic, 2017-03-22:
Work item breakdown:
1.  Refactor master's slave list and slave's engine list traversal to use device_for_each_child( ) utility.
      ( already factored into the community reviewed source)
2.  Set primary slave's SRSIM hub sourced interrupt masks according to discovery of new hub links and their locations
3.  Update fsi_master_irq( ) to detect non local engine IRQ sources (hub)
4.  On detect of hub sourced IRQ retrieve the hub connected slave that originated the IRQ
      a.  Read in primary slave SRSIS
      b.  Retrieve slave on index indicated by SRSIS
5.  Read in the IRQ status register of the hub connected slave and dispatch client engine's interrupt handlers
      a.  Clear interrupt condition bits 
6.  Differentiate engine interrupts and general FSI bus error interrupts on hub slave.
      a.  Create FSI IRQ handler to deal with these errors and leverage the existing error handling framework in FSI.
6.  Merge common primary and hub slave client engine dispatch handlers into one utility.
7.  Create a round robin scheduler to fairly handle all hub sourced interrupts
8.  Check for unexpected sources of interrupts
      a.  CMFSI sourced IRQs,  no links so no interrupts should be reported
      b.  HMFSI sourced IRQs off of an upstream HMFSI engine, no links so no interrupts should be reported

wghoffa, 2017-04-13:
Do you know when the work is planned to resolve this?

closed 2017-05-10 by rfrandse