openbmc/openbmc #1226: OpenFSI is segfaulting

shenki, 2017-03-01:
This is stopping me from booting Zaius

```
root@zaius:~# pdbg probe -a
Segmentation fault
```

```
Unable to handle kernel NULL pointer dereference at virtual address 0000003c
pgd = bc070000
[0000003c] *pgd=bd2b4831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#296] ARM
CPU: 0 PID: 5808 Comm: sh Tainted: G      D W       4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
task: bcd6e9a0 ti: bcdb2000 task.ti: bcdb2000
PC is at store_scan+0xc/0x90
LR is at kernfs_fop_write+0x140/0x1ac
pc : [<80350d70>]    lr : [<801f79e4>]    psr: a0000013
sp : bcdb3eb0  ip : 80350d64  fp : 00000000
r10: 00000051  r9 : 01f73bf0  r8 : bcdb3f78
r7 : bd28248c  r6 : bd230f40  r5 : 00000002  r4 : 00000000
r3 : 00000002  r2 : bd230f40  r1 : 80717f04  r0 : bdcb5d10
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: bc070008  DAC: 00000051
Process sh (pid: 5808, stack limit = 0xbcdb2188)
Stack: (0xbcdb3eb0 to 0xbcdb4000)
3ea0:                                     00000002 bd282480 bd230f40 801f79e4
3ec0: 00000000 00000000 801f78a4 bce89d20 8070200c 01f73bf0 01f73bf0 00000002
3ee0: 00000000 801a32d0 8070200c 01f74bf4 bcdb3fb0 80705d0c 01f74bf0 000001ff
3f00: 4a4fe71c 801012f4 58b443d9 393b8026 58b61442 068f2e5a 58b61442 068f2e5a
3f20: 000012e2 00000000 00000001 bce89d20 bcdb3f78 744b1990 00000002 bce89d20
3f40: bcdb3f78 01f73bf0 01f73bf0 801a3548 bce89d20 01f73bf0 00000002 bce89d20
3f60: 8070200c 8070200c bce89d20 01f73bf0 00000002 801a36d4 00000000 00000000
3f80: 00001000 744b1990 00000000 00000002 01f73bf0 4a51b140 00000004 80102384
3fa0: bcdb2000 801021c0 00000002 01f73bf0 00000001 01f73bf0 00000002 00000000
3fc0: 00000002 01f73bf0 4a51b140 00000004 00000002 7eb0d9ec 000e9124 00000000
3fe0: 00000000 7eb0d964 4a4482a4 4a4a00dc 60000010 00000001 00000000 00000000
[<80350d70>] (store_scan) from [<801f79e4>] (kernfs_fop_write+0x140/0x1ac)
[<801f79e4>] (kernfs_fop_write) from [<801a32d0>] (__vfs_write+0x28/0x12c)
[<801a32d0>] (__vfs_write) from [<801a3548>] (vfs_write+0xb8/0x144)
[<801a3548>] (vfs_write) from [<801a36d4>] (SyS_write+0x54/0xac)
[<801a36d4>] (SyS_write) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: 0001a36e e92d4070 e1a05003 e590404c (e594003c) 
---[ end trace f530c079033f1d06 ]---
```

cbostic, 2017-03-01:
Attempted to recreate on zaius6 but could not.   Possibly some sequence leading up to the probe is required?

```
root@zaius:/sys/bus/platform/devices/fsi-master# pdbg probe -a
ucd9000 0-0064: Device ID UCD90160|2.3.4.0000|110603
Overall System Topology
=======================

Kernel FSI Backend@0
    MFSI Port@100000
        FSI2PIB@1000
    FSI2PIB@1000

CFAMs Present
=============

0: Kernel FSI Backend@0
1: MFSI Port@100000

Processors/Chiplets Present
===========================

Processor-ID 0: FSI2PIB@1000
Processor-ID 1: FSI2PIB@1000
```

cbostic, 2017-03-01:
Appears same build as described above:
```
Linux version 4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 (jenkins@openbmc-builder2) (gcc version 5.3.0 (GCC) ) #1 Fri Feb 24 22:51:37 UTC 2017
```

cbostic, 2017-03-01:
I hit the same seg fault by:
```
$ root@zaius:/sys/bus/platform/drivers/fsi-master-gpio# echo fsi-master > unbind
$ pdbg -b fsi -d p9w -p 0 getcfam 0xc09
$ gpioutil -p d0 -d out -v 1
$ obmcutil poweroff
```

Apparently obmcutil poweroff  results in a echo command to the fsi-master-gpio scan file.
Since the gpio fsi master is unbound at this point the scan attempts to dereference a master-> pointer.   

Two issues I see here.
1.  Why is obmcutil poweroff  attempting to echo anything to the scan file?
2.  FSI driver needs to remove the scan file on unbind.

cbostic, 2017-03-01:
Fix put up for review:
http://patchwork.ozlabs.org/patch/734385/

rfrandse, 2017-03-27:
Patches Merged

closed 2017-03-27 by rfrandse

bradbishop, 2017-03-27:
I don't have the exact commit but this is supposed to be fixed as of at least openbmc/linux@168b8cb (probably earlier).