openbmc/openbmc #1227: slpd spamming journal

shenki, 2017-03-01:
Running `v1.99.2-143-g0286b0c-dirty` (0286b0c) every few seconds I'm seeing this:

```
Mar 01 01:21:00 zaius slpd[841]: Channel::Read : DataIn Fd[5] Req[56] Recv[56]
Mar 01 01:21:00 zaius slpd[841]: SLP unable to find the service=service:directory-agent
Mar 01 01:21:00 zaius slpd[841]: SLP Error rc=10
Mar 01 01:21:00 zaius slpd[841]: Channel::Read : DataIn Fd[5] Req[75] Recv[75]
Mar 01 01:21:00 zaius slpd[841]: SLP unable to find the service=service:directory-agent
Mar 01 01:21:00 zaius slpd[841]: SLP Error rc=10
Mar 01 01:21:02 zaius slpd[841]: Channel::Read : DataIn Fd[5] Req[75] Recv[75]
Mar 01 01:21:02 zaius slpd[841]: SLP unable to find the service=service:directory-agent
Mar 01 01:21:02 zaius slpd[841]: SLP Error rc=10
Mar 01 01:21:06 zaius slpd[841]: Channel::Read : DataIn Fd[5] Req[56] Recv[56]
Mar 01 01:21:06 zaius slpd[841]: SLP unable to find the service=service:directory-agent
Mar 01 01:21:06 zaius slpd[841]: SLP Error rc=10
```

geissonator, 2017-03-01:
#983 greatly reduced this but yeah, for some reason Zaius systems only are getting all this invalid SLP traffic.  Not sure what to do though, we probably want to know about this, but we really only want to know once :)

shenki, 2017-03-01:
Here's my workaround in case others are interested:
```
root@zaius:~# systemctl stop slpd-lite
root@zaius:~# systemctl mask slpd-lite   
Created symlink from /etc/systemd/system/slpd-lite.service to /dev/null.
```

ratagupt, 2017-03-22:
I have minimized the logs.Now it will be one..
SLP unable to find the service=service:directory-agent.

https://gerrit.openbmc-project.xyz/#/c/3170/
But I feel for that we should bring iptables for such kind of DOS attack.It would be good in other cases for eg:- rest/NetIpmid
1) Malicious attack:- TCP Half open socket

rfrandse, 2017-05-27:
https://gerrit.openbmc-project.xyz/#/c/3170/
resolves #1227

closed 2017-05-27 by rfrandse