openbmc/openbmc #1228: REST logging interface Stopped functioning [Witherspoon]

gkeishin, 2017-03-01:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-6"
VERSION_ID="v1.99.3-6-g1603890-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-6"
BUILD_ID="v1.99.3"
root@witherspoon:~# cat /etc/version
20170228221548

IBM-witherspoon-ibm-OP9_v1.14_1.7
        op-build-v1.14-60-g7462097
        buildroot-2016.11-4-g6b9fd80
        skiboot-8c51fda
        hostboot-6966799
        linux-4.9.8-pa8cf74e
        petitboot-v1.4.1-p38036cb
        witherspoon-xml-c75ecc4
        occ-4141b5f
        hostboot-binaries-446c2ea
        capp-ucode-1bb7503
```

` At BMC standby`

```
/xyz/openbmc_project/ 
{
  "data": [
    "/xyz/openbmc_project/sensors", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/ObjectMapper", 
    "/xyz/openbmc_project/software", 
    "/xyz/openbmc_project/logging"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
```

`Power on and the logging went kaput. Probably an Object Maaper or logging issue here`

```
Feb 22 06:17:54 witherspoon ipmid[888]: Received Host Event
Feb 22 06:17:54 witherspoon ipmid[888]: A host system event was received
Feb 22 06:17:58 witherspoon ipmid[888]: Received Host Event
Feb 22 06:17:58 witherspoon ipmid[888]: A host system event was received
Feb 22 06:18:00 witherspoon ipmid[888]: Failed to call the method: Input/output error
Feb 22 06:18:00 witherspoon ipmid[888]: Dest: :1.10, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:49 witherspoon ipmid[888]: Failed to call the method: Input/output error

/xyz/openbmc_project/ 
{
  "data": [
    "/xyz/openbmc_project/sensors", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/ObjectMapper", 
    "/xyz/openbmc_project/software"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
```

```
root@witherspoon:/etc/systemd/system# cat ./obmc-standby.target.wants/xyz.openbmc_project.Logging.service
[Unit]
Description=Phosphor Log Manager

[Service]
ExecStart=/usr/sbin/phosphor-log-manager
Restart=always
Type=dbus
BusName=xyz.openbmc_project.Logging

[Install]
WantedBy=obmc-standby.target
root@witherspoon:/etc/systemd/system# journalctl --no-pager | grep -i "Phosphor Log Manager"
Feb 22 06:05:11 witherspoon systemd[1]: Starting Phosphor Log Manager...
Feb 22 06:05:15 witherspoon systemd[1]: Started Phosphor Log Manager.
root@witherspoon:/etc/systemd/system# systemctl status xyz.openbmc_project.Logging.service | cat
● xyz.openbmc_project.Logging.service - Phosphor Log Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Logging.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-02-22 06:05:15 UTC; 1h 1min ago
 Main PID: 779 (phosphor-log-ma)
   CGroup: /system.slice/xyz.openbmc_project.Logging.service
           └─779 /usr/sbin/phosphor-log-manager

Feb 22 06:05:11 witherspoon systemd[1]: Starting Phosphor Log Manager...
Feb 22 06:05:15 witherspoon systemd[1]: Started Phosphor Log Manager.
root@witherspoon:/etc/systemd/system# ps -aef | grep phosphor-log-manager
  779 root      46:28 /usr/sbin/phosphor-log-manager
root@witherspoon:/etc/systemd/system#

```

- [x]  Journal log 
[journald_v1.99.3-6-g1603890_20170228221548_logging.txt](https://github.com/openbmc/openbmc/files/810128/journald_v1.99.3-6-g1603890_20170228221548_logging.txt)

gkeishin, 2017-03-01:
There is a ipmi failure while making dbus call https://github.com/openbmc/openbmc/issues/1230   which @tomjoseph83 is looking at separately . So not sure if they are related in some way..

geissonator, 2017-03-01:
George and I discussed.  The summary is that the REST api to look at phosphor-logging interfaces works great at BMC_READY but then later after booting the REST api no longer works to look at phosphor-logging interfaces.  The phosphor-logging service seems to be fine though.  Some things we need:
- A journal of when it works and when it doesn't work
- Use the busctl tool to determine if the DBUS interface is working (i.e. is this just a REST thing).
- Do other REST api's work ok?

gkeishin, 2017-03-01:
The REST API to logging doesn't work since that object is no longer to be seen active when failed.. the busctl I didn't check since as per the `systemctl` and the 'ps of the logging` still show it was running and active as I pasted in my comment above as well.  Perhaps the the journal should have the data for it. So I uploaded which I could think of. If I see again I'll keep the system in failure state to debug.

gkeishin, 2017-03-01:
No longer seen in latest fixes.. all came clean..

geissonator, 2017-03-03:
Please reopen if seen again George

closed 2017-03-03 by geissonator