openbmc/openbmc #1229: Security scanner reporting critical and few medium priority vulnerabilities

prkatti1, 2017-03-01:
Security scanning reporting few high severity & few medium priority issues,
System details 
************************
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-6"
VERSION_ID="v1.99.3-6-g1603890-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-6"
BUILD_ID="v1.99.3"

Criticla issue:
Dropbear SSH Server < 2016.72 Multiple Vulnerabilities

Medium Issues
*******************************
Web Server Generic XSS
Description

The remote host is running a web server that fails to adequately sanitize request strings of malicious JavaScript. A remote attacker can exploit this issue, via a specially crafted request, to execute arbitrary HTML and script code in a user's browser within the security context of the affected site.

Solution
+++++++++++++++++++++++++++++++
Contact the vendor for a patch or upgrade.

SSL Medium Strength Cipher Suites Supported
Description

The remote host supports the use of SSL ciphers that offer medium strength encryption. Nessus regards medium strength as any encryption that uses key lengths at least 64 bits and less than 112 bits, or else that uses the 3DES encryption suite.

Note that it is considerably easier to circumvent medium strength encryption if the attacker is on the same physical network.
Solution

Reconfigure the affected application if possible to avoid use of medium strength ciphers.

SSL 64-bit Block Size Cipher Suites Supported (SWEET32)
Description

The remote host supports the use of a block cipher with 64-bit blocks in one or more cipher suites. It is, therefore, affected by a vulnerability, known as SWEET32, due to the use of weak 64-bit block ciphers. A man-in-the-middle attacker who has sufficient resources can exploit this vulnerability, via a 'birthday' attack, to detect a collision that leaks the XOR between the fixed secret and a known plaintext, allowing the disclosure of the secret text, such as secure HTTPS cookies, and possibly resulting in the hijacking of an authenticated session.

Proof-of-concepts have shown that attackers can recover authentication cookies from an HTTPS session in as little as 30 hours.

Note that the ability to send a large number of requests over the same TLS connection between the client and server is an important requirement for carrying out this attack. If the number of requests allowed for a single connection were limited, this would mitigate the vulnerability. However, Nessus has not checked for such a mitigation.
Solution

Reconfigure the affected application, if possible, to avoid use of all 64-bit block ciphers. Alternatively, place limitations on the number of requests that are allowed to be processed over the same TLS connection to mitigate this vulnerability.

prkatti1, 2017-03-01:
@sivassrr these issues are found when we run security scanner on bmc

prkatti1, 2017-03-01:
Medium issue

mDNS Detection (Remote Network)

Description

The remote service understands the Bonjour (also known as ZeroConf or mDNS) protocol, which allows anyone to uncover information from the remote host such as its operating system type and exact version, its hostname, and the list of services it is running. 

This plugin attempts to discover mDNS used by hosts that are not on the network segment on which Nessus resides.
Solution

Filter incoming traffic to UDP port 5353, if desired.
Output
Nessus was able to extract the following information :

  - mDNS hostname       : witherspoon.local.

  - Advertised services :
    o Service name      : witherspoon._ssh._tcp.local.
      Port number       : 22
    o Service name      : sftp._sftp._tcp.local.
      Port number       : 22
    o Service name      : witherspoon._sftp-ssh._tcp.local.
      Port number       : 22
    o Service name      : phosphor_rest._phosphor_rest._tcp.local.
      Port number       : 443
    o Service name      : obmc_console._obmc_console._tcp.local.
      Port number       : 2200

williamspatrick, 2017-03-09:
@prkatti1 

- Dropbear will be updated soon.
- We can look at disabling some of the crypto ciphers.
- Do you have details on the XSS?  The text here is really generic.
- mDNS is a feature we purposefully added.  Why would this be a security problem?

joseph-reynolds, 2018-05-25:
For dropbear: according to https://matt.ucc.asn.au/dropbear/CHANGES:
Version "2018.76 - 27 February 2018" is the latest version.
- twofish ciphers are now disabled in the default configuration
Version "2015.67 - Wednesday 28 January 2015":
- Add option to disable CBC ciphers.

My interpretation is that the compromised CBC cyphers are still enabled in the latest available version of dropbear; specifically AES and 3DES are still enabled.  (Compared to OpenSSH which disabled all CBC cyphers in 7.6 per https://www.openssh.com/releasenotes.html).

If so, we'll have to change our bitbake recipe to disable CBCs in dropbear.

joseph-reynolds, 2018-05-25:
The XSS will be fixed (I think) when we go to nginx.

joseph-reynolds, 2018-05-25:
The mDNS issue is a security issue because it gives more information to an attacker about how magnify their DOS attacks by 100%.  The main idea is that you should drop mDMS requests if they come from outside your local network.

gtmills, 2019-01-23:
@joseph-reynolds What remains on this?

closed 2019-08-24 by stale[bot]