openbmc/openbmc #1233: IPMI network connection fails

shenki, 2017-03-01:
I was attempting to use ipmi over the network and I could not get it to work:

`"v1.99.2-143-g0286b0c-dirty` (0286b0c)

```
$ ipmitool -H zaius6.aus.stglabs.ibm.com -I lanplus -U root -P 0penBmc
Error: Unable to establish IPMI v2 / RMCP+ session
```

```
Mar 01 23:17:55 zaius netipmid[795]: << RAKP12
Mar 01 23:17:55 zaius netipmid[795]: I> Channel::Read : DataIn Fd[3] Req[17] Recv[17]
Mar 01 23:17:55 zaius netipmid[795]: >> GetChannelCapabilities
Mar 01 23:17:55 zaius netipmid[795]: << GetChannelCapabilities
Mar 01 23:17:55 zaius netipmid[795]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
Mar 01 23:17:55 zaius netipmid[795]: >> openSession
Mar 01 23:17:55 zaius netipmid[795]: << openSession
Mar 01 23:17:55 zaius netipmid[795]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
Mar 01 23:17:55 zaius netipmid[795]: >> RAKP12
Mar 01 23:17:55 zaius netipmid[795]: Failed to get /org/openbmc/control/chassis0 bus name: Connection timed out
```

williamspatrick, 2017-03-02:
Based on #1236 I guess this was user error at this time?  Looks like it was the wrong username / password because we don't have the story for PAM based authentication for IPMI done?

shenki, 2017-03-02:
Yep. We can close this one, or change it to "IPMI placeholder password should match root user placeholder password" to avoid confusion.

tomjoseph83, 2017-03-16:
@shenki Can we get this closed? The default password is changed to match 'root user' password.

geissonator, 2017-03-20:
Closing based on above comments.

closed 2017-03-20 by geissonator