openbmc/openbmc #1240: OpenFSI seg fault on P9 #2 slave read via raw sysfs file

cbostic, 2017-03-02:
Seen on zaius6.   From Matt Light:
```
root@zaius:~# /tmp/croserver-openbmc-debug 
 --- Cronus Socket Deamon Initialized ...
 --- Built on : Feb 17 2017 19:30:19 CST
opened /sys/bus/platform/devices/fsi-master/slave@00:00/raw
adal_mbx_get_register(281A)
before lseek(fd = 5, ...)
lseek rc = 10344
before read(fd = 5, ...)
read rc = 4
adal_mbx_get_register(281A) rc = 4, value = A01E0001
opened /sys/devices/hub@00/slave@01:00/raw
adal_mbx_get_register(281A)
before lseek(fd = 5, ...)
lseek rc = 10344
before read(fd = 5, ...)
Segmentation fault
```
----
```
Unable to handle kernel NULL pointer dereference at virtual address 0000001c
pgd = bd3e4000
[0000001c] *pgd=bd2fa831, *pte=00000000, *ppte=00000000
Internal error: Oops: 17 [#14] ARM
CPU: 0 PID: 2100 Comm: croserver-openb Tainted: G      D W       4.7.10-39eb5a44bfd49fe530f6b68c4d99ab2d8d391be0 #1
Hardware name: ASpeed SoC
task: bde11280 ti: bd2b6000 task.ti: bd2b6000
PC is at fsi_slave_read+0x14/0x30
LR is at 0x0
pc : [<8034f9c4>]    lr : [<00000000>]    psr: 00000013
sp : bd2b7e60  ip : 00000000  fp : 7e9a0aa4
r10: 00000000  r9 : bd31030c  r8 : bd361080
r7 : 00000000  r6 : fffffffc  r5 : 00000000  r4 : 8034fa10
r3 : 00102868  r2 : bd361080  r1 : 00102868  r0 : bc85c000
Flags: nzcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 00c5387d  Table: bd3e4008  DAC: 00000051
Process croserver-openb (pid: 2100, stack limit = 0xbd2b6188)
Stack: (0xbd2b7e60 to 0xbd2b8000)
7e60: bd361080 00000004 8034fa10 8034f9d8 bd361080 00000004 00002868 8034fb64
7e80: 00002868 00000000 8034fb1c 7e9a0ab8 bd361080 801f80fc 00002868 00000000
7ea0: 00000004 00000001 00000004 00000004 bd310300 bd2b7f78 7e9a0ab8 801f7b24
7ec0: 00002868 00000000 00000004 801f7a50 bce93d40 8070200c bd2b7f78 7e9a0ab8
7ee0: 00000004 801a30ac 803cb830 00000418 00000000 00000000 00000000 bd2b7f00
7f00: 00000002 00000000 bd16cc08 00000001 bce930c0 00000000 00000020 00000000
7f20: 00000000 94b83956 00000000 bce93d40 bd2b7f78 94b83956 00000004 bce93d40
7f40: 7e9a0ab8 bd2b7f78 7e9a0ab8 801a323c bce93d40 7e9a0ab8 00000004 bce93d40
7f60: 8070200c 8070200c bce93d40 7e9a0ab8 00000004 801a3628 00002868 00000000
7f80: 00000001 94b83956 00000000 00018038 00018bd8 00003200 00000003 80102384
7fa0: bd2b6000 801021c0 00018038 00018bd8 00000005 7e9a0ab8 00000004 00000005
7fc0: 00018038 00018bd8 00003200 00000003 00003200 00000000 7e9a1748 7e9a0aa4
7fe0: 00000000 7e9a0a74 00052c84 4a53068c 60000010 00000005 e3a05000 ea000084
[<8034f9c4>] (fsi_slave_read) from [<8034f9d8>] (fsi_slave_read+0x28/0x30)
[<8034f9d8>] (fsi_slave_read) from [<8034fb64>] (fsi_slave_sysfs_raw_read+0x48/0x5c)
[<8034fb64>] (fsi_slave_sysfs_raw_read) from [<801f80fc>] (sysfs_kf_bin_read+0x84/0x8c)
[<801f80fc>] (sysfs_kf_bin_read) from [<801f7b24>] (kernfs_fop_read+0xd4/0x174)
[<801f7b24>] (kernfs_fop_read) from [<801a30ac>] (__vfs_read+0x28/0x128)
[<801a30ac>] (__vfs_read) from [<801a323c>] (vfs_read+0x90/0xfc)
[<801a323c>] (vfs_read) from [<801a3628>] (SyS_read+0x54/0xac)
[<801a3628>] (SyS_read) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
Code: e590c0e0 e5d0e0e8 e88d000c e1a03001 (e59c401c) 
---[ end trace 8417936cac786ac9 ]---
```

cbostic, 2017-03-02:
I couldn't recreate this on w5.  All zaius systems are presently occupied.

cbostic, 2017-03-02:
I was able to recreate this on w5 provided I re-scanned the fsi bus.   Seems to be a de-reference of the master->read call where master is NULL.

cbostic, 2017-03-07:
Submitted two patches for review that will take care of this:

http://patchwork.ozlabs.org/patch/735992/
The following depends on the previous:
http://patchwork.ozlabs.org/patch/736019/

bradbishop, 2017-03-27:
@cbostic looks like the first one has been merged but the second one had some feedback?

rfrandse, 2017-03-28:
fixed in head

closed 2017-03-28 by rfrandse