openbmc/openbmc #1241: Unable to open IPMI session: no free sessions left

shenki, 2017-03-03:
Running `v1.99.3-32-g05b7eba` (05b7eba), I ran the following. I was told that we need to restart the ipmi network daemon due to #1110, so I did that, and it still didn't work.

The full journal is attached.

```
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis status
Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis on
Error in open session response message : insufficient resources for session

Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis on
Error in open session response message : insufficient resources for session

Error: Unable to establish IPMI v2 / RMCP+ session
joel@ka1 ~ 
$ ipmitool -H palm5-bmc -I lanplus -P PASSW0RD chassis on
Error in open session response message : insufficient resources for session
```

```
Mar 03 00:01:53 palmetto netipmid[810]: I> Channel::Read : DataIn Fd[3] Req[17] Recv[17]
Mar 03 00:01:53 palmetto netipmid[810]: >> GetChannelCapabilities
Mar 03 00:01:53 palmetto netipmid[810]: << GetChannelCapabilities
Mar 03 00:01:53 palmetto netipmid[810]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
Mar 03 00:01:53 palmetto netipmid[810]: >> openSession
Mar 03 00:01:53 palmetto netipmid[810]: E> No free sessions left: Active: 5 Allowed: 5
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x00000000
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x100a3391
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x2f33acb3
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x41ff21ed
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x4a556d18
Mar 03 00:01:53 palmetto netipmid[810]: E> Active Session: 0x75682194
Mar 03 00:01:53 palmetto netipmid[810]: No free sessions left
Mar 03 00:01:53 palmetto netipmid[810]: openSession : Problem opening a session
```
[journal.txt](https://github.com/openbmc/openbmc/files/815421/journal.txt)

shenki, 2017-03-03:
I noticed we have changed the placeholder password to `OpenBmc`. This issue highlights we are not freeing sessions when authentication fails.

tomjoseph83, 2017-03-17:
Even in the case of authentication failure, the session is created. The maximum session's currently supported is 5 sessions(excluding the Session ID 0). So you are seeing the insufficient resources for session error code. The Session inactivity timeout comes into play '60 sec' and the session gets closed.

The plan is to have a different timeout for Authentication failure, in which case we would have a timeout of '5 sec'. A minor fix is needed to provide that.

closed 2017-03-20 by williamspatrick

Ad1bDaw, 2019-04-29:
I had a similar issue with a surprising twist. This was simply due to a cabling mixup.  I'm running a Dell c6420 and the IPMI cable was actually plugged into one of the LOMs, once changed back to the dedicated port, it worked.